MarlinFirmware/Marlin #16468: CR10V2 with factory fitted BLTouch V3.0 wont deploy

nicwilson58, 2020-01-05:
### Bug Description

CR10_V2 with factory fitted BLTouch will not deploy the BLTouch at all.   I have a working 1.16 that works with BLTouch fine,  I have compared the configuration.h and configuration_adv.h and all looks very much the same with no obvious reasons why it will not deploy.      As a further proof of error,  if I connect the optoprint, when it connects,  the terminal gives an error and terminates the connection.  The error from optoprint is:  

"Recv: Error:STOP called because of BLTouch error - restart with M999
Changing monitoring state from "Connecting" to "Error: STOP called because of BLTouch error - restart with M999""

I looked through other bugs and issues and could not find anything which resolved the issue.

I have attached a zip file with the two configuration files.   I have also included a second set of config files from version 1.16 that works perfectly with BLTouch.
[config 1.16 working.zip](https://github.com/MarlinFirmware/Marlin/files/4022832/config.1.16.working.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Factory unmodifed CR10_V2 with factory fitted BLTouch V3.0,  ( I also tested this on a friends CR10_V2 with a BLTouch fitted after purchase,  the BLTouch is the same and is an original BLTouch V3.0, his machine does the exact same thing.

**Expected behavior:**   BLTouch to deploy on G28, or from BLTouch Menu Deploy

**Actual behavior:**  BLTouch does not deploy on G28 or on BLTouch Menu.

[Config201.zip](https://github.com/MarlinFirmware/Marlin/files/4022831/Config201.zip)

nicwilson58, 2020-01-05:
Hi,

I have since managed to get this to work,  but outside of the configuration files and it was due to changes made in 1.1.7 and I have now tested this to be an issue on 1.1.7 and all versions afterwards,  which is why 1.1.6 works.

Changes to SERVO0_PIN value,  CR10_V2 uses RAMPS_13 and the definition of the SERVO0_PIN was altered in 1.1.7 from 11 to 7.  Changing this back to 11, resolved the issue and the BLTouch now deploys and works properly.

InsanityAutomation, 2020-01-05:
There is a creality specific pins file in new versions. Also, they line up with ramps14 not 13.

nicwilson58, 2020-01-05:
Thanks,  I found "pins_RAMPS_CREALITY.h" which seems to have the correct pins,  however that include file is not referenced anywhere in the code,  only the MELZI_CREALITY one is, so its not clear which board to use.

sjasonsmith, 2020-01-05:
> however that include file is not referenced anywhere in the code

It is referenced from the pins.h file, at least in the bugfix-2.0.x branch I am looking at. The board name is `RAMPS_CREALITY`.

nicwilson58, 2020-01-05:
hi, ok will look at the bugfix-2.0x branch

nicwilson58, 2020-01-05:
Well that solved the BLTouch issue,   but now it wont home.  It moves to the centre of the bed.  deploys the bltouch,  but the Z axis does not move down,  it then retracts the bltouch,  and deploys it again, leaves it deployed with a blue LED and stops.

InsanityAutomation, 2020-01-05:
Try comparing what you have with my branch here https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin_2.0 - Its already been tested on quite a few machines.

nicwilson58, 2020-01-06:
I will try your branch as soon as I compile it as there does not seem to be a pre-compiled version for the CR10-V as there is for most other models. I will let you know how the test goes with BLTouch.

nicwilson58, 2020-01-06:
Before changing the firmware I ran an M119 test.   Om my current 1.1.6 build the Z_min shows Open and shows Triggered when its triggered as expected as it works fine.  On the BugFix2.0 version,  the z_min shows Triggered always,  and even if I invert the Z endstop it still shows Triggered, therefore explaining why Z does not move down.

InsanityAutomation, 2020-01-06:
> As there is no direct machine for CR10 V2, which do you recommend that has been tested, the closest to me seems CR10SV2 but not sure about the lergeFilSensor?

Yes the S in there is erroneous lol I need to kill it sometime. Polarity is forced for a bltouch, can't change it. V2 probes and 1.1.x bugfix as well as 2.0 support a switch mode, so when retracted it always reports triggered. Deploy from LCD then check m119 again.

boelle, 2020-01-08:
@nicwilson58 did you try to deploy from lcd as @InsanityAutomation suggested?

nicwilson58, 2020-01-08:
> BLTouch V3.0 wont deploy (#16468)
> <mention@noreply.github.com>

Hi,I have not had the chance yet as I am now back at work,  I will be doing that when I get a sec to flash the firmware again and run some more tests. Stay tuned I will be back  ;)Nic

nicwilson58, 2020-01-11:
M119 still showed triggered.  However I found a version of your source that is working fine on my CR10V2 + BLTouchj,  so I have included the configuration files here.   When I get chance I will compare them and make the same changes and check if your version works.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4049161/Marlin.zip)

nicwilson58, 2020-01-11:
Hi,   I have made some more changes to the configuration files and now its working.   The main cause of the instability and freezing seemed to be the baud rate of 250000 which most folk seem to be using. 
 Lowering it to 115200 seems to have totally removed the issues.I was having.  It seems to be working fine right now so I will run it for several days and report back then.

nicwilson58, 2020-01-11:
There maybe more users that have issues as I did when trying pre-compiled firmware, or trying to compile their own,  so if its OK I might fork this source and create one specific to Creality.

hilleke-dirk, 2020-01-11:
I have the same problem with the Sidewinder X1. It is a skr-gen-l board.
The marlin 2.0 Version of the wagster-mod works (April 2019) but newer versions have the homing problem.

nicwilson58, 2020-01-12:
Hi,can you describe exactly what it does?Nic

hilleke-dirk, 2020-01-13:
I have the homing-Z problem, liked described obove.
"It moves to the centre of the bed. deploys the bltouch, but the Z axis does not move down, it then retracts the bltouch, and deploys it again, leaves it deployed with a blue LED and stops."

Mesh-bed-levelling works fine, but not the homing.

boelle, 2020-01-18:
@nicwilson58 do you still have problems if using the latest bugfix 2.0.x ?

nicwilson58, 2020-01-19:
I gave up on most firmware either not working with BLTouch or behaving oddly.  I ended up trying the firmware listed above my InsanityAutomation and it worked immediately.  I have since been making changes to that source to better tune the V2 and for Creality printers in general.   I am starting to get requests for it and its working quite well now,  so I may make a fork,   if I get the time and desire from programming for work  ;)

boelle, 2020-01-26:
then we just have to wait for @InsanityAutomation figuring a PR :-)

closed 2020-08-04 by github-actions[bot]