MarlinFirmware/Marlin #16469: [BUG] RAMPS 1.6 on Arduino MEGA 2560 BLtouch shows triggered on zmin

ebraiman, 2020-01-05:
### Bug Description

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4022901/Configuration.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

physical connection:
Using BLtouch v3.1 or V2.0
Connect Servo pin on pin 4 (yellow) , orange on 5v and black on -
Connect Zmin with white wire on Zmin and black wire on ground for Zmin

Open a fresh copy of Marlin 2.0.x
Remove comment for BLTouch
Remove comment for Num_servos and change 3 to 1
Compile and upload.

1. Remove comment for BLTouch
2. Remove comment for Num_servos and change 3 to 1
3. Compile and upload.
4. Open Pronterface and enter gcode M119

**Expected behavior:** [What you expect to happen]

Z-min is open (BLTouch 3.1)
z_min: TRIGGERED (BLtouch 2.0)

**Actual behavior:** [What actually happens]

Z-min is open and (BLtouch 3.1) but does not extend on Z-home stuck in processing
Z-min is triggered (BLtouch 2.0) but does not extend on Z-home stuck in processing

#### Additional Information
Tested on Marlin 1.1.8 and worked correctly

* Provide pictures or links to videos that clearly demonstrate the issue.

https://www.youtube.com/watch?v=R0HeFV9nKCM

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ebraiman, 2020-01-05:
Correction:
Expected behavior: [What you expect to happen]

Z-min is open (BLTouch 3.1)
z_min: open (BLtouch 2.0)

Actual behavior: [What actually happens]

Z-min is open and (BLtouch 3.1) but does not extend on Z-home stuck in processing
Z-min is triggered (BLtouch 2.0) but does not extend on Z-home stuck in processing

boelle, 2020-01-10:
did you test with 2.0.1 or bugfix 2.0.x ?

ebraiman, 2020-01-10:
2.0.1 for sure and bugfix has not worked since late summer.

boelle, 2020-01-10:
2.0.1 is just a frozen copy of bugfix 2.0.x so that should not work either

bugfix 2.0.x is updated at least every 2 days so there is a BIG chance that it has been fixed since late summer

boelle, 2020-01-18:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-18 by boelle