MarlinFirmware/Marlin #16477: Marlin 2.0 homing issue

mcarpenterjr, 2020-01-06:
Bought this..

Beshine 3D Printer Board MKS SGen_L 32-bit Controller Compatible with Marlin2.0 and Smoothie Firmware.Support Multiple Types of Drivers (with TMC2209 5) https://www.amazon.com/dp/B07V1STHWX/ref=cm_sw_r_cp_apa_i_vrUeEbRNEEY7C 

board that came with TMC2209s.

Flashed marlin 2.0 using platformio for vscode, board boots, octoprint connects, I can jog the axises, except when I issue a G28 the z axis jogs up and nothing happens, no bounce or motion in any axis. The machine the sets zero.

Is this a config issue or something more? Anyone work with the above board before?

GMagician, 2020-01-06:
What does report M119?

mcarpenterjr, 2020-01-06:
I see what you're getting at here.

```
Send: M119
Recv: Reporting endstop status
Recv: x_min: TRIGGERED
Recv: y_min: TRIGGERED
Recv: z_min: open
Recv: ok
```
Not sure why X and Y show as triggered and not Z aswell, since no end stops are connected and Sensorless Homing/Probing are enabled.

GMagician, 2020-01-06:
Don't know exactly how sensorless homing works (maybe diag pin signal homes?), but 'triggered' means 'min' input is defined and active and I think this is not correct.

mcarpenterjr, 2020-01-06:
Agreed, even when turning off sensorless homing I get the same issue. Tried shutting off the software limiting also, same affect, machine raises Z and then sets zero.

Vertabreak, 2020-01-06:
unsure if related but i was having homing issues then i enabled end stop noise threshold and the problem went away, worth a try.

boelle, 2020-01-06:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

mcarpenterjr, 2020-01-07:
From the 2.0.1 Branch
[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/4028426/MarlinConfig.zip)

tatusah, 2020-01-07:
What brand and version are those 2209 drivers? Do you have the DIAG pin connected from the drivers to the control board? With some driver versions you might have to do some soldering to get the sensorless homing working.

mcarpenterjr, 2020-01-08:
They're MKS TMC2209 V1.0 __No__ DIAG pins soldered in.
Should I solder some in and try?

I have some other questions, out of the scope of firmware, but I've noticed the board is very similar to the SKR 1.3 boards out there, when putting the jumper in place for the DIAG pin on the limit switches should I unplug the limit switch? And also the plug layout for the driver has 2 plugs on the end for what I would assume is the diag pin and something else, should I solder in the second pin when I solder in the DIAG pin?

ellensp, 2020-01-08:
For so called SENSORLESS_HOMING requires the DIAG pins on the stepper module. These are connected to the min endstops pins so the controller can detect "endstops"

ellensp, 2020-01-08:
You should not have limit switches and be trying to use SENSORLESS_HOMING on the same axis.

GMagician, 2020-01-08:
> You should not have limit switches and be trying to use SENSORLESS_HOMING on the same axis.

or disconnect diag as input pin, disable SENSORLESS_HOMING, and use limit switches

boelle, 2020-01-08:
and check if it also happens on bugfix 2.0.x

mcarpenterjr, 2020-01-14:
So I'm going to close this, it looks to be a hardware issue more so than anything. The increasing number of people I have talked to all seem to have abandoned the board I am working with and gone with something different. I think I'm going to see how an SKR 1.3 will do. Thanks, everyone for the help.

closed 2020-01-14 by mcarpenterjr