MarlinFirmware/Marlin #16478: Control both extruders from LCD or TFT panel for maintenance

1979mgb, 2020-01-06:
My Dual color printer works. I can control both Extruders using Pronterface by selecting either E0 or E1 then selecting how much, how fast and to extrude or retract. 
I would like to be able to do the same from my printers display. Printer Info reports that I have 2 extruders.
Yet choosing the motion function for extruder only allows control of E0
My request is to be able to do the following:
1. [press Encoder Knob, dial to Motion]
2. [press Encoder Knob, dial to Move Axis]
3. [press Encoder Knob, dial to Extruder]
4. [press encoder Knob, dial to select E0, E1, etc.]
5. [press encoder Knob, dial to select 10mm, 1mm, .1mm]
6. [press encoder Knob, dial to select speed]
7. [press encoder Knob, dial + or - direction to choose total distance]
8. [press encoder to back out]

Except for choosing speed, this function already exists for 1 extruder only it seems. I have watched many videos read many forums searched this site and the answers seem to say that this is not a function in marlin. Have read that it is part of the mixer function. 

This would be useful for maintenance such as calibrating all of the extruders or changing filament.
I have dual drive BMG extruders and using the release levers doesn't let go of the filament so I motor it out. 
It simply would be easier than relying on Pronterface which of course is a very useful tool for so much else.
Thank you for considering this matter. If this function already exists and I have simply missed it I apologize.

ellensp, 2020-01-06:
Add CUSTOM_USER_MENUS in Configuration_adv.h to switch tools (ie send T0 or T1)  The extruder tools work on what ever tool is active.   To implement exactly what you want would require rewriting the menu system...

boelle, 2020-01-06:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

1979mgb, 2020-01-07:
Sorry for delay, new to zipping took awhile to figure out how. Here they are I hope.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4028950/Configuration.zip)

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4028951/Configuration_adv.zip)

1979mgb, 2020-01-07:
It should be known that CHANGE FILAMENT is working for both E1 and E2. 
But I would still like to incrementally control both extruders. Just curious why Pronterface refers to tool 0 and 1, Marlin configuration.h refers to E0 and E1 yet the Ender 3 stock LCD lists E1 and E2 in Change Filament but no separate definition in the MOTION/MOVE AXIS/EXTRUDER menu. My TFT35 in Extruder menu shows E0 and "nozzle" button that has 01 labeled on it(of course I only have 1 nozzle).  Nomenclature seems to have no standard. It of course is easy to figure what is meant since only dealing with 2 steppers. Sorry, I am  rambling on the opening line is the important bit.

1979mgb, 2020-01-07:
Make note of edit in opening request.

boelle, 2020-01-08:
btw, is this on bugfix 2.0.x?

1979mgb, 2020-01-09:
Yes it is.

Vertabreak, 2020-01-09:
2 motors 1 nozzle has two options mix or single nozzle sounds like you should define single nozzle.

1979mgb, 2020-01-09:
Hi, I have configured it both ways and tried to control extruders from motion menu. Presently I am on mix. Could try 2 extruders 1 nozzle again just to be sure.

1979mgb, 2020-01-09:
This printer is an A10M with a BTT SKR V1.3 installed.
Now configured as 2 extruder 1 nozzle. Still can only control 1 extruder in Motion menu.
Mixing menu gone.
Change filament Menu still shows both extruders.

Vertabreak, 2020-01-09:
to switch extrudes with mix you have to adjust the mix ratio from 100/0 to 0/100 everything is virtually defined if you use what i call cyclops listed as single nozzle you can control each motor directly.

1979mgb, 2020-01-09:
Not having any problems with mixing or controlling for printing. Looking for maintenance functionality.
For instance extruder calibration. select E0 or E1 dial for 100mm extrude measure repeat until done. 
Change filament menu is wait for the unload cycle, then load and purge.
The motion Menu can select xyz and 1 extruder and tell it exactly what to do and how much and in what direction-but only for 1 extruder. I know that I can do this with Pronterface but connecting to my computer is not always convenient also Pronterface can be tricky  to connect at times.
So when set up as Cyclops how do you control each motor directly from the display controller?
Thanks for all the help.

Vertabreak, 2020-01-09:
with cyclops in the motion control menu you will find E0 & E1 & E2 should you have 3 like i do.

1979mgb, 2020-01-09:
configuration.h define extruders 2 define single nozzle. Is there something else to configure because I still only get 1 extruder to choose from in motion control menu.

Vertabreak, 2020-01-09:
`  
#define EXTRUDERS 2
#define SINGLENOZZLE`

Vertabreak, 2020-01-09:
if you have yet to update marlin today you maybe have a bug in your eeprom limiting extruders to 1.

1979mgb, 2020-01-09:
Thank you for picture, I know I am a noob but did not think I was this bad. I was using a tft display at first and could not figure out how to select another extruder. So put on the Ender LCD and by that time I had switched configuration over to mixing. Now I am back on cyclops and simply did not turn dial enough to see E1 E2. I was pressing the knob thinking it would take me to a selection menu like the TFT does. Now I will try the mixing and turn the knob and see what happens.
Thank you for teaching me how to turn a knob. Sure am glad you folks are patient.

Vertabreak, 2020-01-09:
yea photos and videos help alot the devil is in the details.

1979mgb, 2020-01-09:
Am curious though, Why Extruder, Extruder 1 and Extruder 2 as though there are 3 on my machine?

Vertabreak, 2020-01-09:
the first extruder E0 and the listed E1 are the same entry the menu needs adjusted it should list E0 & E1 ect.

Vertabreak, 2020-01-09:
ill have a look at the code see if i can fix it up and submit the fix.

1979mgb, 2020-01-09:
A speed selection would still be nice, any possibility?

1979mgb, 2020-01-09:
So should I close this issue?

Vertabreak, 2020-01-09:
you can #define ULTIPANEL or exnensible ui and edit the speed of xyze move speed.
`
#if EITHER(ULTIPANEL, EXTENSIBLE_UI)
  #define MANUAL_FEEDRATE { HOMING_FEEDRATE_XY, HOMING_FEEDRATE_XY, HOMING_FEEDRATE_Z, 120 } // Feedrates for manual moves along X, Y, Z, E from panel
  #define SHORT_MANUAL_Z_MOVE 0.025 // (mm) Smallest manual Z move (< 0.1mm)
  #if ENABLED(ULTIPANEL)
    #define MANUAL_E_MOVES_RELATIVE // Display extruder move distance rather than "position"
    #define ULTIPANEL_FEEDMULTIPLY  // Encoder sets the feedrate multiplier on the Status Screen
  #endif
#endif`

1979mgb, 2020-01-09:
Will try this tomorrow, thanks

1979mgb, 2020-01-09:
Vertabreak, configuration.h has oneline
//#define ULTIPANEL 
I am not a writer of code, am learning by watching videos. Do I add what you provided after uncommenting # define ULTIPANEL ? Not sure where,how to add in what you have written.

Vertabreak, 2020-01-09:
the config for ultipanel is in config adv search it

1979mgb, 2020-01-09:
So if I enable ULTIPANEL or EXTENSIBLE_UI in config.h it uses the lines in config.adv and these are the ones I can edit? 
I was thinking it was something to add.
Hopefully I am getting it.

Vertabreak, 2020-01-09:
yes that is what the following line means.
#if EITHER(ULTIPANEL, EXTENSIBLE_UI)
if either of these features are enabled then enable everything inside this if statement until it his endif.

1979mgb, 2020-01-09:
You are a good and patient teacher. Now will have to play with editing the numbers and see what happens.

Vertabreak, 2020-01-09:
i listed what i use above for speeds.
`#define MANUAL_FEEDRATE { HOMING_FEEDRATE_XY, HOMING_FEEDRATE_XY, HOMING_FEEDRATE_Z, 120 } // Feedrates for manual moves along X, Y, Z, E from panel`

1979mgb, 2020-01-09:
Enough for tonight will try tomorrow thank you so much. looking forward to see what this does.

Vertabreak, 2020-01-09:
k ill go back to being a lurking dev lol.

1979mgb, 2020-01-09:
Thank you to all the contributors.

1979mgb, 2020-01-09:
Ok I now have a better understanding of the function of a 2 in 1 out hot end. I can control both extruders from the display motion menu when Printer is configured as Cyclops with 2 extruders 1 nozzle no mixing.
When configured as a mixing Hot end 1 extruder 2 mixing steppers I can not control the extruders individually from display motion menu.

Even so the printer knows it has 2 extruders as they can be individually controlled via change filament menu.
So it would seem that Marlin could be written to allow individual control of the extruders from the motion control menu even if it configured as a mixing printer. 
Anyone willing to give a try?

Vertabreak, 2020-01-09:
You switch mix factor to control additional extruders with mix. Default 100/0 switch to 0/100 to change the motor being controlled.

1979mgb, 2020-01-09:
So if I leave my printer in mixing configuration. I now remove both filaments to clean out nozzle. I now want to check extruder calibration for what ever reason. From the mixing menu I switch from 100/0 to 0/100 Now I can go to motion menu to extruder and the other extruder is now controlled?

Vertabreak, 2020-01-09:
Bingo it's 1 virtual extruder mix factor sets whatnot controls if you Wana empty both set 50/50 and half of the move goes to each.

Vertabreak, 2020-01-09:
If you defined vtools with set mix ratios you can call that vtool with it's set mix factor.

1979mgb, 2020-01-09:
Vertabreak, I ran off to my printer to test this, now I get it. Thank you for your patient teaching once again. The printer does do what I want it to. As you say I just did not understand the relationship.
Not trying to be difficult, but it would seem to be easier from a newbie user aspect if all the motion functions were in one place. I had been blending colors back and forth with mixing for some time without realizing that it had a direct effect on motion menu. That was why I kept using Pronterface.
Thanks again for all the help. Keep up the good work.

Vertabreak, 2020-01-09:
marlin is a patch work of code some times there is no intuitive controls they where never designed with anything else in mind.

Vertabreak, 2020-01-14:
@boelle I think thing one is solved I explained it seems like it's solved.

closed 2020-01-18 by boelle