MarlinFirmware/Marlin #16479: BUG lcd reprapdiscount smart controller with fysetc f6 card

boubouexpress, 2020-01-06:
hello

I had to add the addresses in pin_fystec_f6_13.h to be able to compile

//
// LCDs and Controllers
//
    #define  LCD_PINS_RS      16
    #define  LCD_PINS_ENABLE  17
    #define  LCD_PINS_D4      23
    #define  LCD_PINS_D5      25
    #define  LCD_PINS_D6      27
    #define  LCD_PINS_D7      29
    #define  BEEPER_PIN       37
    #define  BTN_EN1          31
    #define  BTN_EN2          33
    #define  BTN_ENC          35
    #define  SDSS             53 
    #define  SD_DETECT_PIN    49

ellensp, 2020-01-06:
Those pins are all defined in https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/pins/ramps/pins_FYSETC_F6_13.h
Where are your configuration files?

boelle, 2020-01-06:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

boubouexpress, 2020-01-07:
there must have been an update since i downloaded my marlin

closed 2020-01-07 by boubouexpress