MarlinFirmware/Marlin #16481: [BUG] Carriage grinding into endstop on cr10s example config

Marlor, 2020-01-06:
I did an Upgrade from Marlin 1.1.9 to 2.0.1
but the carriage was grinding into the x endstop if i move x to zero, or 3 point probe from UBL.

After a bit of comparing i found this commit:

commit 1d366db764ce2df11b6cac1c1916c6f58bdd5f26

#define MANUAL_X_HOME_POS 4
#define MANUAL_Y_HOME_POS 2

Do i understand this correctly that the touching of the endstop button is set to 4 so 0 would be inside the endstop?

After reverting to the following the printer is acting as expected

#define X_BED_SIZE 300
#define Y_BED_SIZE 300

#define X_MAX_POS 315 // 15mm Extra space on the right end
#define Y_MAX_POS 304 // 4mm Extra space behind the bed

//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0

boelle, 2020-01-07:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Marlor, 2020-01-07:
It is happening in the unchanged example config for Marlin 2.0.1.
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/config/examples/Creality/CR-10S/Configuration.h
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/config/examples/Creality/CR-10S/Configuration_adv.h

boelle, 2020-01-08:
does it also happen in bugfix 2.0.x ?

Marlor, 2020-01-09:
Yes
MANUAL_X/Y is still set and working as intended, it just shouldn't be set.

boelle, 2020-01-10:
so everything is working as intended? or is the example config wrong?

Marlor, 2020-01-12:
Wrong example config, sorry i dont know how to make an pull request.

#define MANUAL_X_HOME_POS 4
#define MANUAL_Y_HOME_POS 2
should be:
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
Dont set 0 inside the endstops

#define X_BED_SIZE 296 // Nozzle is at X4 when homed
#define Y_BED_SIZE 298 // Nozzle is at Y2 when homed
should be:
#define X_BED_SIZE 300
#define Y_BED_SIZE 300
Nozzle is now at 0 homed

#define X_MAX_POS 319 // 15mm Extra space on the right end
#define Y_MAX_POS 306 // 4mm Extra space behind the bed
should be:
#define X_MAX_POS 315 // 15mm Extra space on the right end
#define Y_MAX_POS 304 // 4mm Extra space behind the bed
now that X4 and Y2 are gone

boelle, 2020-01-12:
@thinkyhead could you fix this one?

boelle, 2020-01-28:
btw retest with bugfix 2.0.x: https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

and example configs have moved to their own repro: https://github.com/MarlinFirmware/Configurations

boelle, 2020-01-28:
2.0.1 and even 2.0.2 are older than 2.0.x

boelle, 2020-01-28:
> #define MANUAL_X_HOME_POS 4
> #define MANUAL_Y_HOME_POS 2

endstop are placed at 4,2 so inside the bed, if you want to change that you have to move the endstops away from the bed center first and then change the config

> #define X_BED_SIZE 300
> #define Y_BED_SIZE 300

that is not what the config says

#define X_BED_SIZE 296 // Nozzle is at X4 when homed
#define Y_BED_SIZE 298 // Nozzle is at Y2 when homed

> should be:
> #define X_BED_SIZE 300
> #define Y_BED_SIZE 300

no it should not as the endstops are placed inside  the bed area.... you need to change the hardware first

boelle, 2020-01-28:
will close this one as its clearly as it should be

closed 2020-01-28 by boelle

Marlor, 2020-01-28:
> endstop are placed at 4,2 so inside the bed, if you want to change that you have to move the endstops away from the bed center first and then change the config

If the endstops are placed at 4,2 inside the bed, then 0 is inside the endstops.
Which leads to the carriage grinding into endstop if you move to 0

Simple test:
set
#define MANUAL_X_HOME_POS 10
on any printer.
After homing were is 0?
10 inside the bed or 10 inside the endstop?

boelle, 2020-01-28:
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 319 // 15mm Extra space on the right end
#define Y_MAX_POS 306 // 4mm Extra space behind the bed
#define Z_MAX_POS 400

that is what the config says... so when you do a G29 it thinks its at 0, but in fact its at 4, so you loose at least 4mm on x

its a 300x300mm bed but X max is defined as 319 with 15 mm extra
so 319-15-4=0 so Everything for X is correct, where is the nozzle when you home X (G28 X)? at edge of bed?

same for y, 306-4-2=0, is the nozzle at edge of bed when Y homed (G28 Y)?

https://marlinfw.org/docs/configuration/configuration.html

Manual Home Position
//#define MANUAL_X_HOME_POS 0
//#define MANUAL_Y_HOME_POS 0
//#define MANUAL_Z_HOME_POS 0 // Distance from nozzle to printbed after homing

These settings are used to override the home position. Leave them undefined for automatic settings. For DELTA Z home must be set to the top-most position.

now why would you want to override?

but those 3 would be the only ones wrong if they are

@Vertabreak @thinkyhead @shitcreek any thoughts here?