MarlinFirmware/Marlin #16482: [FR] (feature request title)

Wdfitzpat007, 2020-01-06:
### Description

The problem with setting acceleration values globally with respect to RPM is that stepper motors to not work this way. As RPM increases stepper motor torque decreases exponentially.

It is my opinion that this is one of the main causes of the ringing you see when setting higher acceleration values. Furthermore, lets look at the other main causes: Insufficient machine rigidity, backlash, and unknown nozzle pressure. Also lets bring up another common print issue: lost steps. Lost steps occur when load surpasses motor capacity.

Doing this will lower the load placed on the steppers when at their weakest, but without limiting it when at maximum strength. Now lets see how setting a dynamic acceleration corresponding to RPM impacts each of the aforementioned issues.

-Insufficient machine rigidity 'slop'. Cheap Chinese POM rollers have more slop when you push on them and apply higher than usual load, right? If we lower the load on the motors when moving fast, a side effect will be lowering the load on the linear motion system. I have seen no empirical studies demonstrating this and therefore cannot unequivocally say that this will positively impact print quality when running fast. Lets not kid ourselves here either. Another consideration is the timing belt, when moving faster timing belts have a lower load capacity. I know this because I work engineering linear motion systems and have proven several of my designs inadequate after the fact because I failed to check the load rating at a given speed of the belt width and pitch I was intending to use at the length I was planning on using it at.

-Backlash. The reason you do not see widespread issued with backlash in threaded rods is because the downwards load never exceeds the load of gravity placed on the assembly. This is also why no-one EVER needs to use anti-backlash nuts, ballscrew or threaded rod. In fact, depending on the kind of anti-backlash screw and the strength of the spring, you could decrease the load capacity of the system without incurring appreciable backlash. For example, if you had one of the oil impregnated bronze anti-backlash nuts with the spring side facing down with gravity causing the spring to hold the mounting side to the top profile of the threaded rod against gravity. In that use case if the screw is able to exert less than twice the force of the gravity load without any appreciable deflection you will increase the backlash of the system. The spring end would have to facing 'up' pressed against gravity with the mounting side facing down the spring pressing with gravity. Ballscrew is only better than leadscrew if it has a higher accuracy over distance.

-Backlash part 2 belts. To address confusion on belt reinforce type, the specific modulus of Kevlar 149 is 127 GPa/ (gcm^3), S-glass (aka S-2) fiber 35GPa/ (gcm^3), 1045 steel 26GPa/ (gcm^3). Values used kevlar 186/1.47, glass 89/2.53, steel 205/7.87. This means that for the same level of stiffness a kevlar belt will have 10% higher cross sectional area but only weigh 21% of what the steel belt weighs. Thus making the steel belt 5 times heavier for no reason. Furthermore when using steel the strand size is far greater than kevlar resulting in the belts being far stiffer chewing them when going around small pulleys. Look up "Steel Core Polyurethane belt used with a 16 tooth pulley", there are pictures, real nasty.

Deflection in the belt is not where the error would come from if there were belt related error, the error would come from not using anti-backlash sprocket. GT 3mm sprocket for example have 114 micron worth of clearance. GT 2mm will have less clearance, but I do not have a solid number off hand. Friction, belt pitch, belt pre-load, resulting belt compression (squish into the tooth clearance), and tooth engagement (a function of belt pitch and width) have a role in how much the tooth clearance effects the overall slop of the machine. So it is not as simple as saying that you are limited to 114 micron of accuracy with GT 3mm. Something more to the effect of you have within half the tooth clearance of the drive in each direction. The less load placed on the belt, the less the teeth will wiggle in the drive grooves.

-Calmer decelerators will give the nozzle pressure more time to come down before stopping. Making the plastic flow more predictable.

-Lowering the load when the motors are weaker lowers the load. This is self explanatory it would take more force to cause a lost step without using a larger stepper or unnecessarily increasing the print time.

Making the acceleration dynamic would have many different advantages. Some in-substantial, others very meaningful.

### Feature Workflow

-Calculate RPM
-Estimate acceleration biased on change in RPM and step/mm
-apply a weight to the allowable acceleration value biased on current RPM

-Allow users to change the weight biased on their motors.

Unfortunately programming is not my trade of choice, far from my favorite kind of work and I have no free time at this point in my life to pursue this. However I will toss this idea out there in case someone decides to work on it before future me finds time too.

DerAndere1, 2020-01-07:
Can you propose a function or a diagram that shows how you think acceleration should be calculated from current RPM? Should it be different from S_CURVE_ACCELERATION?

Wdfitzpat007, 2020-01-08:
Identical to S_CURVE_ACCELERATION, simply giving the user the ability to input the desired curve. Unless that is something else I do not know about.

DerAndere1, 2020-01-08:
The bezier curve used for S_CURVE_ACCELERATION is defined from https://github.com/MarlinFirmware/Marlin/blob/a88b91b3ed9eab7ec89045f93031faeb1823e1d4/Marlin/src/module/stepper.cpp#L455 onwards. (more specifically lines  690-1190) You can try to change the constant factors in the calculation of the bezier coefficients, but I guess that probably means you have to change those 500 lines of optimized assembler. Seeing how much work went into that optimized code makes me think that it would be extremely difficult do let the user choose bezier coefficients freely in the config files.

MarkMan0, 2020-01-09:
I guess a linear decrease of acceleration with speed would be an improvement, but finding the right parameters will be hard, and possibly vary from stepper to stepper.
Do they lose torque exponentially in the whole range of RPM? I think the motors only start losing torque after a certain RPM, where the induced voltage is greater than the voltage needed to pass a set current through the windings.

AnHardt, 2020-01-09:
https://www.designworldonline.com/improving-high-speed-performance-of-hybrid-stepper-motors/

or
https://www.google.com/search?sxsrf=ACYBGNSp7_LiPWolh3T9F4IJJTfHoNPSug:1578594455262&q=stepper+motor+torque+rpm&tbm=isch&source=univ&sa=X&ved=2ahUKEwjy9__NkvfmAhVHQEEAHcMIDP8QsAR6BAgIEAE#imgdii=IK_fWZqtMH5UJM:&imgrc=PtyztIeMxYfOYM:

DerAndere1, 2020-01-09:
I see now that this is independent of S_CURVE_ACCELERATION. In the context of enabled S_CURVE_ACCELERATION, this proposal means that the **nominal** acceleration would have to change dynamially with rpm. 

Some thoughts: Lets assume that defined positioning is a minimum requirement, so global acceleration has to be low enough to not loose steps at max feedrate. So I see only one (never the less important) advantage of the proposed feature: You can increase acceleration at low rpm and thereby reduce print duration. 

Regarding 

> It is my opinion that this is one of the main causes of the ringing 

I guess with the proposed feature it will stay the same trade-off between print duration and ringing.

AnHardt, 2020-06-27:
There are a few reasons for closing a FR:
The feature is already there.
The feature can not work for physical or logical reasons.
The feature was replaced by some other functionality so it's not needed anymore.
...

Closing a Feature Request just because currently no one is working on it, is **NOT** a valid reason. It will be a valid request until someone says: This one is resolved or declined.

Automatically closing FRs is **not** a job for a bot.
--------------------
Some on topic thoughts:
Steppers can't run infinitely fast. 
They lose torque with increasing RPM (proportional steprate).
What ever the shape of the curve is, it will end at zero torque when the back EMVis equal to the voltage we can provide.
The force we need has two components.
One, more or less, 'static' component, the force needed to keep the system in movement. The force to overcome the resistances at that speed.
And a 'dynamic' one for accelerating/breaking the speed, resulting from F=m*a.

That means if the naked motor is able to rotate at, let's say, 100 RPM, then the complete system may be able to maintain 95 RPM, once it would be at that speed. That also means that no additional force is left over for the 'dynamic' part. Acceleration has to be zero at that speed. Even closely below that speed we have only a little bit force to accelerate , compared to the full force we have available at zero speed (RPM).

That means maximum speed and maximum acceleration are not independent from each other. To reach maximum speed the acceleration must be minimal. And for a given acceleration we have a lowered maximum speed.

What we currently do when trying to maximize speed fo one axis, is selecting an acceleration value and testing how far we can go up with the speed, or we select a speed and test how far we can go up with the acceleration. (All within the limits of the maximum path/axis length.)

When 'normally' printing, with our synchronized moves, with a very high feedrate (F), normally only one axis is at its limits. With low feedrate (F) it may be, non of the axes is at its speed limit, but always one axis determines the common acceleration.

Knowing about that it may be useful to limit the test speed for an axis to the maximum feedrate we want to use and rising the acceleration instead.

Tuning acceleration for high speeds means having lots of extra power at low speeds and that is the requirement for 'classic jerk speeds'.

Even with perfectly stiff belts we will have 'ringing'. A stepper motor is able to oscillate in a range of 2 full steps. The force/displacement graph follows roughly a sin-function, crossing zero at zero displacement, with the force at its maximum at +- one (full) step displacement, then shrinking down to zero at +- two steps displacement. At higher speeds the maximum force is much lower (down to zero) but the next maximum is still +-1 step away. That is unavoidable. No displacement - no force. I'm quite sure all ringing we see is within this amplitude.

Jerk (speed) has to be small enough to not exceed that two full step limit. When breaking there is already displacement in one direction (else there would not be a breaking force) and acceleration into the opposite direction causes additional displacement into the same direction.

'S-curve acceleration' not only flattens the curve at the top speeds, helping to reach higher speeds, but also helps with low accelerations right after the jerk at the lower end.

'Junction deviation' is just a way to determine jerk (speed) dynamically depending on the junctions angle.

A 'dynamic' **endspeed** dependent acceleration calculation would have to influence the acceleration value, even that feed into the S-curve-acc-calculation. A calculation based on other possible speeds is currently impossible in Marlin - mainly because not known until directly before the block is used - far behind planning based on the acc.

thinkyhead, 2020-07-17:
Your comment prevents it being closed automatically for another 65 days, which comes with a 5 day warning.

thinkyhead, 2020-07-17:
Your comment prevents it being closed automatically for another 65 days, which comes with a 5 day warning.
Since 20 days ago, when I was first working on the bot configuration, FR's have been added as an exemption.
I figured it out all on my own.