MarlinFirmware/Marlin #16484: Add BigTreeTech SKR 1.4 Turbo

Docomar750, 2020-01-06:
Hi 
Can the BigTreeTech SKR 1.4 Turbo be added to Marlin it uses the same pins as a standard 1.4 board but uses a LPC1769 chip (120mhz)

Cheers

Darren

tatusah, 2020-01-06:
Already there, set environment as LPC1769 in platformio.ini

thisiskeithb, 2020-01-06:
This is a dupe of https://github.com/MarlinFirmware/Marlin/issues/16324, but as noted above this board has already been added.

boelle, 2020-01-06:
board has been added and this is a dube of #16324, closing

closed 2020-01-06 by boelle

stantond, 2020-02-05:
I've just had the same problem, though rather than "add the SKR 1.4 Turbo board", it's "make it clear that the SKR 1.4 Turbo board is supported". It's listed as a MOTHERBOARD option, but it would be really helpful to have SKR 1.3/1.4/1.4 Turbo added to the platform.ini file as comments above their appropriate environment options. Currently, searching for SKR in platform.ini shows several other BTT boards, but the boards I've listed aren't listed, so it does give the impression that, by their absence, they're not fully supported. Although I now know this isn't the case, it would save other users moving to these boards some time.