MarlinFirmware/Marlin #16486: M115 misses firmware version

SVLoneStar, 2020-01-06:
### Bug Description

According to https://docs.octoprint.org/en/master/plugins/hooks.html#firmware_info_hook, Octoprint reads the firmware's version from the output of M115 which is expected to be formatted as
`FIRMWARE_NAME:Some Firmware Name FIRMWARE_VERSION:1.2.3 PROTOCOL_VERSION:1.0`

However, current Marlin version(s) report to M115 like this:

`FIRMWARE_NAME:Marlin bugfix-2.0.x (GitHub) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:11111111-2222-3333-4444-555555555555`

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4028093/Marlin.zip)
OctoPrint 1.4rc3, Wanhao Duplicator i3, bugfix-2.0.x as of January 3rd, 2020, SKR 1.3.

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Flash current marlin version
2. Send an M115
3. Look at response...no version included

**Expected behavior:** [What you expect to happen]
Version substring included in M115 response

**Actual behavior:** [What actually happens]
No version information included in M115 response

#### Additional Information

I posted this issue as well in OctoPrint's forum, knowing that it's probably more related to Marlin. Is this because bugfix-2.0.x is a beta and does therefore not have a version number? Might it be possible to tweak this in Marlin?

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-01-06:
I would think that OctoPrint should be updated to account for both formats.

SVLoneStar, 2020-01-06:
While I of course fully agree to this statement, the OctoPrint documentation states at [https://docs.octoprint.org/en/master/plugins/hooks.html#firmware_info_hook](url) that it populates an internal structure (including the firmware version) by using the information provided through M115. So...what other G Code could be used to retrieve the version of a Marlin bugfix-2.0.x installation? Thanks a lot!!

ellensp, 2020-01-07:
this string is defined in language.h  MSG_M115_REPORT. But looking through 1.0.x, 1.9.x and 2.0.x none of them have the string FIRMWARE_VERSION...   perhaps this was pre 1.0,0?

ellensp, 2020-01-07:
But it is easy to add it... just edit the string MSG_M115_REPORT in  language.h. Eg add  FIRMWARE_VERSION:2.0.x   Give it a try see if octoprint picks it up.

SVLoneStar, 2020-01-07:
Thanks! I‘ll try and report back.

boelle, 2020-01-08:
@SVLoneStar when do you have time to report back? just so we know when we can expect that

SVLoneStar, 2020-01-08:
@boelle Hi - up to now I have been playing around with the code of the Octoprint plugin in question (Marlin EEPROM editor) to cope with what's returned by Marlin...I think that it's more meaningful to just deal with what's Marlin reporting in its current version instead of asking Marlin developers to change their code just because of issues with one specific Octoprit plugin.
Just to make sure, I'll as well try to modify 'version.h' (not language.h as stated above) to make sure Ocroptint picks up the version string coded there...but as said, that's from my point of view not the best way to come around the issue I have reported.

Therefore, I'll close this issue. Thanks for all your replies to my post!

closed 2020-01-08 by SVLoneStar