MarlinFirmware/Marlin #16487: [BUG] (short description)

diver197, 2020-01-07:
### Bug Description
Linking everything together...
"C:\\Users\\15205\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-gcc" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -Os --specs=nano.specs -Wl,--defsym=LD_FLASH_OFFSET=0 -Wl,--defsym=LD_MAX_SIZE=524288 -Wl,--defsym=LD_MAX_DATA_SIZE=196608 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common "-TC:\\Users\\15205\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.8.0\\variants\\PRNTR_Vx/ldscript.ld" "-Wl,-Map,d:\\Arduinotmp/Marlin.ino.map" "-LC:\\Users\\15205\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/DSP/Lib/GCC/" -larm_cortexM4lf_math -o "d:\\Arduinotmp/Marlin.ino.elf" "-Ld:\\Arduinotmp" -Wl,--start-group "sketch\\Marlin.ino.cpp.o" "sketch\\SrcWrapper.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\ctrl_access.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\sysclk.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_cdc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_cdc_desc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_composite_desc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\udi_msc.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\uotghs_device_due.c.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\usb_task.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\include\\digipot_mcp4451_I2C_routines.c.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\LCD_pin_routines.c.o" "sketch\\src\\HAL\\HAL_AVR\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\fast_pwm.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\fastio.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\u8g_com_HAL_AVR_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_AVR\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\DebugMonitor.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\EepromEmulation.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\InterruptVectors.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\MarlinSerialUSB.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\Tone.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_shared_hw_spi.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_st7920_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\dogm\\u8g_com_HAL_DUE_sw_spi_shared.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\fastio\\G2_PWM.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\timers.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\usb\\sd_mmc_spi_mem.cpp.o" "sketch\\src\\HAL\\HAL_DUE\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\FlushableHardwareSerial.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\WebSocketSerial.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\i2s.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\ota.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\spiffs.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\timers.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\web.cpp.o" "sketch\\src\\HAL\\HAL_ESP32\\wifi.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\arduino.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Clock.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Gpio.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Heater.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\IOLoggerCSV.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\LinearAxis.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\hardware\\Timer.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\include\\pinmapping.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\main.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\timers.cpp.o" "sketch\\src\\HAL\\HAL_LINUX\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\DebugMonitor.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\MarlinSerial.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\fast_pwm.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\main.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\persistent_store_flash.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\persistent_store_sdcard.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\timers.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\LCD_I2C_routines.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_hw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\u8g\\u8g_com_HAL_LPC1768_sw_spi.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\usb_serial.cpp.o" "sketch\\src\\HAL\\HAL_LPC1768\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\MarlinSerial_AGCM4.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\timers.cpp.o" "sketch\\src\\HAL\\HAL_SAMD51\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\SPI.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\SoftwareSerial.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\dogm\\u8g_com_stm32duino_fsmc.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\dogm\\u8g_com_stm32duino_swspi.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\msc_sd.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\onboard_sd.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\persistent_store_flash.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\persistent_store_sdcard.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\sdio.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\timers.cpp.o" "sketch\\src\\HAL\\HAL_STM32F1\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\SoftwareSerial.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\fastio.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\persistent_store_flash.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\persistent_store_sdcard.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\timers.cpp.o" "sketch\\src\\HAL\\HAL_STM32\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\EmulatedEeprom.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\STM32F4\\timers.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\STM32F7\\TMC2660.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\STM32F7\\timers.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\eeprom_emul.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_STM32_F4_F7\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\persistent_store_impl.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\timers.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY31_32\\watchdog.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\HAL.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\HAL_SPI.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\Servo.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\persistent_store_eeprom.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\timers.cpp.o" "sketch\\src\\HAL\\HAL_TEENSY35_36\\watchdog.cpp.o" "sketch\\src\\HAL\\shared\\HAL_spi_L6470.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\backtrace.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm_arm.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarm_thumb.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarmbytab.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwarmmem.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwinder.cpp.o" "sketch\\src\\HAL\\shared\\backtrace\\unwmemaccess.cpp.o" "sketch\\src\\HAL\\shared\\eeprom_i2c.cpp.o" "sketch\\src\\HAL\\shared\\eeprom_spi.cpp.o" "sketch\ [... cut ...]
#define MOTHERBOARD BOARD_BLACK_STM32F407VE
#define FSMC_GRAPHICAL_TFT

No other modification
arduino-PR-beta1.9-BUILD-116
STM32 Cores 1.8
build.path=d:\Arduinotmp
### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-01-07:
This is a limitation in windows. It cannot handle the long (>32k characters) command lines required. 
The only option I have found is to remove the HAL's that are not being used by your hardware. 
Ie remove unused directories from Marlin/src/HAL/  Do not remove the two .h files for your controller you need to keep "shared", "HAL_STM32_F4_F7" and  "HAL_STM32"

ManuelMcLure, 2020-01-07:
That’s bad design on the part of PlatformIO. There should be a way to create library archives and link those instead of listing every single .o file on the linker command line.

diver197, 2020-01-07:
After compiling in platformio, I can't run after downloading

ellensp, 2020-01-07:
Did you set #define SERIAL_PORT -1 so it uses the usb port to communicate?

diver197, 2020-01-07:
#define SERIAL_PORT 1

diver197, 2020-01-07:
After removing the redundant HAL, it's still too long. In Linux, what's the length

ellensp, 2020-01-07:
@ManuelMcLure affects windows, so breaks arduino ide and platfomio.   Previously arduino had this issue but different cause. It used full plaths for everything,  now it uses relative paths and and marlin is much larger and hits this limit again....

ellensp, 2020-01-07:
Even on Linux this combination does not compile...   but for other reasons 
No one has setup a BOARD_BLACK_STM32F407VE with a FSMC_GRAPHICAL_TFT  (yet)
various pin definitions are missing, and even after adding them I couldn't get this to compile.

diver197, 2020-01-07:
It is possible on MKS - Robin -mini, but platformIO compiled downloads cannot run in STM32, so only Arduino ide can be used

ellensp, 2020-01-07:
MKS - Robin -mini, is a STM32F1, BOARD_BLACK_STM32F407VE is clearly a STM32F4 ....

boelle, 2020-01-07:
so this is not a marlin issue?

ellensp, 2020-01-07:
This should probably be closed as its a windows/platformio issue. But a new Feature request  should be opened to support LCD FSMC_GRAPHICAL_TFT on motherboard BOARD_BLACK_STM32F407VE

ManuelMcLure, 2020-01-07:
> @ManuelMcLure affects windows, so breaks arduino ide and platfomio. Previously arduino had this issue but different cause. It used full plaths for everything, now it uses relative paths and and marlin is much larger and hits this limit again....

It is however less prevalent on PlatformIO for a couple of reasons - the external libraries (such as u8glib) are packaged into .a files before the link, and PlatformIO creates a temporary file that lists all of the .o files and passes the name of that file to the linker instead of creating a huge command line.

boelle, 2020-01-07:
@ellensp would it be best if @diver197 creates that FR ?

closed 2020-01-07 by boelle

asantoni, 2020-02-11:
This is still a problem on Windows - I can't compile Marlin with the Maylan M200 config and instructions  due to this error. 

Is there a clear avenue to compile it on Windows? Will PlatformIO work or not? There's no clear answer in this thread or any others.

Or have we just given up with compiling Marlin on Windows?

ellensp, 2020-02-11:
This is closed because it is a windows limitation.  Windows has a 32k character limit on commands. The link command lists all compiled files and libraries together with relative paths. This can easily exceed 32k characters.  The only work around is to remove unused files from the link process. ie. delete unused HAL directories and if your not using it extensible_ui.

asantoni, 2020-02-11:
I fully understand the reason for the error, thanks. What I don't understand is Marlin's policy on this issue, and why this bug is closed. The problem isn't resolved at all.

If there's a policy change that Marlin can't reliably be build with the Arduino IDE on Windows, then us users would expect the website and documentation to be updated, instead of just sweeping major problems like this under the rug.

thinkyhead, 2020-02-11:
> I fully understand the reason for the error, thanks. What I don't understand is Marlin's policy on this issue, and why this bug is closed. The problem isn't resolved at all.

We are not able to tell Arduino IDE which are unused in the targeted build. To do that we would need to re-package the HAL portions of Marlin as individual libraries, but since the HALs are tightly integrated with Marlin's configurations this is not an option. When and if we do reorganize the HALs into a library structure, Arduino IDE will only build the ones that pertain to the target architecture.

It may be a year or more before someone comes along with the time and talent to accomplish this.

So our only current solutions are:
- Use PlatformIO, which _does_ screen out the unused folders, or
- Remove the unused HAL sub-folders before building with Arduino IDE.

asantoni, 2020-02-11:
Thanks Scott, I appreciate the response. I appreciate that Marlin's in a tough spot on this one. It sounds like Marlin has outgrown the Arduino IDE and perhaps deprecating it should be given some consideration. That Arduino 1.9 has been in beta for over 2 years doesn't instill confidence. 

FWIW the Malyan M200 doesn't currently compile with PlatformIO either. [Some people have figured out how to get the M200 V2 compile to work](https://community.platformio.org/t/marlin-compile-fail-board-board-h-no-such-file-or-directory/11843) over here though.

thinkyhead, 2020-02-11:
Deprecation is a thought. But I'll continue to look for potential tricks, and who knows, maybe Arduino will add support for multiple builds at some point….

Barestos, 2020-05-30:
Hello,

I'm facing the same issu with Arduino 1.8.12 & using Marlin 2.0.5.3 since I want to set My LCD and uncomment "#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER";
First I encounter "C:\Users\bzngi\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc.exe: The filename or extension is too long." but what's ellensp said doesn't work for me...(I mean suppressing HAL's unused directories). 
I also tried to shorten the directory of avr files by deleting sub-folder but when I do this I encounter a second issue : "C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2/bin/avr-g++": file does not exist" so i digged up a bit and find a solution: Board Manager -> Install "Arduino AVR Boards (1.8.1)" -> Restart Arduino IDE -> Install Arduino AVR Boards 1.8.2.  However when i restart IDE and compile Marlin sketch I end up with the first issue filename to long...
I keep looking for a solution without succeed, If anyone have an idea i'll enjoy it!

Thanks

ellensp, 2020-05-30:
USE  PLATFORMIO  Arduino IDE is rubbish for larger projects.
Things to try
1) move marlin to root of a Drive and rename it M
2) delete all unneeded HAL directories
3) remove unused lcd/extui directory 
4) if using a 1284p based board use melzi not atmega1284p (shorter) 
5) write to Microsoft and complain bitterly at their pathetic path length limitations.

EvilGremlin, 2020-05-30:
6. move to Linux and never experience such stupid stuff again, along with half of Cura bugs.
But first is mandatory if you building it under windows.

Barestos, 2020-05-30:
Thank you for the replies,

I've dl PLATFORMIO and tried to compile marlin's program unsuccessfully, with these modification:
I've switched Marlin root to "c:/Temp/Marlin-2.0.x"
I've removed some HAL directories but some remains and since i'm new into this i'm not sure of what to remove. Some help would be appreciated ;) (remains 
[image]
As for the unused lcd/extui directory i'm not sure of what is unused..
Concerning the board it's a Mega 2560.

Now i'm trying to compile on PlatFormIO but i encounter this issue once again : 
avr-g++: error: CreateProcess: No such file or directory
*** [.pio\build\mega2560\firmware.elf] Error 1

And the report point those 2 problems:
"c:/Temp/Marlin-2.0.x/include"  X2

EvilGremlin, 2020-05-30:
> move marlin to root of a Drive and rename it M

ellensp, 2020-05-30:
@Barestos  You haven't told us what motherboard your using...
So I am going to guess something atmega2560 or atmega1284p based.
So you can remove 
SAMD51
STM32
STM32_F4_F7
STM32F1
TEENSY31_32
TEENSY35_36

"And the report point those 2 problems: "c:/Temp/Marlin-2.0.x/include" X2"
These two problems can be ignored. PlatformIO is just wating a include directory that doesn't exists (or need to be there)

Barestos, 2020-05-30:
Thanks ellensp,

I deleted these HAL but I keep having the avr-g++: error.
EvilGremlin Thanks but I don't know exactly how to do it...

I'll dig on the net for this issue, maybe crossing solutions tgther will help me undertsand the thing i'm missing...

EvilGremlin, 2020-05-30:
Exactly the same as you moved it to Temp, just to drive root. And rename folder to one symbol.

Barestos, 2020-05-30:
Thank you EvilGremlin.
I've done this but i still encounter avr-g++ error.
I keep looking but i'm strating to despair, maybe time to go on Linux!

Thank you anyway. If someone has a solution it will be appreciated.

diver197, 2020-05-31:
Now I use  PlatformIO to compile. Now it's perfect, but it's difficult to build  PlatformIO environment. Install Python before other  PlatformIO installations

Barestos, 2020-05-31:
Hi there,

I succeed this morning repeeting the steps indicated above, but désinstalling and réinstalling everything before. 
Thank you guys.