MarlinFirmware/Marlin #16489: [BUG] UART dataloss when serial port 2 is enabled

Xyp, 2020-01-07:
As soon as the serialport 2 is enabled, dataloss is seen on serial port 1. 
The dataloss can be noticed by resend requests from Marlin sent to the host. Data gets only losts in pc to Marlin direction, but on virtual cdc usb ports, as well as native uart. Data is also lost, when serial port 2 is not connected - only enabling it in the firmware makes it fail. 

This occurs on current bugfix 2.0.x tree. 

Only the first few bytes are lost every now and then, sometimes ending up in infinite resend loops. 

Tested on bigtree skr mini e3 with stm32f103 STM controller. 
The issue occurs not only when printing, but also in SD card upload.

Analyzed, if the sending pc is the cause by sniffing the uart traffic with a logic analyzer - there is no issue on pc side.

Current prooven in use solution: disable serial port 2 in configuration.h

Xyp, 2020-01-07:
Adding: another effect is, that printing stops for a few seconds at random points due to the resends. 
This aspect is also solved by disabling serial port 2.

boelle, 2020-01-07:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Xyp, 2020-01-07:
Will attach later this evening including serial sniffed data.

Xyp, 2020-01-07:
Btw: I see, that the ring buffers are declared static in MarlinSerial class. Would that not mean, that there is just a single instance of buffers, shared among several instances of MarlinSerial class?

This would be my understanding with good C background, but limited C++ knowledge (especially when it comes to templates)

boelle, 2020-01-07:
closing until the required info stated in the issue template is attached (configs)

closed 2020-01-07 by boelle

Xyp, 2020-01-07:
You managed to close 3 minutes before I wanted to share the details :-)
I doubt the issue are in the configuration files. Check my comment above on the static declaration. Will debug in that direction further.

Attached the working configuration.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4032040/Marlin.zip)

The diff to the non working one is, that the non working one has line #115 uncommented.

Xyp, 2020-01-15:
The info is there, don't want to reopen?