MarlinFirmware/Marlin #16490: [BUG] PID_EXTRUSION_SCALING : Weird and dangerous results with default values !

Balzac40, 2020-01-07:
Printer : FLSUN Prusa I3+, with a lot of customizations : SKR V1.3, TMC 2130 V3.0 SPI mode, TFT35 V3.0 and also Silicone 220V 750 W heated bed.  

### Bug Description
After compilation Marlin 2.0.1 with this option :
#define PID_EXTRUSION_SCALING
  #if ENABLED(PID_EXTRUSION_SCALING)
    #define DEFAULT_Kc (100) //heating power=Kc*(e_speed)
    #define LPQ_MAX_LEN 50
  #endif

during a consistent printing, after about one hour , this is happening ;

- hotend is no more heated while LCD shows temperature equal to requested one
- of course, filament does not move as hotend is now cold
- heated bed climbs dangerously in temperature, while LCD shows also bed temperature equal to requested one
- moves are not affected...

It seems to me that Marlin tries to compensate hotend by bed ?

Maybe default values are bad, or not present in EEPROM ?

### My Configurations
My printer 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4029892/Marlin.zip)

boelle, 2020-01-08:
does it also happen with bugfix 2.0.x ?

boelle, 2020-01-08:
and remember to use M502 then M500 after flashing

Balzac40, 2020-01-09:
@boelle These days, I have no time to try with bugfix 2.0.x, sorry. I do know that we have to do a M502/M500 cycle after each flashing.
In fact, I've changed my genuine heated bed by a silicone one because my diagnostic was wrong and I have also ordered a new motherboard for the same reason... So, I will wait some time before engaging new tests. Thank you

Vertabreak, 2020-01-09:
strange Ive had this feature enabled on my printers for nearly a year without any issues even on days long prints.

Balzac40, 2020-01-09:
@Vertabreak Please, do you remember your values for M301 C and L ? Other parameters ? Thanks
@boelle What would happen if L parameter is not defined ? Thanks

AnHardt, 2020-01-09:
When the problem sows up next time please have a look at the M105 responses and report them here.

From that we can hopefully see in what part of the code the problem is. If in the calculation of the PWM-values, or in switching the outputs in the right rhythm and frequency based in the PWM-values.

For the 220V 750 W heated bed a slow PWM-frequency and really sending not the shortest puls when zero, is very important.

boelle, 2020-01-09:
> @boelle What would happen if L parameter is not defined ?

i have no clue as i dont have problems, and i'm not a code guy, just the janitor sweeping the floor

Vertabreak, 2020-01-09:
i would need more info to attempt to do anything i have this feature enabled on my fork its been used by my community for about a year no one reported issue like this nor have i seen any.

Balzac40, 2020-01-10:
Hi guys,
Not esay for me to explain correctly all my problem as english is not my native tongue...
Today, this feature is **not** active, and I encounter same things. Good start for my print and one hour later, heated bed is cold, while LCD tells the requested temperatures, tells the same values for actual ones, and tells that hotend is warming... I have stopped !
I will wait my new motherboard and will see what the f.... problem is.

I close this issue ; thank you all !

closed 2020-01-10 by Balzac40

Balzac40, 2020-02-11:
After having replaced my SKR V1.3 by an SKR V1.4 Turbo (with all accesories including power for LCD), everything is fine, and I suppose there was a problem with 12V power. Now, my TFT35 V3.0 screen is brighter and more contrast...
Thank you all again

thinkyhead, 2020-02-11:
We are hearing a lot of worrying things about the SKR 1.3 boards. I am actually removing one from my printer because it is causing the PSU to heat up excessively.