MarlinFirmware/Marlin #16493: [BUG] Marlin 2.0, BOARD_MKS_SGEN creates an error while compiling

GerhardSchroeder, 2020-01-07:
### Bug Description

Have a MKS SGEN 1.0 and try to create a bin file. 
When i use the Board "BOARD_MKS_SGEN", i got the error:
"Marlin\src\HAL\HAL_LPC1768\../../inc/../pins/lpc1769/../lpc1768/pins_MKS_SBASE.h:29:4: error: #error "Oops! Make sure you have the LPC1768 environment selected in your
IDE.""

When i disable the checking in the "pins_MKS_SBASE.h file. Which is included in the pins_MKS_SGEN.h file.

#ifndef MCU_LPC1768
 // #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
#endif

It compiles without any errors. But after uploading to the Board, the Board is not accessible via USB, COM port anymore. 
When i write back the Smoothie firmware, the Board is working again.

### My Configurations

https://www.dropbox.com/sh/k4trz437e9x80uw/AACL9D_LHvTkixpH1_Q2Vi3ha?dl=0

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tatusah, 2020-01-07:
The error notice tries to guide you to check the platformio.ini file setting for the default environment. You likely have something other than LPC1768 set for the `default_envs` if you haven't made any changes to that file.

GerhardSchroeder, 2020-01-07:
Hi, platform settings is default. But i compile with the env:LPC1769 in PlatformIO
BOARD_MKS_SGEN is using a LPC1769 processor.

So far i can see, the problem is in the included pins_MKS_SBASE.h file which has a LPC1768 processor.
Here is an error check
#ifndef MCU_LPC1768
  #error "Oops! Make sure you have the LPC1768 environment selected in your IDE."
#endif
But i need to use LPC1769 for the MKS SGEN

Here is the part from pins_MKS_SGEN.h. In the last line is the include of the pins_MKS_SBASE.h.

/**
 * MKS SGen pin assignments
 */

#ifndef MCU_LPC1769
  #error "Oops! Make sure you have the LPC1769 environment selected in your IDE."
#endif

#define BOARD_INFO_NAME   "MKS SGen"
#define BOARD_WEBSITE_URL "github.com/makerbase-mks/MKS-SGEN"

#include "../lpc1768/pins_MKS_SBASE.h"

boelle, 2020-01-07:
what ide do you use?

tatusah, 2020-01-07:
Ah, sorry mixed up SGEN and SGENL.

tatusah, 2020-01-07:
It's a bit similar situation as with SKR v1.4 and the Turbo variant. Those don't use definition check for the MCU in the pins files as it would lead to similar kind of situation if it wasn't done with extra conditions taking into account the LPC1768 and LPC1769 processors. But that ifndef check aside have you tried setting the `#define SERIAL_PORT -1` it seems to be what is set in the Makerbase github example although it's last updated May 2019. Your example config seemed to have this set to 0.

GerhardSchroeder, 2020-01-07:
@boelle IDE is VS Code
Just comment out the env. checking in the SBASE.h and set the serial port to -1
It compiles without errors, but still no access to the board. Also the board Led's are not blinking like it should be.

tatusah, 2020-01-07:
What firmware did the board come with Marlin or Smoothieware? Do you update the firmware with SD card or through Platformio? I'm wondering if there are some remnants of the other firmware on the board preventing good flash.

ellensp, 2020-01-08:
Yes this is a bug..  pins_MKS_SGEN.h correctly checks for a LPC1769 but then includes ../lpc1768/pins_MKS_SBASE.h which checks for a LPC1768.  generating a PR to fix this

ellensp, 2020-01-08:
This has been fixed in marlin bugfux-2.0.x

closed 2020-01-08 by boelle