MarlinFirmware/Marlin #16494: [BUG] SKR 1.4 + TMC2209 UART => ENDSTOP fail for X, Y and Bltouch

Gerggg, 2020-01-07:
### Bug Description

The end stop of X, Y and Bltouch do not work with the SKR 1.4 card and TMC2209 drivers in UART.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4031514/Marlin.zip)

### Steps to Reproduce

1. compil Marlin bugfix-2.0.x for SKR 1.4 + TMC2209 UART
2. try to make home for X, Y or Z
3. the axis moves a few mm and stops taking the position as 0

Gerggg, 2020-01-07:
I know that by cutting a pin on my drivers the problem should be solved but I don't want to deteriorate my driver. a sofware solution seems more intelligent to me.

randellhodges, 2020-01-07:
This isn't a bug. At most, it is a feature request, but even then, I'm not sure it is physically possible. I'll let the experts chime in on the details.

GMagician, 2020-01-07:
What pin do you have to cut to let it works?

boelle, 2020-01-07:
will close then

what title could @Gerggg use for a FR?

closed 2020-01-07 by boelle

randellhodges, 2020-01-07:
Something like [FR] TMC2209 Support physical endstop switches without removing diag pin.

I would invite @Gerggg to jump in discord, maybe over in the trinamic channel and see if such a thing is even possible. Of course, could just open the FR and see what others say.

GMagician, 2020-01-07:
I think that if it has to cut diag pin, that is used, AFAIK, for sensorless home, only solution he may have is to remap min pins (I think diag is used as min pin now)