MarlinFirmware/Marlin #16496: [FR] Multiple PID settings for more accurate stabilization

italocjs, 2020-01-07:
### Description

The idea is to make multiple PID "autotune routines" for a more accurate stabilization

To explain the problem i found that the pid stabilization gets kinda "off" if the target temperature is too far from the used on the autotune, for example, if i do the autotune using 200C (for PLA) i get a kp ki kd value, and if i do the same autotune for nylon (290C) i get another.   If i have my printer prepared with the PID settings to 290C and try to print PLA at 200, there will be too much overshoot (sometimes even causing the thermal runaway to trigger).   Right now i'm getting away from this by setting the PID values using the slicer/material settings (on prusa slicer its possible to send some "gcodes" depending on the material), but would be a nice thing to have implemented into the firmware

### Feature Workflow

User run the M303 E0 S250 C8 with a aditional parameter "N 50 P 50" for example and the autotuning will run three calibrations, using + ou - 50C from the targeted temperature  effectively having a best setting for 200, 250 and 300C.  then the heater maganer would be able to choose the nearest PID settings to print with.

#### Additional Information

sjasonsmith, 2020-01-07:
I wonder whether you could tune at both temperature extremes, and use linear scales to interpolate PID values for temperatures in the middle?

Perhaps that would not work at all, but if it gave "good enough" results it would avoid the need to store a bunch of different values for different temperature ranges.

italocjs, 2020-01-07:
If it worked would be really easier, i'll try plotting something on excel to see if i can work

sjasonsmith, 2020-09-11:
@italocjs did you do any more experiments with this?

italocjs, 2020-09-11:
> @italocjs did you do any more experiments with this?

Not really,  i'm setting the pids in the start gcode.  when i'm printing at 180-240 i use one pid,  for 240-300 im using another.

jsaintrocc, 2020-12-31:
@italocjs Along those lines I have some simple logic in my printer start gcode (In PrusaSlicer) for this to set custom PID's for most situations.

```
{if first_layer_temperature[0]>=260 && first_layer_temperature[0]<270}
M301 E0 P18.40 I1.44 D58.65
{elsif first_layer_temperature[0]>250}
M301 E0 P23.05 I1.78 D74.73
{elsif first_layer_temperature[0]>240}
M301 E0 P18.33 I1.27 D66.34
{elsif first_layer_temperature[0]>230}
M301 E0 P19.96 I1.43 D69.87
{elsif first_layer_temperature[0]>220}
M301 E0 P21.25 I1.56 D72.52
{elsif first_layer_temperature[0]>210}
M301 E0 P20.58 I1.48 D71.52
{elsif first_layer_temperature[0]>200}
M301 E0 P19.42 I1.38 D68.58
{elsif first_layer_temperature[0]>190}
M301 E0 P21.21 I1.50 D74.85
{elsif first_layer_temperature[0]>180}
M301 E0 P29.89 I2.70 D82.64
{endif}
```

adolson, 2021-05-05:
If there were slots to store multiple PID autotunings to, this could be coupled with the custom preheat settings menu, making the presets option much more valuable, to me, at least (even more if it could load UBL mesh slots!).

Triangulix, 2021-08-08:
I also have issues which seem to origin from the PID tuning only working for a narrow temperature range.
I autotuned tuned for 200°C because that is my default print temperature but I also use high speed profiles which start at 200°C for the first layer and then raise the temperature to 220°C for fast printing. This sometimes trigger a runaway error because the 200°C PID tuning works so badly for 220°C that the heater doesn't reach the temperatur in a reasonable time.

tombrazier, 2022-03-03:
I would be interested to know whether my model predictive control algorithm does better for you than PID. PR #23751.