MarlinFirmware/Marlin #16498: [BUG] Error in marlin 2.0.x/bugfix/dev while compiling

Dario-Ciceri, 2020-01-07:
Hi everyone, sorry in advance for my bad english, anyways i discovered a weird error while compiling the latest version of marlin 2. If you try to enable **both** these setting at line 1000 (on arduino ide):
#define WAIT_FOR_BED_HEATER  and #define PROBING_HEATERS_OFF 

you will get an error relative some missing declaration/definitions!
The solution is simply don't enable that settings and wait for a fixed version of marlin!

The error reported if you try to enable  the setting (#define WAIT_FOR_BED_HEATER ) is:
`C:\Users\name\AppData\Local\Temp\arduino_build_642812\sketch\src\module\motion.cpp: In function 'void do_homing_move(AxisEnum, float, feedRate_t)':

C:\Users\name\AppData\Local\Temp\arduino_build_642812\sketch\src\module\motion.cpp:1256:37: error: 'LCD_MESSAGEPGM' was not declared in this scope

       LCD_MESSAGEPGM(MSG_BED_HEATING);

                                     ^

C:\Users\name\AppData\Local\Temp\arduino_build_642812\sketch\src\module\motion.cpp:1258:7: error: 'ui' was not declared in this scope

       ui.reset_status();

       ^`

boelle, 2020-01-08:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

ellensp, 2020-01-08:
Cannot reproduce. Need Configuration files.

Dario-Ciceri, 2020-01-08:
> Cannot reproduce. Need Configuration files.

Ok I'll do this soon, anyways you don't need any configuration cause it's a standard configuration. You just have to enable the settings

Dario-Ciceri, 2020-01-08:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4035337/Configurations.zip)

ok check this

ellensp, 2020-01-08:
#if HOMING_Z_WITH_PROBE && HAS_HEATED_BED && ENABLED(WAIT_FOR_BED_HEATER)
So needed 3 conditions to trigger this error. Error is LCD_MESSAGEPGM(MSG_BED_HEATING); is called in the routine when a lcd is not enabled.  working on PR to fix

closed 2020-01-09 by thinkyhead