MarlinFirmware/Marlin #16500: MMU2(SKR_1.3 error)

lijinlin19970813, 2020-01-08:
### My Configurations
[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/4033238/Marlin-bugfix-2.0.x.zip)
### Steps to Reproduce
1.SKR_1.3 Motherboard
2.Configure marlin
**Expected behavior:**
Compilation can pass
**Actual behavior:**
[image]
#### Additional Information
Hello everyone, I have a question for you. I have a SKR1.3 motherboard. I want to use it to use the PRUSA_MMU2 function. Simple configuration will report an error, and do I need to configure some pins to communicate with the MMU.

ellensp, 2020-01-08:
You have #define MMU2_SERIAL internalSerial in Configuration_adv.h
Right above that line is the comment. 
```
  // For AVR enable the UART port used for the MMU. (e.g., internalSerial)
  // For 32-bit boards check your HAL for available serial ports. (e.g., Serial2)

``` 
You don't have a AVR so why did you set it to internalSerial?
You need to set this to #define MMU2_SERIAL Serial2  for eg or Serial1 etc

lijinlin19970813, 2020-01-08:
> You have #define MMU2_SERIAL internalSerial in Configuration_adv.h
> Right above that line is the comment.
>
> ```
>   // For AVR enable the UART port used for the MMU. (e.g., internalSerial)
>   // For [...]

Yes, I found something wrong, and now it compiles. Now I just need to update my firmware into the motherboard and plug the MMU2 into the communication serial port I have configured. I want to get a reply before trying to see if it is correct.

lijinlin19970813, 2020-01-08:
I also want to know if any printer can use MMU2 in marlin now.

ellensp, 2020-01-08:
No. That is an over the top sweeping statement. It mean there is higher chance that it might work.
But the machine in question has to have a high quality extruder that can load and unload filament reliably. Ie  every single time without failure which most machines cannot do.   Also some machine just don't have any free serial ports.

nbenzaki, 2020-01-08:
this worked for me:

  // Serial port used for communication with MMU2.
  // For AVR enable the UART port used for the MMU. (e.g., internalSerial)
  // For 32-bit boards check your HAL for available serial ports. (e.g., Serial2)
  //#define INTERNAL_SERIAL_PORT 2
  //#define MMU2_SERIAL internalSerial

/////// this setup works ////
  
  #define MMU2_SERIAL Serial

    #ifdef MMU2_SERIAL
      #define internalSerial=2

      #endif

      //////////////////////////////////

boelle, 2020-01-08:
this looks like a config error and not a marlin bug, closing

closed 2020-01-08 by boelle