MarlinFirmware/Marlin #16503: [BUG] Controller Fan and Case Light repeatedly turn on and off in a loop after uploading 2.0.1 on RAMPS 1.4

JAdams12, 2020-01-08:
Hello everyone! 

I decided to update my X5S to Marlin 2.0 from 1.1.X

I have some of the extra digital pins hooked up to relays for the hotend cooling fan, the controller fan, and a case light. Because I'm using relays (and therefore not PWM pins) I just have all of the PWM values set to 255. Here are the sections from Configuration_adv

`/**
 * Controller Fan
 * To cool down the stepper drivers and MOSFETs.
 *
 * The fan will turn on automatically whenever any stepper is enabled
 * and turn off after a set period after all steppers are turned off.
 */
#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 16           // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 20             // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 255           // 255 == full speed
  //#define CONTROLLERFAN_SPEED_Z_ONLY 127  // Reduce noise on machines that keep Z enabled
#endif

########################################################

/**
 * Extruder cooling fans
 *
 * Extruder auto fans automatically turn on when their extruders'
 * temperatures go above EXTRUDER_AUTO_FAN_TEMPERATURE.
 *
 * Your board's pins file specifies the recommended pins. Override those here
 * or set to -1 to disable completely.
 *
 * Multiple extruders can be assigned to the same pin in which case
 * the fan will turn on when any selected extruder is above the threshold.
 */
#define E0_AUTO_FAN_PIN 17
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed
#define CHAMBER_AUTO_FAN_TEMPERATURE 30
#define CHAMBER_AUTO_FAN_SPEED 255

##################################################

/**
 * M355 Case Light on-off / brightness
 */
#define CASE_LIGHT_ENABLE
#if ENABLED(CASE_LIGHT_ENABLE)
  #define CASE_LIGHT_PIN 23                  // Override the default pin if needed
  #define INVERT_CASE_LIGHT false             // Set true if Case Light is ON when pin is LOW
  #define CASE_LIGHT_DEFAULT_ON true          // Set default power-up state on
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 255   // Set default power-up brightness (0-255, requires PWM pin)
  //#define CASE_LIGHT_MAX_PWM 128            // Limit pwm
  //#define CASE_LIGHT_MENU                   // Add Case Light options to the LCD menu
  //#define CASE_LIGHT_NO_BRIGHTNESS          // Disable brightness control. Enable for non-PWM lighting.
  //#define CASE_LIGHT_USE_NEOPIXEL           // Use Neopixel LED as case light, requires NEOPIXEL_LED.
  #if ENABLED(CASE_LIGHT_USE_NEOPIXEL)
    #define CASE_LIGHT_NEOPIXEL_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
  #endif
#endif`

As you can see I have the controller fan on D16, the hotend fan on D17, and the caselight on D23. 

I didn't make any hardware changes from 1.1.X to 2.0.X, and my old configuration_adv has the same values. Am I missing something?

Thank you

aleknest, 2020-01-08:
I have the same problem on fysetc s6

ManuelMcLure, 2020-01-08:
Try enabling ``CASE_LIGHT_NO_BRIGHTNESS`` - that will ignore ``CASE_LIGHT_DEFAULT_BRIGHTNESS`` and access the pin as a digital pin instead of trying to do PWM.

JAdams12, 2020-01-08:
I think I figured it out, and boy do I feel stupid. The extra pins I was using (D16 D17 D23) are all where the LCD adapter goes on the RAMPS board, and I never uncommented my LCD (this firmware was also used on my Fusebox2 as the kinematics are nearly identical, just with a bigger bed)

So I think the mainboard was trying to send signals to the LCD that wasn't there, and my relay bank was just going crazy!

closed 2020-01-08 by JAdams12

aleknest, 2020-01-08:
Problem still here for me. FYSETC S6, M355 S1 - and i have 4V on led strip.
Connect strip to 12V directli - all right, it's alive
Replace heat bed and led strip (and exchange pins in Marlin) - heat bed work correct.
As i think, software problem with case light 

#define CASE_LIGHT_ENABLE
#if ENABLED(CASE_LIGHT_ENABLE)
  #define CASE_LIGHT_PIN PC8                  // Override the default pin if needed
  #define INVERT_CASE_LIGHT false             // Set true if Case Light is ON when pin is LOW
  #define CASE_LIGHT_DEFAULT_ON true          // Set default power-up state on
  #define CASE_LIGHT_DEFAULT_BRIGHTNESS 255   // Set default power-up brightness (0-255, requires PWM pin)
  #define CASE_LIGHT_MAX_PWM 255              // Limit pwm
  #define CASE_LIGHT_MENU                   // Add Case Light options to the LCD menu
  #define CASE_LIGHT_NO_BRIGHTNESS            // Disable brightness control. Enable for non-PWM lighting.
  //#define CASE_LIGHT_USE_NEOPIXEL           // Use Neopixel LED as case light, requires NEOPIXEL_LED.
  #if ENABLED(CASE_LIGHT_USE_NEOPIXEL)
    #define CASE_LIGHT_NEOPIXEL_COLOR { 255, 255, 255, 255 } // { Red, Green, Blue, White }
  #endif
#endif

Controller fan - work as led strip - 4V