MarlinFirmware/Marlin #16504: [BUG] Boot Loop for Ender 3 SKR v1.3 Dual Extruder setup

LilJashy, 2020-01-08:
### Bug Description

Upon uploading firmware that has been modified to enable dual extrusion, Ender 3 will not boot correctly but gets stuck in a loop, showing the boot screens every 10 seconds.

### Steps to Reproduce

1. Modify bugfix 2.0 to support dual extruders
2. Power on printer to upload firmware
3. Watch it get stuck in boot loop

**Expected behavior:** Printer to turn on normally

**Actual behavior:** Boot loop

[Marlin-bugfix-2.0.x Boot Looper.zip](https://github.com/MarlinFirmware/Marlin/files/4036193/Marlin-bugfix-2.0.x.Boot.Looper.zip)

2kingsrock, 2020-01-08:
Can you please check to see if you have a SD card mounted during boot?  I have seen random reboots possibly similar to what you are seeing when I have a SD card mounted on boot on some of my hardware.  I leave the SD cards unmounted now on any Marlin 2.0 builds.

sjasonsmith, 2020-01-08:
I think @Vertabreak already figured this issue out, I don't know if they posted a PR for it yet.

LilJashy, 2020-01-08:
I've tried both ways, @2kingsrock. and @sjasonsmith, he's working on my problem right now because the fix he expected to work didn't work

Vertabreak, 2020-01-08:
just to verify this ender 3 has the following mods a 2 in 2 out hotend & LPC1768 main board with a4988 drivers.
i have looked over the config i don't see any major issues but there are a few things i find suspect.
try enabling #define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors. see if that helps.

LilJashy, 2020-01-08:
Ah actually I have TMC2208s - forgot to update that after downloading most recent bugfix. But yes everything else is correct. I'm trying your suggestion now.

Vertabreak, 2020-01-08:
should correct the driver timing settings and adjust motors directions as needed for tmc.

LilJashy, 2020-01-08:
You're a wizard, @Vertabreak. That did the trick. Thanks so much!

Vertabreak, 2020-01-08:
yea i had suspected the bug literally got saved to the eeprom so even after we fixed it you still had issue.

thisiskeithb, 2020-01-08:
> yea i had suspected the bug literally got saved to the eeprom so even after we fixed it you still had issue.

And `M502`/`M500` couldn't clear it?

Vertabreak, 2020-01-08:
well he was stuck in a bootloop unable to issue commands at all.

LilJashy, 2020-01-08:
Where can I find the driver timing settings I need for the tmc2208s?

Vertabreak, 2020-01-08:
[image]

boelle, 2020-01-08:
so this one is more or less solved?

closed 2020-01-09 by LilJashy