MarlinFirmware/Marlin #16505: custom DOGM status screen from 1.x to 2.x ?

fred2088, 2020-01-08:
from 1.x, we just change the ultralcd_impl_dogm.h to a new header file in order to create a new look of status screen,

we are digging into the source, found it's sperate into many .cpp and .h  locate into dogm folder.
status_screen_DOGM.cpp is duplicate with different name from dogm folder and try to compile , but got duplicate function error from both files

how to modify this from 2.x ?
we try not to modify the existing dogm releated files, because this may keep our own version without overwrite accidentically.

if this is not easy, the only way is to modify the dogm releated files

Thanks a lot

boelle, 2020-01-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-01-08 by boelle

fred2088, 2020-04-28:
thanks a lot, found the solutions after look deep down to the code ...