MarlinFirmware/Marlin #16506: [BUG] Dual-Z Steps ignored with TMC2208

FlailAway, 2020-01-08:
### Bug Description

DEFAULT_AXIS_STEPS_PER_UNIT changes are ignored
### My TWO Configurations
Mint 19.3+Cinnamon
Win 8.1
Arduino 1.8.10 on both devices
MKS_GEN-L board
TMC-2208 x 5
3 x Z axis motors, two with 2:1 adapter on "Z" and 1 on "E1"
2 x Left-most jumpers under TMC-2208 on board connected, third open
.zip attached
**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1: #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 2000, 500 }
2: change to #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 1000, 500 }

**Expected behavior:** [What you expect to happen]
Z to raise and lower fixed amount according to "2000" or "1000" etc.

**Actual behavior:** [What actually happens]
Changing 2000 to anything else I always get the same Z-travel of 20mm.

#### Additional Information
IF I swap out the TMC2208 for DRV8825, install the 3rd jumper under the DRV8825 and change the 5 x Driver assignments "#define Z_DRIVER_TYPE  TMC2208_STANDALONE" to "#define Z_DRIVER_TYPE  DRV8825" then it works fine.

Is it me or ???

[marlin-bf-1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/4036811/marlin-bf-1.1.9.zip)

randellhodges, 2020-01-08:
Did you reset your EEPROM after updating the firmware? These are also stored in the EEPROM and will be read from there.

`M502` then `M500`

http://marlinfw.org/docs/gcode/M502.html

FlailAway, 2020-01-08:
> `M502` then `M500`

Arrgghhh, my mistake, thanks. I did not realize they were also stored. I guess I should have thought about it. I will close this and slink away in shame.  :)

closed 2020-01-08 by FlailAway