MarlinFirmware/Marlin #16508: Iusses to get Max31855 working

NAPCAL, 2020-01-09:
### Bug Description

Have to enable external SD card support to get the SPI bus active to add a MAX31855.
After getting readings from the MAX31855, the readings are double of the ambient temp. 

### My Configurations

Attached as Marlin_configuation_files.zip

### Steps to Reproduce

1. Connect the Adafruit MAX31855 amp boards CLK to the (EXP2) SD cards CLK line, DO connection to the (EXP2) MISO line. In your boards pin file define a MAX6675_SS_PIN to a pin that is not used on one of the AUX connectors. Connect the CS to the MAX6675 SS PIN defined. Connect power to the Adafruit board.
 
2. Configuration.h; Change the #define TEMP_SENSOR_0 to -3
Enable SDSUPPORT
Compile
 
3. The temp for the hotend will show double the room temp. 
This is because of the MAX6675 uses LSB 2^1, and the MAX61855 uses LSB 2^2.
See the spec sheet for MAX61855 page 10.

**Expected behavior:** [What you expect to happen]
Have the SPI bus active when MAX31855 is selected and SDSUPPORT is not enabled.
Provide the correct temp readings.

**Actual behavior:** [What actually happens]
Reading double the actual temp.

#### Additional Information
I added this code to temperature.ccp to correct the actual double temperature.
Section #if HAS_MAX6675 this code at the end of that section.

   #if ENABLED(MAX6675_IS_MAX31855)
        max6675_temp = max6675_temp / 2;
    #endif

    return int(max6675_temp);
[Marlin_configuation_files.zip](https://github.com/MarlinFirmware/Marlin/files/4038576/Marlin_configuation_files.zip)

  }

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

NAPCAL, 2020-01-09:
Yes I tried two K-type thermocouples that are not from the same manufacture with the same result.  

The testing was with the thermocouples in open air not in contact with anything. 

SPI_HALF_SPEED will not effect the readings from the MAX31855 because it sends data based on the CLK speed, it was a selection to insure testing to get things working. 

I connected my logic analyzer to the EXP2 and got no CLK on any of signal pins until I enabled SDSUPORT.

Also in the Marlin code for the 1st SPI it defaults to 4MHz for the CLK. 

Please see the data sheet referenced in the issue post.

As far as the conversion the +temps ‘binary’ from the table in the data sheet converts to decimal as double because of not using binary bit 1, this has to be compensated to get good readings.

My fix is a temporary one to allow reproduction of the issue.

On a Sparkfun red board, I can use the Adafruit library to get readings from the MAX31855, but that code doesn’t react to the three fault detections, I also tried the Sparfun library it also provided good temp readings and reacted to all three faults.

boelle, 2020-01-14:
@NAPCAL still an issue?

Grogyan, 2020-01-14:
@boelle I believe that the issue is somewhat resolved due to work on the PR for STM32 timer fix.

I'm curious as well if the timer HAL fix, fixes this issue as well

NAPCAL, 2020-01-18:
> @NAPCAL still an issue?

I haven’t been home to try it yet.

boelle, 2020-01-23:
@NAPCAL been home yet?

boelle, 2020-01-26:
anyone else been able to reproduce and confirm this one?

boelle, 2020-01-30:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-30 by boelle

rmcrabb, 2020-05-14:
I just encountered this exact issue. Using the SPI for the SD card my temp readings are doubled. Using the SPI for the stepper drivers I get the correct reading but periodic Max temp faults.

Grogyan, 2020-05-14:
That indicates that there is a HAL issue,
As the code works for the LPC176x for the SKR1.3/1.4

Grogyan, 2020-05-14:
Ie if there is an extra bit shift, this will show up as readings ~doubled

AnHardt, 2020-05-14:
Could it also be the infamous R2 pullup resistor on SKR-PROs TF_MISO? Is TCSS lasting long enough?
Is the temperature always doubled, or is one bit always high? Maybe at a bytes border.

rmcrabb, 2020-05-18:
Can someone point me in the right direction to get this max31855 working?

Grogyan, 2020-05-18:
@rmcrabb you will need to elaborate, as I use the Max31855 for years now, and it is working