MarlinFirmware/Marlin #16517: Recv: echo:busy: paused for user  For no apparent reason, cant continue

italocjs, 2020-01-09:
# Bug Description
Something is calling this paused for user, but i'm having a hard time tracing down what exactly,  filament change is disabled,  there isnt anything that would call this afaik.  i have installed a serial logger to be sure, and there is nothing in the serial stream calling for the pause.  I tried sending commands to continue but didnt work. i cant continue (this printer have no LCD, it use astroprint (which is based os octoprint) to send gcodes over usb)
 
No one:
Printer: i'm gonna pause this
(Just kidding), 

### My Configurations
Attached are config.h / config.adv / serial LOG.
[serial.log](https://github.com/MarlinFirmware/Marlin/files/4042437/serial.log)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4042440/Marlin.zip)

I'm using marlin 2.0.x bugfix, got the version on 6/1/2019

thisiskeithb, 2020-01-09:
Can you attach the gcode you were printing? There might be something sending the commands/bugging out.

italocjs, 2020-01-09:
All the gcodes sent/received since the printer was turned on are in the serial.log.  these gcodes I sent using a small macro that I made (to load / unload filament). The gcodes at least to me seems ok, the only thing "not so ordinary" that I used was the m400, to avoid turning the hotend off before the movement finished.

gandzia44, 2020-01-13:
I got the same bug on 3 printers now. Problem aparece for sure when there is really long print, otherwise it's random on small prints. Everything started when I updated to 2.0.1, before that I was using 2.0.x bugfix from "2019-12-6" and everything was ok.

My mainboard: Arduino Due
Raspberry server: Repetier-Server

italocjs, 2020-01-13:
Hi @gandzia44 , were you using M400 anywhere in your code? i think that when i stopped sending it, the error disappeared (i'm not 100% sure, i only did a single 4h print)

italocjs, 2020-01-16:
I have printed around 30h without any issues so far, the only thing i did was not using M400, so something might be wrong there.   

I was using M400 to avoid the nozzle changing temperature before time (while moving),  i dont know why but this was happening, maybe the buffer giver some kind of priorty to M104-109 commands
.

boelle, 2020-01-23:
a bit confused, is this one solved?

italocjs, 2020-01-23:
I wouldn't say "solved" as i didn't had time to test the emergency parsers, but it will probably be. I'll try compiling it this week and report back

closed 2020-01-23 by italocjs