MarlinFirmware/Marlin #16519: M421 Q offsets not being applied

Erikthedude, 2020-01-10:
My printer always reads higher on the right side when I perform a G29 (Bilinear with a Bltouch). Since I cannot use UBL (lacking memory), I wanted to still use Bilinear leveling but adding a offset to certain points. Let say I perform a simple 2x2 leveling grid and get these results:
`Recv:       0      1
Recv:  0 +0.000 +0.185
Recv:  1 +0.135 +0.123`

If I start my print with this mesh, the entire right side of my print is about 0.5mm off the build plate. So I want to add a -0.5mm offset to both points located on the right on the x axis, I added this after my G28 and G29:
`M421 I1 J0 Q-0.5
M421 I1 J1 Q-0.5`

However, no changes appears when I try to print something that uses all my print area, the right side is still 0.5mm off the bed. If I cancel the print and do a M503, I see that the offsets are being applied to both specified points, but that doesn't translate to the print, z axis motors are not moving when going to the right side of the bed:
`Recv: echo:  G29 W I0 J0 Z0.00000
Recv: echo:  G29 W I1 J0 Z-0.31500
Recv: echo:  G29 W I0 J1 Z0.13500
Recv: echo:  G29 W I1 J1 Z-0.37750`

I'm also aware of the ABL Expert plugin on Octoprint which can be used to apply probe correction on axixes, but I want a specific offset for specific points, which the plugin doesn't offer. I tried looking up for anyone doing the same thing, but couldn't find any usefull information about using M421 Q and Bilinear leveling. Any help would be greatly apprecied!

boelle, 2020-01-10:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Erikthedude, 2020-01-10:
Sure, here's my config:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4046515/Marlin.zip)

And if that matters, my full start gcode:

> M280 P0 S160 ; BLTouch alarm release
> G4 P100 ; delay for BLTouch
> G28 ; home
> G29
> M420 S1
> M421 I1 J0 Q-0.5
> M421 I1 J1 Q-0.5
> M900 K0.14 ; K Factor for linear advance
> G1 X10 Y20 Z0.4 F3000 ; get ready to prime
> G92 E0 ; reset extrusion distance
> G1 Y100 E20 F600 ; prime nozzle

Erikthedude, 2020-01-10:
Interresting enough, I just validated that my offseted mesh is in fact used when I do a G26,  all 4 points are leveled and both z axis motors are compensating, but it is completely ignored when I lauch a print

boelle, 2020-01-24:
how often do you update? bugfix is updated almost daily

boelle, 2020-01-30:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-30 by boelle