MarlinFirmware/Marlin #16520: [BUG] Encoder changes temperature with a huge skips

VijitCoder, 2020-01-10:
> Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

No, I'm not. The link is broken.

### Bug Description

On the printer go to temperature changing. **Slowly** turn the encoder - at the first few steps the temperature will be changed with a small steps, then suddenly skips 100C twice. And it happens in any direction.

The only way to tune up the temperature is to rotate encoder for a few steps, then wait 5-6 seconds, rotate a bit more and etc. If you're be lucky, the skips will happen not too often.

### My Configurations

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4045544/configs.zip)

### Steps to Reproduce

I made a [video](https://youtu.be/eq1b6uEYrCg). I think it should illustrate the problem better than words.

**Expected behavior:** Predictable changes according encoder steps. 

**Actual behavior:** Unpredictable changes by encoder. 

#### Additional Information

Printer: Creality Ender 3
Main board: SKR mini E3 Dip, v1.1

I tried to change `ENCODER_10X_STEPS_PER_SEC` and `ENCODER_100X_STEPS_PER_SEC` - useless. Also, when I switched off `ENCODER_RATE_MULTIPLIER` - encoder works, but veeery slowly changes the values.

thisiskeithb, 2020-01-10:
This might be related to https://github.com/MarlinFirmware/Marlin/issues/16298.

VijitCoder, 2020-01-10:
Yes, I saw that report. Maybe I posted the same after that, but in my case no reduction in values of `ENCODER_10X_STEPS_PER_SEC` and `ENCODER_100X_STEPS_PER_SEC` produced results.It seems, they are ignored at all. Or they should be extremely small, I don't know for now.

boelle, 2020-01-10:
@VijitCoder so you have the same problem as in #16298 ?
if so it might be better to report there instead so we only have one issue open for the same problem

VijitCoder, 2020-01-11:
I'm not sure that this is the same problem. There was said:

> pixel corruption on display

In my case no corruption but the speedy changes of the temperature value. 

So I see the only coincidence - 32-bit controller.

borland1, 2020-01-11:
In file Configuration.h, under section "Encoder Settings", there are default override settings for ENCODER_PULSES_PER_STEP and ENCODER_STEPS_PER_MENU_ITEM.   Have you tried increasing these to see the affect.

VijitCoder, 2020-01-12:
Just checked:
```
#define ENCODER_PULSES_PER_STEP 4
#define ENCODER_STEPS_PER_MENU_ITEM 1
```
No effect.

borland1, 2020-01-12:
Those are the default values.  Did you try overriding those with increased values, like say 16 and 4.  ?

VijitCoder, 2020-01-13:
> Those are the default values

No, by default they are switched off.

I don't understand, what the relation between my problem and these options. Nevertheless, I has tried several values and stopped on this:

```
#define ENCODER_PULSES_PER_STEP 8
#define ENCODER_STEPS_PER_MENU_ITEM 1
```

`ENCODER_PULSES_PER_STEP 8` doesn't solving the problem,  but make it harder to reproduce cause now I need to rotate encoder a bit longer. In the end the temperature jumps +-100C again, if I will rotate encoder fast and long enough. I mean, config `ENCODER_10X_STEPS_PER_SEC` is still ignored, and maybe `ENCODER_100X_STEPS_PER_SEC` ignored too.

`ENCODER_PULSES_PER_STEP 16` leads to very slow reaction in the menu at all. But remains the problem with a temperature tuning.

borland1, 2020-01-13:
They are overrides, so you can only change the values, they are already switched on.

The pulses per step and steps per menu item are how fast the menus respond to rotary action, while the  encoder 10x and 100x are for how often the firmware looks at the encoder for changes in knob rotary movement.

You need to find a sweet spot for your encoder, but that is could be a time consuming process.

VijitCoder, 2020-01-13:
Is that what you're assuming OR is that what for you made these options? 

According your logic

>  how often the firmware looks at the encoder

the higher values should give to user more precises value changes, right? But it doesn't. Anyway, I've tried multipliers from default values (30/80) down to 5/5. No difference at all.

Going further, here is the copy-paste from my Configuration_adv.h
```
// Change values more rapidly when the encoder is rotated faster
#define ENCODER_RATE_MULTIPLIER
#if ENABLED(ENCODER_RATE_MULTIPLIER)
  #define ENCODER_10X_STEPS_PER_SEC   5  // (steps/s) Encoder rate for 10x speed
  #define ENCODER_100X_STEPS_PER_SEC  5  // (steps/s) Encoder rate for 100x speed
#endif
```
In the first line provided explicit description: these multipliers should change values `+-5` if encoder makes 10/100 steps per second. Did you see the video from the bug description? Is it looks like +-5°C per 10 steps/s of encoder? I don't think so. It's more like +-100°C.

If there is nothing else that can help me with a problem, close the report. I'll try to debug the firmware, when (if) I'll have time for it.

VijitCoder, 2020-01-13:
What a nice surprise! I tried again the default values
```
  #define ENCODER_10X_STEPS_PER_SEC   30
  #define ENCODER_100X_STEPS_PER_SEC  80
```
and it works pretty well!

I think, I misunderstanding the purpose of these options or comments to them. The bigger multipliers leads to slower values changes. Something like this.

This case can be closed. Thanks for all.

closed 2020-01-13 by VijitCoder