MarlinFirmware/Marlin #16521: [FR] Ender-3 Pro Configuration Example

aduffner, 2020-01-10:
Hello, I am not sure if this is the right way to ask for this:

### Description

Please add an Ender-3 Pro Configuration Example for newbies like me. The Ender-3 Config is not the same and different in some things (for example bed size and some temp settings).

I could provide my edited config, but as I said I am a newbie and don't want to share settings which could be wrong.

Vertabreak, 2020-01-10:
if you post yours ill check it an submit it once its in adjustments can always be made if needed.

thisiskeithb, 2020-01-10:
The Ender-3 and Ender-3 pro are the same firmware-wise except for a few minor difference (Status screen, name, etc.) that weren’t enough to justify adding a new config in the past since we already have an Ender-3 config.

Has there been a recent hardware refresh on the pro?

ellensp, 2020-01-10:
Aparently yes, some have tmc2208's https://creality.com/info/se-termine-3-se-termine-3-pro-se-termine-5-firmware-code-source-mis-niveau-i00025i1.html

thisiskeithb, 2020-01-10:
I guess we shouldn’t be surprised that silent updates continue to happen without revision numbers or dates.

Sure makes maintaining configs a chore 🤪

Vertabreak, 2020-01-10:
geeetech does the same thing i hate it ....

thisiskeithb, 2020-01-10:
Can anyone find a link to an Ender-3 Pro being sold with the new silent board (with TMC2208s)? 

Perhaps Creality's firmware download is for Ender-3 Pro users who upgraded their stock board to the silent one.

aduffner, 2020-01-10:
Hello everyone,
thanks for all your responses.
 
My Ender-3 Pro have a build size of 220x220x250mm and the board from Creality is their Melzi Board in Version 1.1.4, delivered with the TMC2208S from factory.
I have screwed it open to look this up and flash the missing boot loader, which I needed to update the firmware to the current marlin one.

Here are my config files @Vertabreak, thank you for checking them ☺️.
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/4046755/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/4046756/Configuration_adv.txt)

thisiskeithb, 2020-01-10:
> My Ender-3 Pro have a build size of 220x220x250mm

Ender-3s are sold as 220x220, but can use the full 235x235mm, so it was updated here to use the full build volume.

> The board from Creality is their Melzi Board in Version 1.1.4, delivered with the TMC2208S from factory.

Do you have the updated Pro status screen or a photo by chance? Creality's download page only includes compile firmware.

There shouldn't be much to change other than status screen, printer name, driver types & possibly direction.

aduffner, 2020-01-10:
Hello @thisiskeithb,

> Ender-3s are sold as 220x220, but can use the full 235x235mm, so it was updated here to use the full build volume.

I can't really believe in this, because when I have 235x235mm setted in the firmware it wouldn't stop at the "end" of the axis and makes my x motor struggling.

> Do you have the updated Pro status screen or a photo by chance? Creality's download page only includes compile firmware.

In this video you can see the TMC2208:
https://youtu.be/_wEZOAFZhB4?t=115

thisiskeithb, 2020-01-10:
> I can't really believe in this, because when I have 235x235mm setted in the firmware it wouldn't stop at the "end" of the axis and makes my x motor struggling.

We should get a few more reports to make sure Creality has really made the bed/printing area smaller on the Pro. It could just be how your particular machine was constructed.

> In this video you can see the TMC2208:
https://youtu.be/_wEZOAFZhB4?t=115

I’m looking for the text/graphic that says “Ender-3 Pro” similar time the existing configs.

aduffner, 2020-01-10:
>We should get a few more reports to make sure Creality has really made the bed/printing area smaller on the Pro. It could just be how your particular machine was constructed.

I have measured it, the bed is 235mm or a few more mm sized square.

> I’m looking for the text/graphic that says “Ender-3 Pro” similar time the existing configs.

This you get in the Video 5 seconds before, on minute 1:50.
Or do you mean especially my printer?

thisiskeithb, 2020-01-10:
> This you get in the Video 5 seconds before, on minute 1:50.

I see it. I also found a different Ender-3 Pro status screen with similar style to the Ender-5 Pro used above, so I'll create one with the sans serif font 🙂.

So now we'll have two Ender-3 Pro models: One with 2208s and one with A4988s to maintain backward compatibility with older Ender-3 Pros. I'm sure the Ender-5 Pro will be next...

PGrave, 2020-01-10:
@thisiskeithb , check if this is what you want, regarding Ender 3 Pro graphic (change the extension from .txt to .h)
[_Statusscreen.txt](https://github.com/MarlinFirmware/Marlin/files/4047094/_Statusscreen.txt)

aduffner, 2020-01-10:
> So now we'll have two Ender-3 Pro models: One with 2208s and one with A4988s to maintain backward compatibility with older Ender-3 Pros. I'm sure the Ender-5 Pro will be next...

Thanks a lot, man. I really appreciate it.

So ... should I close this thread?

Lord-Quake, 2020-01-10:
@PGrave The screen is already available: https://github.com/MarlinFirmware/Marlin/issues/12537

PGrave, 2020-01-10:
> 
> 
> @PGrave The screen is already available: #12537

@Lord-Quake , if it is available, I've never seen it before! So it is that I've made one for myself, which you can see, is different from yours: 
[image]

PGrave, 2020-01-10:
@Lord-Quake , as @thinkyhead said on the original #12537, the updated status file would be on a separate folder, inside Creality example configs. It isn't there, in any 2.0 branch. Maybe it could be done now, and future related questions/issues avoided.

Lord-Quake, 2020-01-10:
@PGrave Correct. I never noticed that it isn't included as I always use my own files. I also noticed that the file is actually outdated now. I created a version that is more centered on the screen.
 
Here the newer version if interested: [_Statusscreen.zip](https://github.com/MarlinFirmware/Marlin/files/4047464/_Statusscreen.zip)

Aianir, 2020-01-12:
> > In this video you can see the TMC2208:
> > https://youtu.be/_wEZOAFZhB4?t=115

His board was an early silent board marked version 1.1.4, Creality screwed up the version numbering. Newer (at least as pr. december 2019) Ender 3 Pro's come with motherboard version 1.1.4 that have the original A4988 drivers. You can buy as an addon the "silent 1.1.5", which have TMC 2208s. 

https://www.creality3dofficial.com/products/creality-silent-mainboard-v1-1-5

tamsencooperlumira, 2020-01-19:
> His board was an early silent board marked version 1.1.4, Creality screwed up the version numbering. Newer (at least as pr. december 2019) Ender 3 Pro's come with motherboard version 1.1.4 that have the original A4988 drivers. You can buy as an addon the "silent 1.1.5", which have TMC 2208s.

This is the reason I have looked into firmware upgrades. Having only owned a 3d printer since December (so still very new) The silent 1.1.5 boards with TMC 2208s (and they do make an incredible difference to the sound) have lost a feature from the LCD screen 'filament change' and I have been trying to find a way to get that back. Looking online, its hard to find a clear Configuration_adv.h for the Ender 3 pro with the 1.1.5 board. The official Creality response to my question about it was "This option is not added to our latest version of the V1.1.5 motherboard. You can heat the nozzle above 185 degrees, then set a command of moving E-axis to send the filament in and out." 

So any support in finding a clear way to upgrade the firmware of an Ender 3 pro with 1.1.5 board to get this feature back would be greatly appreciated.

Aianir, 2020-01-19:
> The official Creality response to my question about it was "This option is not added to our latest version of the V1.1.5 motherboard. You can heat the nozzle above 185 degrees, then set a command of moving E-axis to send the filament in and out."

The Creality boards are pretty limited in memory size, so they may have had to remove the filament change command to get enough room to fit the 2208 driver.

closed 2020-03-20 by aduffner