MarlinFirmware/Marlin #16524: MULTI_NOZZLE_DUPLICATION limited to 2 copies.

SlickNickeL, 2020-01-10:
Currently only hotend E0 and E1 will engage.

CONDITION:
M605 S2 P7, Sets status of copies for extrusions on E0,E1,E2. 

RESULTS:
Only E0 and E1 are active. E2 is never active under any P(1-15)

PROPOSAL:
The current P(number) does not address a condition past 3 extruders (and currently doesn’t work)
Allow all connected extruders to duplicate from E0.

Simplify P.

P1= (on) with one copy from E0 to E1, Total print count = 2
P2= (on) with two copies from E0 to E1,E2, Total print count = 3
P3= (on) with three copies from E0 to E1,E2,E3, Total print count = 4
And so on.

I’m familiar with the discussion, to be able to exclude an extruder or turn one off in mid print. Please use KISS first and then address this issue. 
Out of 120 continuous 3 head duplication runs under Repetier, we have had only one issue from a filament delamb where a flaw was inserted, but the print finished with no problems to the other two prints. We can simply disengage the extruder manually if need be, but I understand that many machines can’t do this. I think this is a low priority at this time. A well thought out manufacturing fff printer, has contingencies for such problems built into the design.

boelle, 2020-01-10:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

SlickNickeL, 2020-01-10:
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/4046539/Configuration.h.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4046541/Configuration_adv.zip)

boelle, 2020-01-24:
how often do you update? bugfix is updated almost daily

SlickNickeL, 2020-01-24:
I did the latest bugfix before posting this. Are you able to get three heads to print simultaneously since then?

boelle, 2020-01-24:
no i dont have a printer with 3 heads

my post was more a question since your issue could have been fixed during the last 13 days

InsanityAutomation, 2020-01-24:
No, that section of the code hasn't been touched. I'll make it around to there when the gtr10 arrives and I can rig up some high stepper count test stations.  It didn't make it out of China in time. The original code was written blind so with hardware in hand debug should be relatively quick.

SlickNickeL, 2020-01-24:
My initial thought, was everything is still in place for 2 extruders, but it was never tied to a third or forth extruder. Marlin has always done 2 heads.  I suggest that all code, clones from E0, and leave PID per head alone, so each print head maintains its own PID and Extrusion  Multiplier. Let software control this aspect. Let me know when to test, I will send it to the floor for testing. We REALLY want to use Marlin as our firmware.

SlickNickeL, 2020-01-30:
How is this going? Anything yet?

InsanityAutomation, 2020-01-30:
Board to debug with hasn't hit the us yet... Hopefully next week..

SlickNickeL, 2020-01-30:
I have a Rumba board I can throw some steppers drivers on and get it out to you.

InsanityAutomation, 2020-01-30:
If it comes down to it and doesn't show up soon, I'll end up using a 6 driver skr pro probably. Just hoping to use something I plan to actually have 4 extruders love on! Day job just let up from holiday mayhem and my weekend plumbing problems are hopefully dealt with. Next 2 weekends should see some progress on my project list!

SlickNickeL, 2020-01-30:
Let me know how your build goes. We can go to 4 extruders at one time but would have to dump the Rumba. We are researching autoleveling multi head printers, and have some interesting findings I can share if you are interest. This is for fixed head gantry.

InsanityAutomation, 2020-01-30:
I can share some info on my concept of you'd like, can email me d.menzel [at] insanityAutomation.com

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

SlickNickeL, 2020-07-03:
This was the testing after the code was produced for multi head Duplication. We tested it and it didn’t work. #16534 was the results, and now the thread has been closed by Bo. This thread ties directly into the Feature for more than 2 heads printing simultaneously. If I recall, none of the heads could be addressed correctly with all kinds of jitters and stuck prints, with no more than 2 heads ever working.  So it looks as if the code was produced, but it never worked.

sjasonsmith, 2020-07-03:
@SlickNickeL I've re-opened this issue for you. Since this topic was idle for a long time, it is possible something has changed. Would you mind updating your configuration files so they build with the latest `bugfix-2.0.x` branch, and verify if this problem still exists? If so, please attach those updated configuration files so it is easier for somebody else to join in and help.

sjasonsmith, 2020-07-03:
@InsanityAutomation, have you done anything else related to this feature (including using it!) since this went idle about 6 months ago?

SlickNickeL, 2020-07-04:
I tested on January 10 2020. 

I asked if there was any progress on this in May, and I was told no. 

I asked the person who was working on it, a couple of weeks ago and he said nothing has been done to this as yet. Said he didn’t have the time.

No sense in uploading any files, since I simply brought in the latest build, and tested it. I haven’t seen anyone address this issue.

I’m a patient person. Let people do their thing, and I know these things take time. Problem is, if it’s canceled then what’s the point?  

There seems to be some miscommunication or a lack of knowledge of what’s going on. I have followed this closely from its conception. I really don’t want to start all over. 

I do not write C++code, I am a developer and user. I do understand some code and it’s outlay, and write proficiently in G/M code for programing, and very much aware of the differences in the various code between Repetier and Marlin.  

There has yet to be a unification of Ditto/ Duplication in G-code commands between what Repetier has done and Marlin has yet to do. I have already posted these differences.

I asked the individual who was working on it to keep me abreast of any changes, so that I can test the code. This was maybe last week or the week before.

closed 2020-08-09 by github-actions[bot]