MarlinFirmware/Marlin #16525: [BUG] M108 not breaking "pause for user" loop

italocjs, 2020-01-10:
### Bug Description
The M108 some times doenst break the loop, effectively keeping me out of the printer (cant continue clicking on LCD because there is no LCD in this machine, i print sending gcodes over usb)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4046391/Marlin.zip)

### Steps to Reproduce
Open any host (tried with repetier and simplify3d, same result):
Sent:  M0/M1
Wait 10-20s
Send M108

**Expected behavior:** [What you expect to happen]
Send M0/M1  - Printer pauses
Send M108 - Printer continues

**Actual behavior:** [What actually happens]
Send M0/M1 - Printer pauses
Send M108 - Nothing

#### Additional Information
I'm using 2.0.x bugfix from 06/01/2019

italocjs, 2020-01-10:
another thing i noticed that i dont understand a reason is: why does M0 block a M105 response? the host should be always aware of the temps.

ellensp, 2020-01-11:
Marlin (more or less) takes in commands sequentially and processes them. M0 holds up the command queue waiting for a lcd click.  The only exceptions to this are things handled by the EMERGENCY_PARSER it handles M108, M112, M410, M876 but nothing else. 
You have not enabled #define EMERGENCY_PARSER in Configuration_adv.h so after you send M0 the only way out is to click the lcd to continue. (or power off)

italocjs, 2020-01-11:
hmm, i'll try enabling the emergency_parser.   thanks for the tip!

This is pretty much what i wanted, a way to get out of the M0 for printers that doesn't have a embedded LCD.  i'm using a raspberry with astroprint in one machine, and a raspberry / simplify3d in another to control the printers, in both i cant "unpause" when i call the M0 for filament changes.

I'll try testing it, for now i'll close this issue since this ill probably work.

closed 2020-01-11 by italocjs