MarlinFirmware/Marlin #16526: [ISSUE] MarlinSerial::flush()  Duplicate comments

desmeckert, 2020-01-10:
NON Critical 

### Bug Description

MarlinSerial.cpp lines 133-136 comments are immediately duplicated
and the comment is completed on line 137-141.

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
My configuration is not relevant.

### Steps to Reproduce
Read the code

No steps needed.

**Expected behavior:** [What you expect to happen]
Minimal download file size.

**Actual behavior:** [What actually happens]
source code bloat/

#### Additional Information

boelle, 2020-01-10:
i only see 56 lines in total (https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/HAL_LPC1768/MarlinSerial.cpp)

so in which folder ?

desmeckert, 2020-01-11:
That is MarlinSerial.h.

I said MarlinSerial.cpp.

Easiest is search for the flush() method.

desmeckert, 2020-01-11:
https://github.com/RichCattell/Marlin/blob/master/MarlinSerial.cpp

Is one version.

boelle, 2020-01-11:
> That is MarlinSerial.h.

no its not, look at the link clearly say MarlinSerial.cpp

> https://github.com/RichCattell/Marlin/blob/master/MarlinSerial.cpp

that is not main marlin, main marlin links start with https://github.com/MarlinFirmware/Marlin/

closed 2020-01-14 by boelle