MarlinFirmware/Marlin #16529: [BUG] #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."

thetechy2, 2020-01-10:
#error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."
when trying to compile and upload a marlin 2.0 build for skr 1.4 using visual studio code and platform io.
i input my configs and it gives me the error listed above even though i have "USE_ZMIN_PLUG" enabled as well as "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN"

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4048618/Marlin.zip)

### Steps to Reproduce

1-download latest marlin 2.0
2- configure for level sensor on zmin plug on skr 1.4
3-build

**Expected behavior:** a successful build/upload

**Actual behavior:** #error "Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN requires the Z_MIN_PIN to be defined."

#### Additional Information
brand new board and nothing plugged into it besides usb

italocjs, 2020-01-11:
I read the config.h,  maybe the issue is in the homing direction (it is going towards the ZMAX, not ZMIN), but i'm not sure if it can work like that.  another thing that might be the issue is that the pin is not defined for the board  you defined ( #define MOTHERBOARD BOARD_BIGTREE_SKR_V1_4 ) .  take a look in its config inside of Marlin\src\HAL

ellensp, 2020-01-11:
Whomever created the pins_BTT_SKR_V1_4.h defined X_STOP_PIN, Y_STOP_PIN and Z_STOP_PIN  instead of all the MIN_PIN and MAX_PIN's
Give this pins_BTT_SKR_V1_4.h file a try 
[pins_BTT_SKR_V1_4.h.zip](https://github.com/MarlinFirmware/Marlin/files/4048814/pins_BTT_SKR_V1_4.h.zip)
(I don't have this board to try, but it now compiles)

italocjs, 2020-01-11:
If it works it might be a good idea to commit these changes to avoid future problems

thisiskeithb, 2020-01-11:
> Whomever created the pins_BTT_SKR_V1_4.h defined X_STOP_PIN, Y_STOP_PIN and Z_STOP_PIN instead of all the MIN_PIN and MAX_PIN's

The SKR 1.4 only has stop pins for each axis because it only has one set of endstops, whereas the SKR 1.3 has both min & max sets.

ellensp, 2020-01-11:
@thisiskeithb not according to the pin diagram from BTT https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4PIN.pdf

thisiskeithb, 2020-01-11:
> @thisiskeithb not according to the pin diagram from BTT https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4PIN.pdf

Did you mean to link something else? There are only stop pins (the others are used for other features):
[image]

ellensp, 2020-01-11:
Oh I see... they repurposed the max's endstop plugs... I didnt see that..

boelle, 2020-01-11:
so is this a user misunderstanding then?

thetechy2, 2020-01-11:
ok so i have a max endstop for the z and an inductive sensor for bed leveling i guess i dont need (and cant use with the current pins configuration) the endstop? maybe have 2 pin layouts for the 1.4 as theyve changed the assignments from the 1.3 for other sensors. i see my error in assuming the plug assignments would be the same, ugh i just should have gone for 2209 drivers to have sensorless homing =/

ellensp, 2020-01-11:
OK. plug all you endstops into the formally called Min endstop plugs. plug your probe into the pwrdet plug (formally z-max) 
In Configuation.h disable "#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN" instead use "#define Z_MIN_PROBE_PIN" ans set the pin to "P1_00"

thetechy2, 2020-01-11:
thank you for all the help everyone, i think ive got it!

ellensp, 2020-01-11:
just to make sure it doesn't interfere also add #define PS_ON_PIN -1 after you defined the MOTHERBOARD in Configuration.h ie 
```
#define MOTHERBOARD BOARD_BIGTREE_SKR_V1_4     
#define PS_ON_PIN -1
```

ellensp, 2020-01-11:
@boelle yes it was a miss understanding. not a bug. seems all sorted now.

closed 2020-01-11 by thetechy2

Talha909, 2020-01-14:
> thank you for all the help everyone, i think ive got it!

Hi,  Is your problem solved. Can you share the config files.

Zero3d, 2020-02-17:
I added a lines in the pins_BTT_SKR_V1_4_Turbo.h file and changed Z_STOP_PIN to Z_MIN_PIN.  Left #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN enabled.  Seems to solve my problem. This is for Anycubic Kossel Linear Plus. 

#define X_STOP_PIN         P1_29
#define Y_STOP_PIN         P1_28
#define Z_MIN_PIN         P1_27
#define X_MAX_PIN          P1_26
#define Y_MAX_PIN          P1_25
#define Z_MAX_PIN          P1_00