MarlinFirmware/Marlin #16532: DATA STORED IN FLASH EEPROM DOES NOT PERSIST MORE THAN ONE POWER OFF/ON/OFFCYCLE AFTER AN M500[BUG] (short description)

Thinkersbluff, 2020-01-11:
### Bug Description

### My Configurations

[MyConfigs.zip](https://github.com/MarlinFirmware/Marlin/files/4049020/MyConfigs.zip)

### Steps to Reproduce

1. Power-On the system
2. Set Fade Height to a non-zero value (e.g. 10), on the Bed Leveling menu.
3. Use Manual Bed Leveling, to store at least one non-zero mesh data point.
4. Select "Store Settings".
5. Change Fade Height to a different value (e.g. 9).
6. Use "Edit Mesh" to modify any mesh value (e.g. X0, Y0)
5. Select "Load Settings".
6. Verify Fade Height changes back to the value set at step 2.
7. Verify the modified mesh value is restored to the value stored at step 3.
6. Power-Off the system
7. Power-On the system
8. Auto-Home or G28 the system
9. Verify the Fade Height is the value stored at step 2.
10. Verify the modified mesh value is the value stored at step 3.
11. Power-Off the system.
12 Power-On the system.
13. If you have reproduced this "BUG", then the Fade Height value and all of the Mesh values will once again be Zero.
14. Repeat steps 1-10.
15. Select "Store settings" or issue an M500 command.
16. Power-Off the system.
17 Repeat steps 7 to 10.  If you have reproduced the "work around" for this bug, steps 9 and 10 will pass correctly.

**Expected behavior:** [What you expect to happen]
I expect all of the data I store in EEPROM to be restored every time Power is turned back ON, regardless of how many times I cycle Power OFF/ON, after I originally store that data.  I do not expect to have to deliberately store that data every time, before I turn Power off.

**Actual behavior:** [What actually happens]
If "Store Settings" is selected on the Bed Leveling menu, after creating mesh data or after setting a non-zero value for Fade Height, then power is turned OFF/ON, the data is restored (which is correct).  
If the power is then cycled OFF/ON again, without first issuing a second Store Settings (M500) command, the data stored is not restored (**this** behaviour appears to be a bug.)
If, instead, the Store Settings menu is selected or an M500 code is issued, before cycling Power OFF, then the originally stored values are again restored, as they should be, when Power is turned back ON.

#### Additional Information

* I am observing these issues with the 11 Dec release of Marlin 2.0.x, on an SKR Mini E3 v1.2.
* When I comment-out the line "#define FLASH_EEPROM_EMULATION" in the file pins_BTT_SKR_MINI_E3.h, then Marlin 2.0.x reverts to using the SD card to store EEPROM.dat.  In that configuration, the EEPROM does work as expected.

thisiskeithb, 2020-01-11:
> I am observing these issues with the 11 Dec release of Marlin 2.0.x, on an SKR Mini E3 v1.2. 

Can you try with the latest [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) release to see if your issue persists?

tpruvot, 2020-01-11:
bullshit, too much caps

Thinkersbluff, 2020-01-11:
Sorry, thisiskeithb.  
I am a total NOOB with Marlin and with PLATFORMIO/VISSTUDIO. I am having trouble compiling a firmware.bin with the 11 Jan 2020 Marlin 2.0.xbugfix...  
I first tried copy/pasting my Configuration.h and Configuration_adv.h files into the Marlin folder of the 11 Jan 2020 Marlin2.0.xbugfix and compiling with the platformio.ini environment set to default_envs = STM32F103RC_bigtree_NOUSB, as I do for the 11 Dec 2019 version.  That compile failed with, Error: Unknown environment names 'STM32F103RC_bigtree_NOUSB'. Valid names are 'BIGTREE_SKR_PRO, rumba32_f446ve, mks_robin_lite3, FLYF407ZG, STM32F103RC_bigtree_USB, STM32F103CB_malyan, ARMED, mks_robin_pro, sanguino_atmega1284p, DUE_debug, STM32F103RC_bigtree_512K_USB, mks_robin, FYSETC_F6_13, mks_robin_mini, linux_native, STM32F4, esp32, STM32F103RE_bigtree, teensy31, teensy35, melzi_optiboot, FYSETC_S6, sanguino_atmega644p, STM32F103RE_bigtree_USB, STM32F103RE, at90usb1286_dfu, mks_robin_nano, BIGTREE_BTT002, STM32F103RC_bigtree, melzi, DUE_USB, mks_robin_lite, LPC1768, STM32F407VE_black, STM32F103RC, STM32F103RC_fysetc, STM32F103VE_longer, megaatmega2560, jgaurora_a5s_a1, SAMD51_grandcentral_m4, STM32F103RC_bigtree_512K, chitu_f103, mks_rumba32, at90usb1286_cdc, DUE, megaatmega1280, LPC1769, STM32F7, rambo, include_tree'"

I tried changing the environment name to STM32F103RC_bigtree_USB, but maybe that was wrong??

The next compile threw two errors about two variable name changes required in Configuration_adv.h:
"#error "FILAMENT_UNLOAD_RETRACT_LENGTH is now FILAMENT_UNLOAD_PURGE_RETRACT. Please update Configuration_adv.h."
"#error "FILAMENT_UNLOAD_DELAY is now FILAMENT_UNLOAD_PURGE_DELAY. Please update Configuration_adv.h."
I made both of those edits AND added the line  "#define FILAMENT_UNLOAD_PURGE_FEEDRATE      25  // (mm/s) feedrate to purge before unload" which I found in the GitHub version of Configuration_adv.h for this release.

Now my compile throws the following two errors:
"Compiling .pio\build\STM32F103RC_bigtree_USB\src\src\lcd\dogm\u8g_fontutf8.cpp.o
Marlin\src\lcd\dogm\status_screen_DOGM.cpp: In function 'void _draw_bed_status(bool)':
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:238:16: warning: unused variable 'isHeat' [-Wunused-variable]
     const bool isHeat = BED_ALT();
                ^~~~~~"
"Linking .pio\build\STM32F103RC_bigtree_USB\firmware.elf
c:/users/think/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio\build\STM32F103RC_bigtree_USB\firmware.elf section `.rodata' will not fit in region `rom'
c:/users/think/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: region `rom' overflowed by 7180 bytes     
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RC_bigtree_USB\firmware.elf] Error 1"

I am willing to keep trying, until I can test this bug for you, but I am stumped as to what to do next.
Has support for STM32F103RC_bigtree_NOUSB been renamed? or is the SKR MINI E3 v1.2 actually supported by a different environment?

sjasonsmith, 2020-01-11:
It sounds like you are missing this fix which is in 2.0.1 (NOT 2.0):

https://github.com/MarlinFirmware/Marlin/pull/16118

I’m on my phone so can’t check your configs easily, but I am guessing you have print counters enabled, which will write to the flash every time you turn on the printer. This was corrupting the flash contents prior to this fix.

Thinkersbluff, 2020-01-11:
NOTE: Sorry for "SHOUTING"...
All caps in the bug title was not my deliberate choice. I thought the form forced that on me.
 Does the form include some tagging around the placeholder text that I might have used in ignorance?  
Pls. try not to scare noobs who report bugs in good faith.  Everybody is just trying to be helpful. Some of us may accidentally irritate the more experienced.  
The rules of conduct here are wise but not always observed...

Thinkersbluff, 2020-01-11:
Exactly right, Jason.

Thank you for your quick & helpful response.
I have updated the bug thread with what I learned.

I was caught off-guard by a PlatformIO environment name change for the SKR E3 mini at 2.0.1. Hoping I guessed correctly which one to use.
I have noted two cautions thrown by the compiler, in case they need further attention.  I won’t “risk” raising a bug report on those points, since firmware.bin did compile anyway.

Reported bug now fixed.
New issue for me was that firmware.bin was not renamed after the flash, so the flash kept repeating on each power cycle until I removed the SD card.  Again, no plan to report as a BUG, but noted in case it actually is significant.

Thank you for all you do to help our community.

Cheers,
spj

#16118<https://github.com/MarlinFirmware/Marlin/pull/16118>

sjasonsmith, 2020-01-11:
@Thinkersbluff, thanks for letting us know it now works. Could you close the issue now that it is resolved?

> I have noted two cautions thrown by the compiler, in case they need further attention.  I won’t “risk” raising a bug report on those points, since firmware.bin did compile anyway.

There are always some warnings caused by some of the external libraries. I mostly recall seeing them in LCD and NeoPixel libraries. Those are safely ignored. Ideally Marlin would not generate warnings in its own code, but it can be hard to get everything right with so many possible combinations.

Thinkersbluff, 2020-01-11:
Bug fixed at 2.0.1, ref:  #16118<#16118>

closed 2020-01-11 by Thinkersbluff