MarlinFirmware/Marlin #16534: [BUG] TOUCH_UI_FTDI_EVE Error on 32 Bit boards and the newest bugfixes

TB1405, 2020-01-11:
i ordered today a 4.3" Graphical LCD Touchscreen, 480x272, SPI, FT800 from HAOYU STAR. 
i tried to comlile the marlin 2.0.x and marlin 2.0.x bugfixes fromt today (11.01.2020). 

Lets start with the issue  on the 2.0.x branch
compiling is possible with ramps and Fysetc F6. (mega2560)
but if i activate the mmu2 feature following error message comes up
is it maybe because we have to activate 5 extruders ?

`Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\screens\nudge_nozzle_screen.cpp: In static member function 'static void NudgeNozzleScreen::onRedraw(draw_mode_t)':
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\screens\nudge_nozzle_screen.cpp:76:54: error: 'getNozzleOffset_mm' was not declared in this scope
         format_position(str, getNozzleOffset_mm(X, E1), getNozzleOffset_mm(Y, E1), getNozzleOffset_mm(Z, E1));
                                                      ^
Compiling .pio\build\megaatmega2560\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\screens\screens.cpp.o
*** [.pio\build\megaatmega2560\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\screens\nudge_nozzle_screen.cpp.o] Error 1`

if i try to compile the 2.0.x branch with SKR PRO or Fysetc S6 this Error appears. (all extra features deactivatet)
https://pastebin.com/tRVjJH4e

now to the 2.0.x bugfixes branch from today.

with ramps
https://pastebin.com/F9ZbKkgw

with skr pro 
https://pastebin.com/7g7HfQtr

with skr 1.3
https://pastebin.com/uv3W4YzM

ellensp, 2020-01-11:
All those pastebins posts and not a one example config file for us to try and replicate this with...

TB1405, 2020-01-11:
> All those pastebins posts and not a one example config file for us to try and replicate this with...

i will add example files I thought it was getting too convoluted.

boelle, 2020-01-11:
config files are the first requirement, its also listed in the issue template if you read that one :-/

ellensp, 2020-01-11:
just tried Ramps1.4 + MMU2 + TOUCH_UI_FTDI_EVE (+min requirements, EXTRUDERS 5, NOZZLE_PARK_FEATURE,  LCD_HAOYU_FT800CB , OTHER_PIN_LAYOUT and ADVANCED_PAUSE_FEATURE)  and it compiled without issues. 

But BOARD_BIGTREE_SKR_V1_3, after adjusted MMU2_SERIAL compile fails with the same error 'strcat_P' was not declared in this scope. 

BIGTREE_SKR_PRO with your configs also gives same error you posted "no matching function for call to 'min(int, uint8_t&)'"  
 
The last two issue have nothing to do with MMU2  just TOUCH_UI_FTDI_EVE on those boards.

TB1405, 2020-01-11:
> just tried Ramps1.4 + MMU2 + TOUCH_UI_FTDI_EVE (+min requirements, EXTRUDERS 5, NOZZLE_PARK_FEATURE, LCD_HAOYU_FT800CB , OTHER_PIN_LAYOUT and ADVANCED_PAUSE_FEATURE) and it compiled without issues. [...]

strange it wont work on my config with ramps. With wich branch did you compile succesfully

any idea how to fix it for skr pro

thisiskeithb, 2020-01-11:
I tried with an SKR 1.3/`LPC1768` a little while ago unsuccessfully as well. Manually mapping pins still threw errors and I couldn't figure out how to get it to work after spending an evening on it.

TB1405, 2020-01-11:
i hope the filenames are not confusing 
[Confi_2.0.x_ramps_noMMU_CompSucces.zip](https://github.com/MarlinFirmware/Marlin/files/4049827/Confi_2.0.x_ramps_noMMU_CompSucces.zip)

[Confi_2.0.x_fysetcS6_NoSucces.zip](https://github.com/MarlinFirmware/Marlin/files/4049832/Confi_2.0.x_fysetcS6_NoSucces.zip)

[Confi2.0.x_ramps_withMMu2NoSucces.zip](https://github.com/MarlinFirmware/Marlin/files/4049837/Confi2.0.x_ramps_withMMu2NoSucces.zip)
following error message appear. if i deactivate the mmu2 feature compiling is succesfully again
`
Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\screens\nudge_nozzle_screen.cpp: In static member function 'static void NudgeNozzleScreen::onRedraw(draw_mode_t)': Marlin\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\screens\nudge_nozzle_screen.cpp:76:54: error: 'getNozzleOffset_mm' was not declared in this scope format_position(str, getNozzleOffset_mm(X, E1), getNozzleOffset_mm(Y, E1), getNozzleOffset_mm(Z, E1)); ^ Compiling .pio\build\megaatmega2560\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\screens\screens.cpp.o *** [.pio\build\megaatmega2560\src\src\lcd\extensible_ui\lib\ftdi_eve_touch_ui\screens\nudge_nozzle_screen.cpp.o] Error 1 `

[Confi_bugfix2.0.x_ramps_noSucces.zip](https://github.com/MarlinFirmware/Marlin/files/4049852/Confi_bugfix2.0.x_ramps_noSucces.zip)

TB1405, 2020-01-11:
> I tried with an SKR 1.3/`LPC1768` a little while ago unsuccessfully as well. Manually mapping pins still threw errors and I couldn't figure out how to get it to work after spending an evening on it.

did you received already your ftdi display ? i think i read in a post you had ordered one

thisiskeithb, 2020-01-11:
> did you received already your ftdi display ? i think i read in a post you had ordered one

I picked up a 5" FTDI EVE FT810/800x480 TFT and that's what I was trying to get working a while ago. I don't remember the specific errors, but pins were fully defined.

TB1405, 2020-01-11:
also from HAOYU? did you tried it with a arduino and ramps. With ramps compiling is succesfully

ellensp, 2020-01-12:
I managed to get SKR 1.3 and TOUCH_UI_FTDI_EVE to compile..  but can't test it.  I did have to use software SPI.

ellensp, 2020-01-12:
Took your "Confi_bugfix2.0.x_ramps_noSucces.zip" and it compiles fine with a few warning. see https://pastebin.com/ajkhzdUx

ellensp, 2020-01-12:
Confi2.0.x_ramps_withMMu2NoSucces.zip does not compile. You also enabled BABYSTEPPING. and that is causes it not to compile. This seems to be a bug in src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/screens/nudge_nozzle_screen.cpp line 75 should be "#if HOTENDS > 1" not "#if EXTRUDERS > 1"

ellensp, 2020-01-12:
regarding TOUCH_UI_FTDI_EVE on lpc1768  (eg skr 1.3).  
This is just to document what I've found so it so far
Apply PR #16540,  enable #define CLCD_USE_SOFT_SPI    (only works with sofware spi so far)
Add the following to src/HAL/HAL_LPC1768/HAL.h (I added at line 50, after includes) 
```
#define strcat_P(dest, src) strcat((dest), (src))
#define pgm_read_ptr_far pgm_read_ptr
#define pgm_read_ptr_near pgm_read_ptr
#define pgm_read_word_far pgm_read_word
```
It will now compile, but I cant test it (I don't have this display :( )

thisiskeithb, 2020-01-12:
I'll see if I can break out the screen tomorrow and test on an SKR 1.3/1.4. Thanks!

TB1405, 2020-01-12:
> Took your "Confi_bugfix2.0.x_ramps_noSucces.zip" and it compiles fine with a few warning. see https://pastebin.com/ajkhzdUx

With wich branch did you test bugfixes or stable?

> regarding TOUCH_UI_FTDI_EVE on lpc1768 (eg skr 1.3).
> This is just to document what I've found so it so far
> Apply PR #16540, enable #define CLCD_USE_SOFT_SPI (only works with sofware spi so far)
> Add [...]

I hope my Display arrives in 2 or 3 weeks. 
Maybe @thisiskeithb could say tomorrow more if its works

ellensp, 2020-01-12:
@TB1405  "With wich branch did you use test bugfixes or stable?" The wrong one... stable. sigh... I need new glasses..  
With bugfix I do get errors related to  OnPidTuning(const result_t rst)    on ramps...

ellensp, 2020-01-12:
As "punishment" for my blindness I have tracked down fixed the issue with bugfix not compiling with ftdi_eve_touch_ui.   
can someone confirm this now compiles before I do a PR?
src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/marlin_events.cpp is missing a new event OnPidTuning
replace with [marlin_events.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/4050471/marlin_events.cpp.zip)
As before I cannot test this, the logic is taken from extui_dgus_lcd.cpp and converted to ftdi_eve_touch_ui syntax.

TB1405, 2020-01-12:
@ellensp thank you :)
i made also your changes and compiling with ramps and mmu2 feature is now succesful
i added also 
`#define strcat_P(dest, src) strcat((dest), (src))
#define pgm_read_ptr_far pgm_read_ptr
#define pgm_read_ptr_near pgm_read_ptr
#define pgm_read_word_far pgm_read_word`

compiling with skr does now also works.

ellensp, 2020-01-13:
Got TOUCH_UI_FTDI_EVE to compile on BOARD_BIGTREE_SKR_PRO_V1_1  
Here is how.
You need to apply PR #16540 for this board also.
Then add the following lines to src/HAL/HAL_STM32/HAL.h
#define min(a,b) ((a)<(b)?(a):(b))
#define max(a,b) ((a)>(b)?(a):(b))

It now compiles and for both software and hardware SPI ( CLCD_USE_SOFT_SPI defined or not)

TB1405, 2020-01-13:
@ellensp 
now compiling on skr pro is also possible :) 

For fysetc S6 i got a eeprom error but is also without #define TOUCH_UI_FTDI_EVE i think that is a problem with my branch 

i tried also to compile with #define MMU2_MENUS without succes but i think that isnt supported yet in combination with FTDI_eve. 
There isnt any file in arlin/src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/screens for mmu2 menu

thisiskeithb, 2020-01-14:
Compiling & flashing an SKR 1.4 with `TOUCH_UI_FTDI_EVE` works, but I'm not getting any output on an FT810. Wiring is correct and I'm getting 5v on the correct pin, so I'm not sure why there's no output. I can't confirm that the TFT ever worked, so it could be dead.

TB1405, 2020-01-14:
@thisiskeithb maybe you could try it with a example code on a arduino and example from the eve libary. If it wont work the Display is probably defective

thisiskeithb, 2020-01-14:
It doesn't even light up with 5v applied, so it's probably DOA. Too bad I waited so long to test it because now I can't get a refund. Oh well!

TB1405, 2020-01-14:
@thisiskeithb oh thats bad. 
Did you ordered from HAOYU / HOTMCU

thisiskeithb, 2020-01-14:
> Did you ordered from HAOYU / HOTMCU

Yes

TB1405, 2020-01-14:
> > Did you ordered from HAOYU / HOTMCU
> 
> Yes

There is a only Chat for techsupport if you explaine the Situation maybe they will send one again.

boelle, 2020-01-14:
@TB1405 so still a bit left on this one before its solved?

thisiskeithb, 2020-01-14:
@boelle None of us have a working screen to test on yet.

thisiskeithb, 2020-01-14:
@marciot: Have you tried `TOUCH_UI_FTDI_EVE` on an SKR 1.3/`LPC1768`?

TB1405, 2020-01-14:
i had now ordered a FTDI EVE FT800 Display 4,3" VM800B43A-PL from germany. should arrived maybe this week. 
i hope this will work with the #define LCD_FTDI_VM800B35A and change the resolution.

ellensp, 2020-01-15:
I got TOUCH_UI_FTDI_EVE compiling on LPC1768/9 with hardware SPI to compile. If someone wants to try it. You still need to apply #16540 
and add the following to  src/HAL/HAL_LPC1768/HAL.h  (added a new last line)  
```
#define strcat_P(dest, src) strcat((dest), (src))
#define pgm_read_ptr_far pgm_read_ptr
#define pgm_read_ptr_near pgm_read_ptr
#define pgm_read_word_far pgm_read_word
#define SPI_MODE0 0x00      
```
In src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/spi.cpp
Change SPI::spi_read_bulk to the following.
```
  void SPI::spi_read_bulk (void *data, uint16_t len) {
    uint8_t* p = (uint8_t *)data;
    #ifndef CLCD_USE_SOFT_SPI
      #if EITHER(TARGET_LPC1768, TARGET_LPC1769)
        spiRead(p,len);
      #elif
        ::SPI.transfer(p, len);
      #endif 
    #else
      while (len--) *p++ = spi_recv();
    #endif
  }
```
In regards to SPI_MODE0 it is only referenced once in src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/spi.cpp 
"SPISettings SPI::spi_settings(SPI_FREQUENCY, MSBFIRST, SPI_MODE0);"
The HAL_LPC1768  originally defines SPI_MODE3 but not SPI_MODE0, so perhaps, the following would be a better option. 
"SPISettings SPI::spi_settings(SPI_FREQUENCY, MSBFIRST, SPI_MODE3);"  
Try both SPI MODES perhaps.

TB1405, 2020-01-17:
today arrived my FTDI EVE FT800 Display 4,3" VM800B43A
i checked it with a arduino uno and test sketch the display does work. 

i tried it with enabled `#define LCD_FTDI_VM800B35A`  and `LCD_HAOYU_FT800CB` both uses the FT800 chip but it wont turn on. 

i tried it with S6 and SKR 1.3 
for S6 i also used SW SPI because i didnt know witch spi marlin would use. S6 board supports 3. 
`  #define OTHER_PIN_LAYOUT  // Define pins manually below
  #if ENABLED(OTHER_PIN_LAYOUT)
    // The pins for CS and MOD_RESET (PD) must be chosen.
    #define CLCD_MOD_RESET  PC9
    #define CLCD_SPI_CS    PA8

    // If using software SPI, specify pins for SCLK, MOSI, MISO
    #define CLCD_USE_SOFT_SPI
    #if ENABLED(CLCD_USE_SOFT_SPI)
      #define CLCD_SOFT_SPI_MOSI PD2
      #define CLCD_SOFT_SPI_MISO PC12
      #define CLCD_SOFT_SPI_SCLK PD1
    #endif
  #endif` 

tomorrow I will check if it is because the VM800B35A is a 3.5 inch tft. mine is 4,3 inch.
But during the Arduino test sketch the display still turned on when I selected the VM800B35A. Only the format was wrong. 

@thisiskeithb maybe your display isnt doa. if i connect mine to 5v and ground there is nothing light up. only if the control is successful it goes on

thisiskeithb, 2020-01-17:
> maybe your display isnt doa. if i connect mine to 5v and ground there is nothing light up. only if the control is successful it goes on

I had *hoped* that was the case, but hadn't looked into it since I got sucked into another project. I'll keep watching this issue and help out when I can.

TB1405, 2020-01-18:
> I got TOUCH_UI_FTDI_EVE compiling on LPC1768/9 with hardware SPI to compile. If someone wants to try it. You still need to apply #16540
> and add the following to src/HAL/HAL_LPC1768/HAL.h (added a new last line)
> 
> ```
> #define strcat_P(dest, src) strcat((dest), (src))
> #define pgm_read_ptr_far pgm_read_ptr
> #define pgm_read_ptr_near pgm_read_ptr
> #define pgm_read_word_far pgm_read_word
> #define SPI_MODE0 0x00      
> ```
> 
> In src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/spi.cpp
> Change SPI::spi_read_bulk to the following.
> 
> ```
>   void SPI::spi_read_bulk (void *data, uint16_t len) {
>     uint8_t* p = (uint8_t *)data;
>     #ifndef CLCD_USE_SOFT_SPI
>       #ifdef TARGET_LPC1768  //also works for LPC1769
>         spiRead(p,len);
>       #elif
>         ::SPI.transfer(p, len);
>       #endif 
>     #else
>       while (len--) *p++ = spi_recv();
>     #endif
>   }
> ```
> 
> In regards to SPI_MODE0 it is only referenced once in src/lcd/extensible_ui/lib/ftdi_eve_touch_ui/ftdi_eve_lib/basic/spi.cpp
> "SPISettings SPI::spi_settings(SPI_FREQUENCY, MSBFIRST, SPI_MODE0);"
> The HAL_LPC1768 originally defines SPI_MODE3 but not SPI_MODE0, so perhaps, the following would be a better option.
> "SPISettings SPI::spi_settings(SPI_FREQUENCY, MSBFIRST, SPI_MODE3);"
> Try both SPI MODES perhaps.

why SPI Mode 3 i think the LPC only have SPI_0 and SPI_1 SPI_1 is used for onboard i think so. 
if i change SPI_MODE3  to SPI_MODE0 there comes a error that SPI_MODE3  isnt defined

TB1405, 2020-01-18:
alright i was able to repair my old arduino mega. 
the display does now turn on. i had correct the resolution from 320x240 to 480x272 but this only works if i change the resolution in the board.h file. (Change in config.adv doesnt change anything maybe a bug?) 
But problem is now i got the text message "Release to begin screen calibration" if i disconnect the MISO pin if the MISO Pin is connected there is while booting only a shiny green backround with green rectangle and after this the display is black but backlight still on

TB1405, 2020-01-18:
i dont if this a marlin problem or it isnt enough to set the correct resolution for the VM800B43A Display.
but the "Release to begin screen calibration" is shown correct so it should works or not ?
im done for today.

TB1405, 2020-01-19:
i got the display to work. But only with the marlin version from  FYSETC https://github.com/FYSETC/Marlin-2.0.x-FYSETC/tree/S6/BUG-FIX-CLCD?fbclid=IwAR0WQ6GFOD8_tBAQGZei5TsJ3Nuijf2q-uqrg6zIuC3CLbtIZ1uxGDJXtdE and it only works if i use the Fysetc S6 i tried also with arduino mega but then the display shows only black screen with backlight on. 
so i think there are still some bugs in the TOUCH_UI from marlin. 
but now iam know the VM800B43A is compatbible.

TB1405, 2020-01-20:
i close this i opened a new issue with a more detailed description i think this wasnt only a compiling issue. 
new issue https://github.com/MarlinFirmware/Marlin/issues/16628

closed 2020-01-20 by TB1405