MarlinFirmware/Marlin #16535: [BUG] Motors stuttering on SKR Mini E3 v1.2

chand1012, 2020-01-11:
### Bug Description

I just upgraded my Ender 3 to use a 32-bit SKR Mini E3 v1.2 and everything looked fine on first boot, when homing, and when leveling the bed. As soon as the print started, the motors started to stutter with every movement. This has something to do with how the Z-offset is calculated during a print, as it didn't happen when the bed was manually leveled or when I was using the `AUTO_BED_LEVELING_LINEAR` instead of my preferred `AUTO_BED_LEVELING_BILINEAR`.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4049448/Configuration.zip)

### Steps to Reproduce

1. Have a printer with an SKR Mini E3 v1.2
2. Download and run Marlin with the config linked
3. Run a print with the bed autoleveling before the print.

**Expected behavior:** Smooth movement on the X and Y axes.

**Actual behavior:** Stuttering when moving the X and Y axes.

#### Additional Information

* [SKR Mini E3 v1.2](https://www.amazon.com/BIQU-Control-Integrated-TMC2209-Printer/dp/B07YW86YS7)
* Video of the issue [here](https://imgur.com/a/6G3iFgH).
* More config info can be found [here](https://github.com/chand1012/Marlin-Ender3-BLTOUCHv3).
* I was able to fix the issue by switching from `AUTO_BED_LEVELING_BILINEAR` to `AUTO_BED_LEVELING_LINEAR`. I would prefer bilinear as when I am not using a glass bed the surface can be a little uneven.

Lord-Quake, 2020-01-11:
Are you running the latest version of marlin?
As of this post it's v2.0.1 : https://github.com/MarlinFirmware/Marlin/releases

chand1012, 2020-01-11:
> Are you running the latest version of marlin?
> As of this post it's v2.0.1 : https://github.com/MarlinFirmware/Marlin/releases

Actually I have been basing my config off of the [bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) branch.

Lord-Quake, 2020-01-11:
OK, then my question would be if you have the same problem with the official version 2.0.1?

boelle, 2020-01-11:
> Are you running the latest version of marlin?
> As of this post it's v2.0.1 :

no that is not the case.... bugfix 2.0.x is

2.0.1 is a frozen copy of bugfix 2.0.x, just look at when each was last updated

Lord-Quake, 2020-01-11:
@boelle OK, understand... I was always under the impression that the Marlin base code for download is under "Latest release" and considered stable. Obviously bugs found in v2.0.1 are corrected in bugfix v2.0.x. 
BTW, in my opinion it would make no sense then to have a link to v2.0.1 on "Latest release" if the bugfix version is to be used as latest version for download.

Anyway, since I have the same setup as the author and I'm not having the issues with v2.0.1 I would like to help the person by analyzing the problem under the same conditions.

chand1012, 2020-01-11:
Strange. v2.0.1 seems to fix most of the problem, but there is still a bit of stuttering on fast moves. Is there any way to mitigate that?

Lord-Quake, 2020-01-11:
Post both configs and I will look.

boelle, 2020-01-11:
> BTW, in my opinion it would make no sense then to have a link to v2.0.1 on "Latest release" if the bugfix version is to be used as latest version for download.

that you have to take up with our maintainer @thinkyhead 

i never looked at what is "release" but rather what is updated the most and most recently

chand1012, 2020-01-12:
I copied the config from above and used it for Marlin 2.0.1. Here it is again if needed.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4050085/Configuration.zip)

Lord-Quake, 2020-01-12:
@chand1012  In my opinion that isn't good practice. I would always use the configs that come with the FW version that is being used. Other FW configs versions might have different default parameters set or additional/removed settings. Please use the configs that came with FW v2.0.1., see if you still have a problem and if so post your FW v2.0.1 configs here.

boelle, 2020-01-18:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-01-18 by boelle