MarlinFirmware/Marlin #16541: [BUG] UBL incorrectly mapping nozzle location to matrix; M421 updates wrong entry

GarthSnyder, 2020-01-12:
I'm using Marlin 2.0.1 compiled from the repo. Configuration files are attached. The hardware is a frankenprinter that started as an Evnovo Sidewinder X1 V4 but now has a BLTouch and Bigtree SKR V1.3 mainboard.

I'm trying to convert to UBL, but I have no prior experience with it. I successfully tested out some smaller matrix sizes, and now I want to generate a full 15x15 bed map. The UBL and probe margins are both 2. The bed size is 306 x 306. UBL correctly leaves two rows and two columns unfilled after running G29 P1.

I don't have a UBL-compliant display and controller, so I want to fill in some of the missing entries by hand with M421, as mentioned in the UBL top-level doc. But neither G29 nor M421 seems to have a clear idea of the mapping between bed coordinates and matrix entries.

In the transcript below, the nozzle moves to X2 Y304, which is the upper left coordinate of the matrix as displayed by G29 T. That is, it should correspond to row 14, column 0. However, after the move, G29 T shows row 12, column 1 as being the closest matrix entry. I could understand being one off, but this is three orthogonal moves off (like a knight's move in chess). So that's issue 1; not sure if it's just a display bug or is representative of deeper UBL state.

Then, I run M421 C Z0.08 to set the Z offset for the closest matrix entry to 0.08mm. That does set a matrix entry to 0.08, but it's the entry at row 12, column 2. Wrong, and not even in agreement with the current entry marked by G29.

The stride for this matrix should be a little over 20mm. If you try other coordinates, both G29 T and M421 cell selections seem to be consistently wrong, but G29 T seems to be an offset, whereas I get the impression M421 is trying to limit changes to entries that already contain values.

Please let me know if there's any additional info I can provide. This seems to be repeatable.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4050190/Configuration.zip)

```
Send: G0 X150 Y150 Z1
Recv: ok
[...]
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  2,304)                                                                                                      (304,304)
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv: 14 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 13 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
Recv:    |
Recv: 12 |   .       .     +0.106  +0.077  +0.067  +0.036  +0.022  -0.015  -0.025  -0.003  +0.032  +0.011  -0.002  -0.000  -0.034
Recv:    |
Recv: 11 |   .       .     +0.108  +0.098  +0.073  +0.052  +0.046  +0.010  +0.013  +0.042  +0.061  +0.052  +0.045  +0.060  +0.022
Recv:    |
Recv: 10 |   .       .     +0.022  +0.017  +0.020  +0.011  +0.020  -0.014  +0.001  +0.041  +0.069  +0.073  +0.090  +0.116  +0.098
Recv:    |
Recv:  9 |   .       .     +0.032  +0.030  +0.026  -0.002  +0.018  -0.012  -0.002  +0.038  +0.068  +0.056  +0.057  +0.080  +0.046
Recv:    |
Recv:  8 |   .       .     +0.057  +0.038  +0.036  +0.020  +0.026  -0.002  -0.003  +0.026  +0.052  +0.046  +0.046  +0.083  +0.042
Recv:    |
Recv:  7 |   .       .     +0.011  -0.007  +0.013  +0.011  +0.026  +0.005  +0.023  +0.073  +0.108  +0.121  +0.140  +0.168  +0.151
Recv:    |
Recv:  6 |   .       .     +0.059  +0.046  +0.048  +0.036  +0.035  +0.010  +0.016  +0.048  +0.073  +0.086  +0.096  +0.115  +0.083
Recv:    |
Recv:  5 |   .       .     +0.101  +0.077  +0.048  +0.026  +0.028  -0.002 [-0.028] +0.010  +0.030  +0.017  +0.011  +0.038  -0.015
Recv:    |
Recv:  4 |   .       .     +0.022  +0.007  +0.005  -0.004  +0.001  -0.034  -0.029  +0.022  +0.046  +0.046  +0.038  +0.077  +0.056
Recv:    |
Recv:  3 |   .       .     +0.065  +0.038  +0.048  +0.030  +0.020  +0.013  +0.007  +0.036  +0.061  +0.060  +0.052  +0.083  +0.073
Recv:    |
Recv:  2 |   .       .     +0.140  +0.096  +0.077  +0.046  +0.030  -0.018  -0.030  +0.001  +0.006  -0.005  -0.028  -0.028  -0.047
Recv:    |
Recv:  1 |   .       .     +0.058  +0.022  +0.001  +0.001  +0.007  -0.024  -0.030  +0.011  +0.036  +0.026  +0.030  +0.041  -0.000
Recv:    |
Recv:  0 |   .       .     +0.058  +0.038  +0.035  +0.013  +0.026  -0.005  -0.012  +0.021  +0.048  +0.042  +0.038  +0.055  +0.017
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv:     (  2,  2)                                                                                                      (304,  2)
Recv: 
Recv: ok
[...]
Send: G0 X2 Y304 Z1
Recv: ok
[...]
Send: G29 T
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  2,304)                                                                                                      (304,304)
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv: 14 |   .       .       .       .       .       .       .       .       .       .       .       .       .       .       .
[... 51 lines cut ...]
Recv:    |
Recv:  9 |   .       .     +0.032  +0.030  +0.026  -0.002  +0.018  -0.012  -0.002  +0.038  +0.068  +0.056  +0.057  +0.080  +0.046
Recv:    |
Recv:  8 |   .       .     +0.057  +0.038  +0.036  +0.020  +0.026  -0.002  -0.003  +0.026  +0.052  +0.046  +0.046  +0.083  +0.042
Recv:    |
Recv:  7 |   .       .     +0.011  -0.007  +0.013  +0.011  +0.026  +0.005  +0.023  +0.073  +0.108  +0.121  +0.140  +0.168  +0.151
Recv:    |
Recv:  6 |   .       .     +0.059  +0.046  +0.048  +0.036  +0.035  +0.010  +0.016  +0.048  +0.073  +0.086  +0.096  +0.115  +0.083
Recv:    |
Recv:  5 |   .       .     +0.101  +0.077  +0.048  +0.026  +0.028  -0.002  -0.028  +0.010  +0.030  +0.017  +0.011  +0.038  -0.015
Recv:    |
Recv:  4 |   .       .     +0.022  +0.007  +0.005  -0.004  +0.001  -0.034  -0.029  +0.022  +0.046  +0.046  +0.038  +0.077  +0.056
Recv:    |
Recv:  3 |   .       .     +0.065  +0.038  +0.048  +0.030  +0.020  +0.013  +0.007  +0.036  +0.061  +0.060  +0.052  +0.083  +0.073
Recv:    |
Recv:  2 |   .       .     +0.140  +0.096  +0.077  +0.046  +0.030  -0.018  -0.030  +0.001  +0.006  -0.005  -0.028  -0.028  -0.047
Recv:    |
Recv:  1 |   .       .     +0.058  +0.022  +0.001  +0.001  +0.007  -0.024  -0.030  +0.011  +0.036  +0.026  +0.030  +0.041  -0.000
Recv:    |
Recv:  0 |   .       .     +0.058  +0.038  +0.035  +0.013  +0.026  -0.005  -0.012  +0.021  +0.048  +0.042  +0.038  +0.055  +0.017
Recv:         0       1       2       3       4       5       6       7       8       9      10      11      12      13      14
Recv:     (  2,  2)                                                                                                      (304,  2)
Recv: 
Recv: ok
```

Vertabreak, 2020-01-12:
G29 P3 smart fill not good enough?

GarthSnyder, 2020-01-12:
> G29 P3 smart fill not good enough?

The bed's kind of wavy:

[image]

G29 P3 interpolation gives a value of 0.165 for that upper left corner, so that's 0.085 off. Not optimal considering that it's nearly half the layer height. Not P3's fault, though - that corner is 2 cells away from any real data, and the topography isn't smooth.

GarthSnyder, 2020-01-12:
If you fill out the matrix with P3, M421 starts working correctly but G29 T still highlights the wrong cell. So it looks like one possible workaround is to P3 everything, adjust the key cells, delete all the previously-interpolated values, and rerun P3.

rado79, 2020-01-12:
I guess sometimes that the UBL z-correction has an x ​​/ y offset to my bed during printing. 
as if the mesh had shifted. on higher fist layer it is hardly recognizable, if first layer is low (0.12-0.15)
it looks like the z-correction is not working properly
SKR 1.3, corexy, UBL 10x10, checked nozzle to probe offset

boelle, 2020-01-18:
@GarthSnyder is this also a problem when using bugfix 2.0.x? that one is newer than 2.0.1

GarthSnyder, 2020-01-18:
I will check and report back.

Roxy-3D, 2020-01-19:
I don't have any way to check out the nozzle mapping right now.  (I'm out of town and away from my printers)     

But it should be fairly easy to check (and verify) the M421 mesh corrections.    What ever the value is at a location that needs to be corrected...   You can do a M421 Ix Jy Zz and set the Z to some obvious and known value.  When you do a G29 T, it should be reflected in the correct location.

Roxy-3D, 2020-01-19:
@rado79  

> I guess sometimes that the UBL z-correction has an x ​​/ y offset to my bed during printing.
as if the mesh had shifted. on higher fist layer it is hardly recognizable, if first layer is low (0.12-0.15)
it looks like the z-correction is not working properly

One easy check is doing this:
- load a mesh
- activate UBL
- move the nozzle to a Z Height of .5mm
- move the nozzle around the bed to different places
- Do you see the Z Axis correcting as the nozzle moves around?

GarthSnyder, 2020-01-19:
> @Roxy-3D : ...it should be fairly easy to check (and verify) the M421 mesh corrections. What ever the value is at a location that needs to be corrected... You can do a M421 Ix Jy Zz and set the Z to some obvious and known value. When you do a G29 T, it should be reflected in the correct location.

This form of M421 is explicitly documented _not_ to work with UBL. Is the [documentation](http://marlinfw.org/docs/gcode/M421.html) incorrect?

Roxy-3D, 2020-01-19:
> This form of M421 is explicitly documented not to work with UBL. Is the documentation incorrect?

Perhaps the documentation could be clearer.   But I think you are mis-reading the UBL comments.  In the code...   UBL does support both the (I&J) form as well as (X&Y) form to specify mesh coordinates.   

Please give M421 a try and see if it works as expected.
```
#elif ENABLED(AUTO_BED_LEVELING_UBL)

  /**
   * M421: Set a single Mesh Bed Leveling Z coordinate
   *
   * Usage:
   *   M421 I<xindex> J<yindex> Z<linear>
   *   M421 I<xindex> J<yindex> Q<offset>
   *   M421 I<xindex> J<yindex> N
   *   M421 C Z<linear>
   *   M421 C Q<offset>
   */
  inline void gcode_M421() {
  ```

GarthSnyder, 2020-01-19:
> But I think you are mis-reading the UBL comments.

Ah, right you are! Thanks.

rado79, 2020-01-19:
@Roxy-3D  the Z Axis does correcting when the nozzle moves around, but sometimes it feels like there is an offset in XY direction from UBL mesh to bed. hard to repoduce for me.
I rarely use automatic leveling, have a nice flat bed (330x330), but sometimes i try UBL and my results are worse. 
I didn't study ABL/UBL because I don't really need it. Maybe my problem depends on a twistet x axis.

boelle, 2020-01-23:
@GarthSnyder i assume your issue it not solved yet?

boelle, 2020-01-26:
will close this as it was a misunderstanding of commments

> > But I think you are mis-reading the UBL comments.
> 
> Ah, right you are! Thanks.

GarthSnyder, 2020-01-26:
No, I think this bug is still at large, although there may be internal implementation connections that I'm not aware of. Let me summarize:

* M421 C Z sets the value of the wrong matrix location. This is the gravamen of the bug report and is still true as far as I can tell.

* G29 T also shows the wrong current location. However, I suspect that it might be showing the probe location rather than the nozzle location. That would make sense, I just need to verify that.

* @Roxy-3D commented that "it should be fairly easy to check (and verify) the M421 mesh corrections. What ever the value is at a location that needs to be corrected... You can do a M421 Ix Jy Zz and set the Z to some obvious and known value. When you do a G29 T, it should be reflected in the correct location." 

* I didn't realize that the M421 I J Z syntax applied to UBL and I asked about this. And just to confirm, M421 I J Z does work with UBL and works correctly. However, although this is a useful workaround, I don't see any direct relevance to the fact that M421 C Z does not work; there is still a problem here.

* Current status is that we're waiting for me to verify that 1) bug behavior is the same in bugfix-2.0.x, and that 2) G29 T is actually showing the probe location but is showing that accurately.

boelle, 2020-06-20:
@GarthSnyder been out for a while so asking if this one is still at large?

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle