MarlinFirmware/Marlin #16543: [FR] Auto-re-init SD card

richfelker, 2020-01-12:
Requirement to explicitly release/init SD card each time it's removed and inserted is tedious and unnecessary if the card is being used read-only. For boards with the SD-CD pin connected, the "print from SD card" function could automatically re-read the partition table and FAT iff the media has been changed since last use of the menu. For boards without it connected, it could automatically re-read on each use (same as manually clicking twice on release/init card on the menu).

richfelker, 2020-06-24:
Ping.

thinkyhead, 2020-06-24:
The SD card must be released from the menu on setups that have a shared SD card. For a setup which has no SD detect pin, I suppose we could periodically look to see if the SD card is there by attempting to read it. However, that is a blocking operation, and it is likely to cause periodic pauses in the UI and possibly pauses in printing.

richfelker, 2020-06-24:
For the case with no SD detect pin, you don't need to probe while printing since either (1) the print is from SD card, and removing the card would be a hard error, or (2) the print is over serial interface, in which case the SD card is not involved. Just probing while idle should be fine.

But if some/most board have card detect pins, it probably suffices to offer the proposed feature just for boards that do.

sjasonsmith, 2020-09-11:
@richfelker this already exists if you have an SD_DETECT pin. Are you seeing problems suggesting it does not work?

As @thinkyhead mentioned, you will not see this on boards which can expose the SD over USB (such as LPC1768 boards), because it would interfere with host operations.

richfelker, 2020-09-12:
OK, it looks like my board (Creality Melzi) just doesn't have the pin, or at least Marlin doesn't have it configured to be used.

sjasonsmith, 2020-09-12:
I just looked at a Creality Melzi I have around, and I do not see any traces connected to the needed pin.
Based on the discussion here it seems like this is already implemented to the extent desired. It is unfortunate that these Melzi boards don't have the necessary pins to support the feature.

closed 2020-09-12 by sjasonsmith