MarlinFirmware/Marlin #16545: [FR] Add little extrusion prior to retraction when changing filament

BastR, 2020-01-12:
### Description

I use an MMU2 clone with Marlin 2.0 and the filament removal procedure associated with the MMU2 allows me to remove the filament dozens of times without problems. On the other hand, when I use the filament change without having activated the MMU2, I often have jams because of the bad shape of the tip (often thickened). 
I think it would be good to do a small extrusion before starting the filament removal procedure in order to have clean tips just like Marlin does with the MMU2.

Maybe adding this would be enough:

```
 #define FILAMENT_CHANGE_INITIAL_EXTRUSION_FEEDRATE     50  // (mm/s) Extrude filament to make good tips. This can be pretty fast.
 #define FILAMENT_CHANGE_INITIAL_EXTRUSION_LENGTH       2  // (mm) The length of filament for a complete unload.

```

1. M702 or M600
2. Park extruder
3. Do little extrusion too make good tips
4. Retract the filament from the extruder
5. Insert new filament

yet-another-average-joe, 2020-01-12:
This is funny, I was about to ask for the same feature, when I logged to GitHub today, and I first saw your FQ   !

I had problems when adding some retraction after the print ends (end-gcode), and the filament cools down : jams. I gave up with end-gcode retraction, and now let the filament cool down in the hotend. Before every print, I preheat the hotend, extrude 10 mm, then unload the filament, and load another filament.

Extruding before loading really makes a difference : as you wrote, the filament bit is much nicer, and no more jams (orginial E3D V6 full metal + original 450mm Capricorn + TriangleLabs BMG extruder)
Before I did that, I had problems with the frilament tip, too large, and not going through the extruder (really a pain !)

Pre-extruding softens the filament bit in the empty nozzle (compensates oozing), when retracting, it will be stretched, and this makes things much smoother and easier !

This is VERY easy to code. At the begining of unload_filament() function, in source/feature/pause.cpp, just add these 2 lines :

`
  do_pause_e_move((FILAMENT_CHANGE_INITIAL_EXTRUSION_LENGTH) * mix_multiplier, (ADVANCED_PAUSE_PURGE_FEEDRATE) * mix_multiplier);
  safe_delay(FILAMENT_CHANGE_INITIAL_EXTRUSION_DELAY);
`
The delay after pre-extrusion is there to let the pressure release in the Bowden tube.
As you can see, I use the load purge feedrate (no real reason to use a different value)

In Configuration_adv.h, also add :

`
#define FILAMENT_CHANGE_INITIAL_EXTRUSION_DELAY        5000
`

One picture that shows the improvements :

- 1 default unload funtion
- 2 pre-extrusion, no delay before unloading (immediate retraction + ram purge)
- 3 pre-extrusion, 5 sec delay before ram purge and normal unloading

3 is much better and goes smoothly throught the BMG (with the "type 2" bits, I had to pull with some effort)

[image]

BastR, 2020-01-14:
@yet-another-average-joe wow, thank you for your answer. You've already done all the work.

I've never done it, but I'm going to try to push a PR for it with your help.

Thanks you

yet-another-average-joe, 2020-01-14:
Great ! 

(I have problems with PRs from Git app, and didn't really investigate)

BastR, 2020-01-19:
@yet-another-average-joe I followed your advice and I looked at the code in the unload() function, and I think that finally I don't need FR, I explain myself:

```
// Retract filament
  do_pause_e_move(-(FILAMENT_UNLOAD_PURGE_RETRACT) * mix_multiplier, (PAUSE_PARK_RETRACT_FEEDRATE) * mix_multiplier);

  // Wait for filament to cool
  safe_delay(FILAMENT_UNLOAD_PURGE_DELAY);

  // Quickly purge
  do_pause_e_move((FILAMENT_UNLOAD_PURGE_RETRACT + FILAMENT_UNLOAD_PURGE_LENGTH) * mix_multiplier,
                  (FILAMENT_UNLOAD_PURGE_FEEDRATE) * mix_multiplier);

  // Unload filament
  #if FILAMENT_CHANGE_UNLOAD_ACCEL > 0
    const float saved_acceleration = planner.settings.retract_acceleration;
    planner.settings.retract_acceleration = FILAMENT_CHANGE_UNLOAD_ACCEL;
  #endif

  do_pause_e_move(unload_length * mix_multiplier, (FILAMENT_CHANGE_UNLOAD_FEEDRATE) * mix_multiplier);
```

I think that what I'm asking for is already present in the code, a small retraction is made, the extruder comes in its parking position, then it performs a purge extrusion and finally retracts the whole filament.

For me it's similar to the ramming sequence from the MMU2 but without the cooling moves:

```
  #define MMU2_RAMMING_SEQUENCE \
    {   1.0, 1000 }, \
    {   1.0, 1500 }, \
    {   2.0, 2000 }, \
    {   1.5, 3000 }, \
    {   2.5, 4000 }, \
    { -15.0, 3600 }, \
    { -32.9, 1200 }, \
    {  -9.4,  600 }, \
    {  -4.7,  360 }, \
    {  10.0,  1059 }, \
    { -10.0,  1088 }, \
    {  10.0,  1116 }, \
    { -10.0,  1144 }, \
    {  10.0,  1172 }, \
    { -10.0,  1201 }, \
    { -30.3, 2000 }
```

I will try to adapt the ramming sequence to make FILAMENT_UNLOAD_PURGE_LENGTH longer and have better tips. I'm closing the FR for now and I'll reopen it if the results are not good. 

Thanks you for your help

closed 2020-01-19 by BastR

yet-another-average-joe, 2020-01-19:
You are describing the "ram purge". It is **very** different !

- retraction
- cooling : the filament cools down and the bit gets hard
- fast extrusion using hardened filament, as fast as possible, with maximum acceleration = ram purge
- unloading

see https://github.com/MarlinFirmware/Marlin/issues/10873
and (post 8) : https://forum.e3d-online.com/threads/edge-and-scaffold-not-printing-correctly.2736/