MarlinFirmware/Marlin #16546: [BUG] #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."

kramer04, 2020-01-12:
Bug Description
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL.cpp:30:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:2070:4: error: #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
   #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
    ^~~~~
Compiling .pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/msc_sd.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/onboard_sd.cpp.o
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:2070:4: error: #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
   #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
    ^~~~~
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/HAL.cpp.o] Error 1
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/Servo.cpp:26:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:2070:4: error: #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
   #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
    ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp:29:
Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/SanityCheck.h:2070:4: error: #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
   #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
    ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_swspi.cpp:21:
Marlin/src/HAL/HAL_STM32F1/dogm/../../../inc/SanityCheck.h:2070:4: error: #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
   #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
    ^~~~~
In file included from Marlin/src/HAL/HAL_STM32F1/../../inc/MarlinConfig.h:38:0,
                 from Marlin/src/HAL/HAL_STM32F1/onboard_sd.cpp:15:
Marlin/src/HAL/HAL_STM32F1/../../inc/SanityCheck.h:2070:4: error: #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
   #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
    ^~~~~
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/HAL_SPI.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_fsmc.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/Servo.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/dogm/u8g_com_stm32duino_swspi.cpp.o] Error 1
*** [.pio/build/STM32F103RC_bigtree_512K/src/src/HAL/HAL_STM32F1/onboard_sd.cpp.o] Error 1
when compiling with VSCODE + platformio the last Marlin bugfix-2.0.x

### My Configurations
Ender-3 + SKR-MINI-E3 V1.2 with **TMC2209** + TOUCHMI

### Steps to Reproduce
uncomment #define ENDSTOP_INTERRUPTS_FEATURE

First i see in platformio TMC driver is define to TMCStepper@>=0.5.2,<1.0.0
but previous Marlin has TMCStepper@>=0.6.1,<1.0.0.
The last driver are bugged ?

sjasonsmith, 2020-01-12:
Try it with the bugfix-2.0.x. I think I already removed this sanity check for your board, but I can’t quite remember.

kramer04, 2020-01-12:
I'm using the last Marlin bug fix-2.0.x
and do you know why in previous bugfix-2.0.x it's TMCStepper@>=0.6.1,<1.0.0 and now TMCStepper@>=0.5.2,<1.0.0 ?

sjasonsmith, 2020-01-12:
> do you know why in previous bugfix-2.0.x it's TMCStepper@>=0.6.1,<1.0.0 and now TMCStepper@>=0.5.2,<1.0.0 ?

Yes, that is because you are not using the latest bugfix-2.0.x.
This is the line from bugfix-2.0.x:

`TMCStepper@>=0.6.1,<1.0.0`
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/platformio.ini#L29

kramer04, 2020-01-12:
> Yes, that is because you are not using the latest bugfix-2.0.x.
> This is the line from bugfix-2.0.x:
>
> `TMCStepper@>=0.6.1,<1.0.0` [...]

ok my mistake, i close it and will try tomorrow with last marlin bugfix-2.0.x

closed 2020-01-12 by kramer04