MarlinFirmware/Marlin #16547: E0 y E1 extruder dont work in Marlin 2.0.x bugfix SKR 1.4 Turbo

mjaureguiar, 2020-01-12:
Hi

I have an SKR 1.4 Turbo. I've installed TMC2208 driver (UART mode) in X, Y, Z axis (this works) and i have tested A4988 (motor is stuttlering and dont change direction), DRV8825 driver (motor doesn't work) and TMC2208 driver (UART mode works but doesn't with bmg bondtech 3:1 - put in spreadCycle) in E0. Linear Advance is disabled in all tests.

i have changed the following parameters SQUARE_WAVE_STEPPING (Enable/Disabled), MINIMUM_STEPPER_PULSE (Disabled/2), MINIMUM_STEPPER_POST_DIR_DELAY (Disabled/650), MINIMUM_STEPPER_PRE_DIR_DELAY (Disabled/650) all of this didn't work.

I already tested motor E0 in X axis TMC2208 Driver (it works) also A4988 driver in Z axis (it also works).
[MarlinMiguel.zip](https://github.com/MarlinFirmware/Marlin/files/4050715/MarlinMiguel.zip)

I thought that the problem was the E0 pins, so i changed them with E1 pins. This didn't work so the problem continues. 
I even changed the wire but nothing happened.

I use Visual Code Studio with Platformio Core 4.0.3 (Platformio Core 4.1 showed communication terminal problems with USB port, It acted like an USB storage mass)
Attached you will find my configuration file.

I'd really appreciated your help 0n this.
Thanks in advanced, looking forward for your orientation
Miguel

### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Put A4988 driver in E0 
2. Put DRV8825 driver in E0
3. Put TMC2208 UART in E0

**Expected behavior:** It supposed to work.

**Actual behavior:** It doesn't work.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

nalimcos, 2020-01-12:
In my experience, stuttering and single-direction movement happen with mis-wired steppers. Cables are made to different standards and sometimes the center two wires need to be swapped. Although, different wiring standards on the same board would be a bit strange.

Also, try disabling EEPROM_SETTINGS or sending M502 before running your tests, you may have an old incorrect setting overriding what you set in configuration.h.

mjaureguiar, 2020-01-12:
I tried to invert the cables and it didn't work with the a4988 driver.
Thanks

mjaureguiar, 2020-01-12:
[image]
In channel 1 is step from microcontroller and channel 2 is direction. Driver A4988.
[image]
In channel 1 is step from microcontroller. TMC2808.

why is it so different?

sjasonsmith, 2020-01-12:
Your scales are totally different, so it isn't a very fair comparison.
You are looking at pulses on the 4988, and square wave stepping on the TMC2208, where each edge counts as a step.

If you disable square wave stepping, you would probably see that the 2208 pulses are shorter than the 4988 pulses.

Marlin will always use the same pulse length for all your steppers, so if you have one 4988 installed it will coerce the TMC pulse length to match. If you want to examine pulse length for different drivers make sure you configure them all for the same type.

boelle, 2020-01-12:
So this is kind of a mis understanding on how things work?

sjasonsmith, 2020-01-12:
> So this is kind of a mis understanding on how things work?

Only for the scope images. I have no idea what is actually causing the extruders not to work.

ellensp, 2020-01-12:
A A4988 only requires 3 digital signals, you may as well check the enable pin in low, and check the vmot pins have power. But from what I see so far the issue is on output of stepper driver. Either bad wiring or insufficient Vref.  You are plugging the stepper motor into the correct output plug?  there are two Z plugs in the middle...

mjaureguiar, 2020-01-13:
It is true that they are not the same scales, but when you place it, not even the wave happened that went to a4988. As for what ellensp says I corroborated it and it's fine. Thanks

sjasonsmith, 2020-01-13:
> It is true that they are not the same scales, but when you place it, not even the wave happened that went to a4988. As for what ellensp says I corroborated it and it's fine. Thanks

I don’t understand what you are saying. There shouldn’t be a “wave” to a 4988 driver, it will be a series of pulses. Each pulse is usually about 1-3 _microseconds_, and you are trying to see those on a 90 _millisecond_ scale. That is why it looks so strange. You cannot see the pulses properly at your current scope resolution.

You see a wave for the 2208, because it is using square wave stepping. Each pulse is replaced by a single edge, generating a sequence that looks more like a square wave.

ManuelMcLure, 2020-01-13:
“Motor is stuttering and won’t change direction” is very likely a wiring problem between the driver and the motor. If any of the wires is broken or making intermittent contact that’s exactly the symptoms that would show up. The extruder wires on direct drive printers are especially vulnerable to this because the X carriage motion flexes the cables constantly so metal fatigue sets in.

horndogut2k3, 2020-01-14:
I have the same problem with the same 1.4 turbo board and Tmc2209. I put the 1.4 none turbo in, it's as good as gold. This is not a wiring problem, it's a firmware problem. The only other drivers I have are the tmc5160. After I recover from tinkering burn out I might try them out. I don't see much of a hurry just to upgrade the identical board except for 20 extra mhz processor and buggy firmware.

dabivs, 2020-01-17:
I have this exact issue on two SKR 1.4 non-turbo boards (I'm praying that it works with a turbo board though I'm not optimistic.) I'm running TMC 2208 V3.0 UART's across the board with Anet A8 stock motors (also tried with a new motor to see if I could get a different result) and I get the same issue. The motor is loaded (can't spin) but barely moves when I try to extrude 10mm of filament. If I swap the cable over to the X motor it runs fine and if I swap the stepper over to the E0 DIP I get the same results, motor loads but barely any movement. When I tried swapping over to E1 on the previous board it worked just fine.

sjasonsmith, 2020-01-17:
@dabivs, could you reproduce the issue, then use M122 to report the microstepping used for the motor with issues? I would like to know if it is reporting 256 microsteps instead of the 16 that is configured by default. You will need to enable TMC_DEBUG to for M122 to report that information.

dabivs, 2020-01-17:
@sjasonsmith I had tried that in the past (Did this on the second board while working to try and get it fixed with some people on the Bigtree Facebook Group) and it did report 256 as you mentioned on the E0 I can run it again for you if you'd like though

dabivs, 2020-01-17:
@sjasonsmith 

Just for clarity here's one I just ran 

Connecting...
Printer is now online.
>>> m122
SENDING:M122
		X	Y	Z	E
Enabled		false	false	false	false
Set current	675	675	800	675
RMS current	673	673	795	1215
MAX current	949	949	1121	1713
Run current	21/31	21/31	25/31	21/31
Hold current	10/31	10/31	12/31	10/31
CS actual	10/31	10/31	12/31	0/31
PWM scale	12	12	14	1
vsense		1=.18	1=.18	1=.18	0=.325
stealthChop	true	true	true	false
msteps		16	16	16	256
tstep		max	max	max	max
pwm
threshold
[mm/s]
OT prewarn	false	false	false	false
off time	3	3	3	3
blank time	24	24	24	36
hysteresis
-end		-1	-1	-1	-3
-start		1	1	1	6
Stallguard thrs
DRVSTATUS	X	Y	Z	E
stst		*	*	*	*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:0A:00:00
		Y	0xC0:0A:00:00
		Z	0xC0:0C:00:00
		E	0x80:00:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK

sjasonsmith, 2020-01-17:
@dabivs, would you be able to test with a modified version of TMCStepper? I didn't post it for the original issue, because that user was having problems even with A4988 drivers, and if this works would only apply to TMC 2208/9 drivers.

In your platformio.ini file, you will need to replace the TMCStepper dependency in the [env:LPC1769] with the following:
`TMCStepper=https://github.com/sjasonsmith/TMCStepper/archive/stopListeningOnBegin.zip`

Please communicate back to me whether it worked. I've been trying to collect information from impacted people to see if it solves issues for anyone, as I cannot reproduce the problem on my SKR 1.3 (but have on AVR and STM32 controllers).

That download won't be around forever, but if it solves the problem I will try to get it added to the official TMCStepper repo.

sjasonsmith, 2020-01-17:
A little more information. That branch of mine is intended to resolve an issue that could prevent the last stepper from being configured correctly. Based on your M122 results, it may very well resolve your issue.

dabivs, 2020-01-17:
sooo I haven't run M122 again but after building with that link the motor appears to be moving at the correct speed now. 

I don't know if it helps but this is what was in there before TMCStepper@>=0.5.2,<1.0.0

sjasonsmith, 2020-01-17:
Thanks for letting me know. I’ve re-opened a previously abandoned PR to TMCStepper to include this change.
https://github.com/teemuatlut/TMCStepper/pull/101

horndogut2k3, 2020-01-19:
I'm happy to report that since using your Link in Platformio:
"TMCStepper=https://github.com/sjasonsmith/TMCStepper/archive/stopListeningOnBegin.zip"
 my SKR V 1.4 turbo is now operating on all steppers. Thanks

sjasonsmith, 2020-01-19:
> I'm happy to report that since using your Link in Platformio:
> "TMCStepper=https://github.com/sjasonsmith/TMCStepper/archive/stopListeningOnBegin.zip"
> my SKR V 1.4 turbo is now operating on all steppers. Thanks

@teemuatlut accepted my change into TMCStepper, so you can now point to the TMCStepper master instead of my fork. It's exactly the same code, but by personal branch will get deleted at some point. I'm not posting an update to platformio.ini just yet, since I don't know when it will go in a released version of TMCStepper.
> "TMCStepper=https://github.com/teemuatlut/TMCStepper/archive/master.zip

sjasonsmith, 2020-01-19:
@mjaureguiar, even though we've fixed the issue for other people with TMC drivers, your original issue seemed different. Are you still having trouble with your board?

mjaureguiar, 2020-01-20:
I have solved the problem for my case. The problem is the connection of the board with the cable to the motor. I didn't have problems with the dupont connector, but i did have problems with the XH2.54 connector. In the case of XH2.54 the cable could not be longer than 70 centimeters.

boelle, 2020-01-23:
closing since its solved 

> I have solved the problem for my case.

closed 2020-01-23 by boelle