MarlinFirmware/Marlin #16549: 5 axis [FR] (feature request title)

nyarfa, 2020-01-12:
### Description

Plus "A" and "C" axis for 5 axis motion mode. The firmware doesn't including more axis for an future printer type. Some board can use more motor controller. For example the BigTree skr pro 1.1 have three extruder motor controller. If you use only with one extruder you have two free motor controller for A and C axis.

### Feature Workflow

1. [First Action]
Need to expand the motion drives with two new axis
2. [Second Action]
Need to be expand the home with two new axis
3. [and so on...]
Need to be test it with a machine

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

DerAndere1, 2020-01-14:
This can be closed a duplicate of issue https://github.com/MarlinFirmware/Marlin/issues/15717 . Also  related: issue https://github.com/MarlinFirmware/Marlin/issues/7679

Other firmwares that alledgedly have support for more than 4 axes: 

For CNC milling, Laser engravers or 3D-printing (FDM):
- Repetier Firmware 2.0 (https://github.com/repetier/Repetier-Firmware/tree/dev2). It is currently in development but you can already try it on Ardunino Due-based boards. 
- Smoothieware smoothie for SmoothieBoard or compatibles (MKS-sBase, MKS Gen L?, Azteeg X5)

For 5 axis CNC milling only:
- grbl-Mega-5X for Aduino Mega (Atmega 2560)-based boards (RAMPS 1.4)

italocjs, 2020-01-30:
Honestly i think implementing it in the firmware will be the easiest part, the biggest challenge will be making a slicer that uses the 4th and 5th axis (we dont even have one that uses arcs yet)

nyarfa, 2020-01-31:
Yes I know :( I spoke with Daniel Ahlers who made the non-planar
post-processor. This is his answear:

"Hi,

5 axis 3D printing is one of the next big things in 3D printing.
Unfortunatly, there is no automatic toolpath generation existing yet. I
am currently not working on this.

Best Regards,
Daniel Ahlers "

I don't give up!!!

King regards,
Gabor Nyarfadi

Italo Soares <notifications@github.com> ezt írta (időpont: 2020. jan. 30.,
Cs, 22:59):

hobiseven, 2020-02-28:
Well, there could be uses of Marlin already in 2+2 axis, X/Y and U/V secondary cartesian coordinates, in hot wire cutting applications. 
A simple CAM for this is Jedicut, jedicut.com, which takes 2 DXF files as path for eaxh plane, and generates Gcode.  So not only 3d printing...

thinkyhead, 2020-02-28:
Duplicate of #15717

closed 2020-02-28 by thinkyhead