MarlinFirmware/Marlin #16550: [FR] Adjustable Extruder Direction via LCD Menu

pmjdebruijn, 2020-01-12:
Currently, it's only possible to adjust the extruder steps/mm via de LCD menus, not the extruder direction.

However, many printers are shipped with very simple extruder designs, these are often trivial to replace, except for one point: extruder direction inversion (for geared extruders).

While that can already be adjusted via firmware rebuild, this is a significantly more difficult task for many people as opposed to just replacing some mechanical parts, and tightening some bolts.

So it would be very nice if the extruder direction could be adjusted via the menu, so replacing an extruder becomes that much more easy for any Marlin user in the future, as vendors adopt Marlin 2.0 for their factory firmwares.

For example: Settings, Advanced Configuration, Steps/mm, then one of these three:

  1. E0/inverted true/false
  2. E0/directory: factory/inverted
  3. E0/type: direct/geared

A feature like this could replace INVERT_E0_DIR, but I would argue to implement this on top. So INVERT_E0_DIR defines the direction for the factory extruder, while the lcd menu provides on "override" on top. But in the end that's a bit of a matter of taste I guess.

I did some digging, and presumably these files would be involved for implementing this:
 module/configuration_store.cpp (persistant storage)
 src/lcd/menu/menu_advanced.cpp (lcd menu)
 src/module/stepper/indirection.h (actual behavior)

ellensp, 2020-01-13:
Not needed, all you need to do is swap two pins over on the stepper cable plug and the stepper turns in the opposite direction.

Talha909, 2020-01-13:
which cable color need to change

ellensp, 2020-01-13:
There is no real color standard on stepper motors wires.  To change directions at the controller of the stepper cable swap pins 1 & 2 or pins 3 & 4 (but not both!) or you turn the entire plug around 180 degrees (if the shape of the plug will let you) NB always have power off when doing any hardware changes.

pmjdebruijn, 2020-01-14:
I know that swapping the wires is possible, but this is again not something many people are particularly keen to do, as there's risk of messing up the connector, leaving you with a temporarily broken printer.

So I still think there's a good case to be made for adding this capability in software.

DrumClock, 2021-03-04:
Hi, it would be enough to extend the M569 by

`M569 P0 S0`
         
`Pnn `    Motor driver number
`Sn`        Direction of movement of the motor (s) attached to this driver: 0 = backwards, 1 = forwards (default 1)

How can RepRam FW do it

KimmoHop, 2025-01-02:
ProUI uses C562 command for this and redefines INVERT_E0_DIR with variable.
It is **really** useful (_like setting min and max X and Y from the UI, trying to find how far the axes move freely -> how close to far edges the probe can get_) and saves from re-building and re-flashing. Orbiter always runs the wrong way (it could also be me) ;)

This is really really old FR, but with some guidance, if maintainers see value in this, I might have time to look at :)

macron0, 2025-11-05:
> There is no real color standard on stepper motors wires. To change directions at the controller end of the stepper motor cable swap pins 1 & 2 or pins 3 & 4 (but not both!) or you turn the entire plug around 180 degrees (if the shape of the plug will let you) NB always have power off when doing any hardware changes.

About the power. I am sure my printer was turn off (my ender 5 has very loud fans)  But usb was still connected to pc. could that have destroyed my stepper driver om my mobo? coz putting all the wires back in original order the stepper is not moving I think I broke it.

thinkyhead, 2025-11-05:
Note that this is a duplicate of the Feature Request "make everything adjustable at runtime."