MarlinFirmware/Marlin #16555: [BUG] USB connection breaks

cburghardt, 2020-01-13:
I am using a Sovol SV01 which is shipped with Marlin 1.6.1. I have upgraded it to Marlin 2.0 using the latest version from CopterTec (https://github.com/CopterTec/Marlin-Sovol_SV01/releases).
The printer is connected via USB to Raspberry 4 running Octoprint. This setup worked flawless with Marlin 1.6 but since upgrading to 2.0 all prints are simply stopping at random times because the USB (serial) conversation breaks down.
The error message is described here: https://community.octoprint.org/t/octoprint-randomly-loses-connection-to-the-printer-with-a-serialexception/228

According to CopterTec the issue is known and stems from an increased performance requirement of Marlin 2.0 which kills the USB connection. The recommendation is to use SD Card and ditch Octoprint. However I would like to know how that can be fixed or worked around.

ellensp, 2020-01-13:
Unfortunately we cant help unless you use the official marlin github repository... a) who knows what the CopterTec people might have done b) if it was a bug it may have already been fixed and they haven't updated their repository.   

I would get a copy of the official Marlin bugfix-2.0.x  branch and use the Config files from CopterTec, Compile and test that.  If this has issues. Then we may be able to help.

cburghardt, 2020-01-13:
The changes are actually straight forward:
https://github.com/CopterTec/Marlin-Sovol_SV01/tree/master/config/examples/SOVOL/SV01-BLTouch

But I get your point and will give it a shot with the bugfix branch.

boelle, 2020-01-13:
closing as we do not support forks/copies

closed 2020-01-13 by boelle