MarlinFirmware/Marlin #16561: [FR] Need a command to set PT100 temp offset

karabas2011, 2020-01-14:
In common case PT100 reading is about 4-6C higher at room temperature.
I think it is constant or almost constant (difference will be larger at high temp ) along the range. I need a  command to set temp offset to subtract from sensor reading.

thinkyhead, 2020-07-13:
I see that `temperature.cpp` has this…
```cpp
max31865.temperature(100, 400)  // 100 ohms = PT100 resistance. 400 ohms = calibration resistor
```
This is where you would subtract 6.  But maybe you can also fix it by putting in a more accurate calibration resistance value.

AnHardt, 2020-07-13:
Indeed this could be the wheel to turn.
The original Adafruit modules claim to have a **!!!430!!!** Ohm 0.1% low temperature drift reference resistor. The Chinese clones copy in Adafruits module description - but you never know.

For showing to high temperatures also excessive long or thin wires, with a high resistance, in between module and PT100 could be the reason. 3- and 4-wire modes are currently not supported by Marlin so this can't be easily compensated away.

thinkyhead, 2020-07-15:
Well, that might be part of it. See if `max31865.temperature(100, 430)` produces a better result, @karabas2011.

karabas2011, 2020-07-15:
I use ulimaker amplifier clone 
https://aliexpress.ru/item/32829690782.html?spm=a2g0o.productlist.0.0.41625bd4nrXo2l&algo_pvid=2da03427-664a-4c68-a703-b85f6224d172&algo_expid=2da03427-664a-4c68-a703-b85f6224d172-6&btsid=0b8b034115948150769077098ec33a&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

I believe it is not max31865 and I set #define TEMP_SENSOR_0 20
So PlatformIO does not highlight the above string...

ellensp, 2020-07-15:
This is apparently the circuit diagram for that board https://wiki.e3d-online.com/images/8/83/PT100_Schematic_%28v1%29.png
Uses a INA826AIDGKR

ellensp, 2020-07-15:
This is the same as https://github.com/MarlinFirmware/Marlin/issues/18420

karabas2011, 2020-07-15:
It is powered from stabilized 3.3v  (SKR1.3//Sbase/SKR1.1pro)
So I did not catch what to do

AnHardt, 2020-07-15:
>  I set #define TEMP_SENSOR_0 20

> It is powered from stabilized 3.3v (SKR1.3//Sbase/SKR1.1pro)

>  *    20 : Pt100 with circuit in the Ultimainboard V2.x with 5v excitation (AVR)
>  *    21 : Pt100 with circuit in the Ultimainboard V2.x with 3.3v excitation (STM32 \ LPC176x....)
 
And why the hell not 21?

ellensp, 2020-07-16:
As AnHardt politely pointed out.
You have a 3.3v supply, you must use the 3.3 table ie 21 not 20.

karabas2011, 2020-07-16:
Because no difference. Hystorically it was only one PT100 - 20
when I had seen 21 I tried but it shows the same room temp  as 20.

karabas2011, 2020-07-16:
I tried  but 21 shows room temp 1C (!). So it is unusable. Possibly incompatible amplifier board

ellensp, 2020-07-16:
you mention several controllers, which controller did you use and how did you wire in the pt100 amplifier? 
For eg, the skr 1.4 doesn't have any nice places to tap 3.3v so did you use 5?

karabas2011, 2020-07-16:
I use all of them. When I got any, I simply remove resistor from thermistor port. Then I look for 3.3v pin and power Amplifier from it. I have got similar reading on all of the boards.
I have post already amplifier board I use. I tried different temp_sensors defines but 20 is only that works.

AnHardt, 2020-07-16:
Wow! What happened? You accidentally gave us more than one new information per message.

> I simply remove resistor from thermistor port.

That would have been the next question. Did you do that right? You did not accidentally remove the wrong resistor or removed the capacitor or made a bridge instead? All more on the unlikely side - on 3 boards.

What is the temperature reading when the module is not plugged in? Concentrate on one of the boards.

Before the next reply i expect to get: Configuration.h, Configuration_adv.h, Platformio.ini, pins-file, a closeup photo from the place the resistor is removed, a ( photo ) documentation about - how the module is connected to the board. For only one board - with all information about the same board. I'm tired of guessing.

Bob-the-Kuhn, 2020-07-16:
Please take a look at [PR 16731 ](https://github.com/MarlinFirmware/Marlin/pull/16731) for a similar discussion on PT100, PT100 amplifiers and 5V vs. 3.3V.

---

I've ended up using a PT1000 sensor with a 1000 Ohm 0.1% pullup and no amplifier.

---

I have a different view about how to handle thermal inputs than a lot of others.

I maintain, in a hobbyist world (only a few printers), that it is the repeatability of the temperature measurement that is critical.  

If you are following the recommendation to do test prints with each new batch of filament then you are determining the best INDICATED temperature for your system.  It doesn't matter what the indicated temperature is, just that you can reliably get back to it.

Under this approach the 4-6C offset you are seeing at room temperature is of no concern.

karabas2011, 2020-07-17:
> Wow! What happened? You accidentally gave us more than one new information per message.
>
>
> That would have been the next question. Did you do that right? You did not accidentally remove the wrong [...]

What's the problem with removing pullup resistor? Its common way. I did it only after connecting to manual recommended ADC ports. Then I compared temp reading with removed resistor port. The same.  I prefer to use common THx ports on the boards and save spare pins.
If you did not remove resistor you get ~45C reading on sensor type 20.

karabas2011, 2020-07-17:
> Please take a look at [PR 16731 ](https://github.com/MarlinFirmware/Marlin/pull/16731) for a similar discussion on PT100, PT100 amplifiers and 5V vs. 3.3V.
>
> I've ended up using a PT1000 sensor with [...]

Personally I always go the same way. But went with PT100: 
1 with repeatability (between batch of printers) in mind 
2 customers expect to see "real" temp on cold printer. Now it is always higher and higher than bed temp (NTC termistor)
and I suppose 200-250 temps shifted too.

Bob-the-Kuhn, 2020-07-17:
I think I'm seeing two threads here:
- A request to add an offset option for the PT100.
- Attempts to "fix" your system so the PT100 works correctly.

If there is a need to "fix" your system then I think that should be the topic of a separate issue.

I'd prefer to limit discussion in this thread to if adding an offset option is something we should add to Marlin.

Bob-the-Kuhn, 2020-07-18:
Please take a look at PR #18687.  

I believe this implements your request.  If yes then please close this issue.

Bob-the-Kuhn, 2020-07-19:
Sorry but more knowledgeable people pointed out some potential unwanted interactions between my simplistic approach in PR #18687 and the thermal protection features. 

The only other easy option I see is to create a custom thermistor table that has the offset built into it.

Quas7, 2020-09-03:
thermistor_21.h might have a bug. See also #19234.

The look-up table of ADC values is depending on both OP-Amp supply voltage as well as ADC-reference voltage.
As long as ADC-reference and INA826 supply voltage are identical, the look-up table stays the same as ADC values are ratios of the ADC-reference voltage.

The conversion OV_SCALE is therefore not needed for 3V3 microcontrollers setups, in my opinion.
Also explains, why we get a temperature reading of '1' at room temp.

sjasonsmith, 2020-09-11:
This seems closely related to, and possibly a duplicate of #18809.

sjasonsmith, 2020-11-10:
Recently calibration resistance values were made configurable for PT100 devices: https://github.com/MarlinFirmware/Marlin/pull/19695

Once the correct resistance value is used, I suspect that eliminates the need for the offset requested here.

Does anyone feel this still needs to stay open, or can it be closed?

sjasonsmith, 2020-12-02:
Given the earlier comments in this FR and the apparent lack of interest since my last post I am closing this FR.

closed 2020-12-02 by sjasonsmith