MarlinFirmware/Marlin #16563: [BUG] [bugfix-2.0.x] Problem with timer TIM6 and missing definitions of digitalPin and SystemClock_Config functions for STEVAL_F401VE variant

fradaxx, 2020-01-14:
### Bug Description
Trying to compile the bugfix-2.0.x version for the STEVAL-3DP001V1 board with PlatformIO, I first obtain errors on TIM6 timer which doesn't exist in the F401VE processor. I solved this first problem thanks to @ Bob-the-Kuhn changing TIM6 to another unused timer but nonetheless the compilation succedees, the final linking fails with many error about the unavailability of digitalPin and SystemClock_Config definitions:

`...
Compiling .pio\build\STM32F401VE_STEVAL\src\src\sd\usb_flashdrive\lib-uhs2\usbhost.cpp.o
Linking .pio\build\STM32F401VE_STEVAL\firmware.elf
.pio/build/STM32F401VE_STEVAL/FrameworkArduino/HardwareSerial.cpp.o: In function `HardwareSerial::HardwareSerial(void*)':
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareSerial.cpp:226: undefined reference to `digitalPin'
.pio/build/STM32F401VE_STEVAL/FrameworkArduino/pins_arduino.c.o: In function `analogInputToPinName':
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\cores\arduino/pins_arduino.c:61: undefined reference to `digitalPin'
.pio/build/STM32F401VE_STEVAL/FrameworkArduino/stm32/hw_config.c.o: In function `hw_config_init':
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/hw_config.c:63: undefined reference to `SystemClock_Config'
.pio/build/STM32F401VE_STEVAL/FrameworkArduino/stm32/usb/usbd_conf.c.o: In function `HAL_PCD_ResumeCallback':
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb/usbd_conf.c:325: undefined reference to `SystemClock_Config'
.pio/build/STM32F401VE_STEVAL/FrameworkArduino/stm32/usb/usbd_conf.c.o: In function `OTG_FS_WKUP_IRQHandler':
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\usb/usbd_conf.c:434: undefined reference to `SystemClock_Config'
.pio/build/STM32F401VE_STEVAL/FrameworkArduino/wiring_analog.c.o: In function `analogWrite':
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring_analog.c:133: undefined reference to `digitalPin'
.pio/build/STM32F401VE_STEVAL/FrameworkArduino/wiring_digital.c.o: In function `pinMode':
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring_digital.c:76: undefined reference to `digitalPin'
.pio/build/STM32F401VE_STEVAL/FrameworkArduino/wiring_digital.c.o: In function `LL_GPIO_ResetOutputPin':
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_gpio.h:913: undefined reference to `digitalPin'
.pio/build/STM32F401VE_STEVAL/FrameworkArduino/wiring_digital.c.o: In function `digitalRead':
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring_digital.c:86: undefined reference to `digitalPin'
.pio/build/STM32F401VE_STEVAL/src/src/HAL/HAL_STM32/fastio.cpp.o: In function `FastIO_init()':
c:\Users\Francesco\Desktop\MarlinFirmware/Marlin\src\HAL\HAL_STM32/fastio.cpp:32: undefined reference to `digitalPin'
.pio/build/STM32F401VE_STEVAL/src/src/MarlinCore.cpp.o:c:\Users\Francesco\Desktop\MarlinFirmware/Marlin\src/MarlinCore.cpp:295: more undefined references to `digitalPin' follow
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F401VE_STEVAL\firmware.elf] Error 1
====================================================================================== [FAILED] Took 91.85 seconds ======================================================================================

Environment                   Status    Duration
----------------------------  --------  ------------
megaatmega2560                IGNORED
megaatmega1280                IGNORED
rambo                         IGNORED
FYSETC_F6_13                  IGNORED
FYSETC_F6_14                  IGNORED
sanguino_atmega644p           IGNORED
sanguino_atmega1284p          IGNORED
melzi                         IGNORED
melzi_optiboot                IGNORED
at90usb1286_cdc               IGNORED
at90usb1286_dfu               IGNORED
DUE                           IGNORED
DUE_USB                       IGNORED
DUE_debug                     IGNORED
LPC1768                       IGNORED
LPC1769                       IGNORED
STM32F103RC                   IGNORED
STM32F103RC_fysetc            IGNORED
STM32F103RC_bigtree           IGNORED
STM32F103RC_bigtree_USB       IGNORED
STM32F103RC_bigtree_512K      IGNORED
STM32F103RC_bigtree_512K_USB  IGNORED
STM32F103RE                   IGNORED
STM32F103RE_bigtree           IGNORED
STM32F103RE_bigtree_USB       IGNORED
STM32F4                       IGNORED
STM32F7                       IGNORED
STM32F401VE_STEVAL            FAILED    00:01:31.848
ARMED                         IGNORED
STM32F103VE_GTM32             IGNORED
STM32F103VE_longer            IGNORED
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
mks_robin                     IGNORED
mks_robin_pro                 IGNORED
mks_robin_lite                IGNORED
mks_robin_lite3               IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
FLYF407ZG                     IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
rumba32_f446ve                IGNORED
mks_rumba32                   IGNORED
include_tree                  IGNORED
================================================================================= 1 failed, 0 succeeded in 00:01:31.848 =================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
`

### My Configurations
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4058700/configuration.zip)

### Steps to Reproduce

1. Download the bugfix-2.0.x version of Marlin (the first one to officially include support for board STEVAL-3DP001V1)
2. Unzip the archive content on a forlder on the Desktop and rename the folder MarlinFirmware
3. Add the folder to an empty VSCode workspace 
4. Following the directions of @Bob-the-Kuhn I modify definition of timers in 

\Marlin\src\HAL\HAL_STM32\timers.h (lines 66 & 70)
\buildroot\share\PlatformIO\variants\STEVAL_F401VE\variant.h ( lines 196 & 199)

to use timers effectively present in the F401VE microcontroller.

**Expected behavior:**
The code should compile and link finely.

**Actual behavior:**
Code compiles but final linking fails with various errors about the definition of digitalPin and SystemClock_Config.
I think that these definitions (as per other variants based on STM32 microcontrollers) should be given, besides other ones, in a variant.cpp file contained in the variant folder devoted to the board (.\buildroot\share\PlatformIO\variants\STEVAL_F401VE). Currently in this folder I see only a variant.h file.
This is the content of the STEVAL_F401VE folder in my case:
[STEVAL_F401VE.zip](https://github.com/MarlinFirmware/Marlin/files/4058716/STEVAL_F401VE.zip)

fradaxx, 2020-01-14:
At last I managed to successfully compile and link the MarlinFirmware for STEVAL-3DP001V1 board using variant files for STM32F401VE extracted from GitHub repository  stm32duino/Arduino_Core_STM32 where I found a profile for the board (https://github.com/stm32duino/Arduino_Core_STM32/tree/master/variants/ST3DP001_EVAL).
I copied this folder into the variants folder of MarlinFirmware (.\buildroot\share\PlatformIO\variants), created a separate "env" profile in platformio.ini using the STM32F401VE_STEVAL profile as sample and built the project.
However, if I upload the firmware on the board, it communicates with the host, accepting the commands but attached motors do not spin at all.
I think this is due to the problems with L6474 stepper drivers on which @Bob-the-Kuhn is working hard by some weeks.
Also the SD Card doesn't work, nonetheless it is enabled in the firmware.
I will investigate in more depth trying to solve these problems.

Bob-the-Kuhn, 2020-01-15:
I have not been able to reproduce the problem.

I downloaded the latest bugfix-2.2.x, added in your configuration files and then compiled using Platformio within Atom.  No compile errors.

I then changed the three files & re-compiled.  Again no erors.

I then opened the project in VScode, deleted the build and compiled.  No compile errors.

I suggest downloading a fresh copy of bugfix.  It now has the L6474 support plus some STEVAL pin fixes.

If that doesn't compile then get a fresh copy of the **.platformio\packages\framework-arduinoststm32** package by deleting that directory and re-compiling.

FYI - PR #16565 has an updated pins file.  It has the correct definitions for the analog inputs.

---

The on-board SD card isn't yet available.  That's next on my list.

Bob-the-Kuhn, 2020-01-15:
ON-board SD card access was easier than I expected.  Replace the following files with the attached.
- \Marlin\src\pins\stm32\pins_STEVAL_3DP001V1.h
- \Marlin\src\HAL\HAL_STM32\HAL_SPI.cpp

[HAL_SPI.zip](https://github.com/MarlinFirmware/Marlin/files/4063238/HAL_SPI.zip)

I'll create a PR for this change tomorrow.

FYI - PR #16565 has been merged.

fradaxx, 2020-01-15:
@Bob-the-Kuhn, thanks for your answers

Unfortunately following your instructions I am not able to compile/link even getting the last version of bugfix-2.0.x. I've also forced the update of .platformio\packages\framework-arduinoststm32 whitout success.

The only way to compile/link correctly is to point to 3DP001V1_STEVAL folder in .platformio\packages\framework-arduinoststm32\variants changing the board_dir value in [platformio] section of platformio.ini or copying folder 3DP001V1_STEVAL into .buildroot\share\PlatformIO\variants.
If I use env = STM32F401VE_STEVAL to compile, I continue to obtain linking error due to digitalPin and SystemClock_Config missing references.
I think this is due to the fact that variant.cpp source file is not present in .buildroot\share\PlatformIO\variants\STEVAL_F401VE folder even after the complete renewal of bugfix-2.0.x in my local repository.

Please note also that if I substitute files pins_STEVAL_3DP001V1.h and HAL_SPI.cpp I obtain a lot of pin redefinition warnings for SDSS pin in pins_STEVAL_3DP001V1.h (lines 247 and 171) and programming the board with the generated firmware makes it unstable (it is reset by the Watchdog approximately every 3-4 secs):

`
Marlin bugfix-2.0.x

echo: Last Updated: 2020-01-15 | Author: (none, default config)
echo:Compiled: Jan 15 2020
echo: Free Memory: 91179  PlannerBufferBytes: 1280
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)

echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J                                                                         <junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
start
echo: Watchdog Reset
Marlin bugfix-2.0.x

echo: Last Updated: 2020-01-15 | Author: (none, default config)
echo:Compiled: Jan 15 2020
echo: Free Memory: 91179  PlannerBufferBytes: 1280
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)

echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J                                                                         <junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
start
echo: Watchdog Reset
Marlin bugfix-2.0.x

echo: Last Updated: 2020-01-15 | Author: (none, default config)
echo:Compiled: Jan 15 2020
echo: Free Memory: 91179  PlannerBufferBytes: 1280
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)

echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J                                                                         <junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00

start
echo: Watchdog Reset
Marlin bugfix-2.0.x

echo: Last Updated: 2020-01-15 | Author: (none, default config)
echo:Compiled: Jan 15 2020
echo: Free Memory: 91179  PlannerBufferBytes: 1280
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)

echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00
start
echo: Watchdog Reset
Marlin bugfix-2.0.x

echo: Last Updated: 2020-01-15 | Author: (none, default config)
echo:Compiled: Jan 15 2020
...
`

Bob-the-Kuhn, 2020-01-15:
I'm at a loss to explain why you can't compile.

I suggest starting over from scratch:
- uninstall platformio in vscode
- delete the **.platformio** directory
- reboot
- install platformio in vscode
- download a fresh copy of bugfix-2.0.x
- unzip it into a NEW directory
- copy your config files into it
- point vscode/platformio at the new directory

Another possibility is to compile with the Arduino IDE.

fradaxx, 2020-01-15:
All steps done without success...
The build with Arduino goes wrong even after installing stm32 support using the board manager.
The true mistery for me remains how it is possible to correctly build for without the variant.cpp file in board directory .buildroot\share\PlatformIO\variants\STEVAL_F401VE! I can't find it whatever is the way I use to download bugfix-2.0.x version (as a zip archive or by GitHub CLI/Desktop).
The only other try I could do is to flash on my board a firmware.bin generated by you. If you agree, please send it in response to this message.
Anyway, please don't worry anymore about this problem! I will try to solve it by myself.
Thank you again for your willingness.

Bob-the-Kuhn, 2020-01-15:
The on-board SD card access changes are on PR #16579

Bob-the-Kuhn, 2020-01-15:
I've finally been able to reproduce the problem.  You are correct - **variant.cpp** is missing.  Below is the file.

[variant.zip](https://github.com/MarlinFirmware/Marlin/files/4067636/variant.zip)

The build process copies my custom variant folder over to the platformio variant and then uses that during the build.  Apparently I created the **variant.cpp** file early in my work and then "lost" it.  But since it was hiding under platformio copy I didn't realize it was missing.

Another PR coming to fix it.

boelle, 2020-01-15:
remember to mention issue in PR so we can see when PR is merged :-D

fradaxx, 2020-01-15:
Perfect, thank you @Bob-the-Kuhn ; finally I'm able to compile/link Marlin for STEVAL-3DP001V1 without errors!
Steppers don't spin yet and SD Card Init fails; however I'll wait that all your queued PRs be effective to try to recompile with all the relevant corrections for this board.
Thank you again for your efforts and willingness!

Bob-the-Kuhn, 2020-01-16:
Please post your configuration files & I'll see if I can spot something.

fradaxx, 2020-01-16:
Here they are:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4069764/configuration.zip)

Please note that downloading and compiling Marlin from your branch https://github.com/Bob-the-Kuhn/Marlin-Bob-2/tree/STM32---add-soft-SPI-so-can-access-on-board-SD-cards-on-STM32F4xx-and-STM32F7xx-boards, and using the attached config files, compilation goes well, the board doesn't restart periodically due to whatchdog error, the SD card works (SD Card Init OK, and I'm able to list its content with M22 command), but motors don't spin.

`start
echo: External Reset
Marlin bugfix-2.0.x

echo: Last Updated: 2020-01-15 | Author: (none, default config)
echo:Compiled: Jan 16 2020
echo: Free Memory: 89795  PlannerBufferBytes: 1280
echo:SD card ok
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)

echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00`
In the attached configuration files I disabled all temperature sensor, also the temperature sensor 0 putting 999 as his type and giving value 230 to DUMMY_THERMISTOR_999_VALUE symbol; I don't know if this is required with this version of Marlin but the old ST's version of the STEVAL-3DP001V1 firmware blocks all motors if temperature sensors report temperatures not compatible with the 3D printing process.

If I give command M17 to enable steppers I don't hear the typical initial "bump" due to the first energize of their coils.

If I give command G0 X10 Y10 Z10 E10 to make motors move, they don't spin but if I give command
M114 Marlin reports the correct final position:

`M17
ok
G21
ok
G0 X10 Y10 Z10 E10
ok
M114
X:10.00 Y:10.00 Z:10.00 E:10.00 Count X:800 Y:800 Z:40000
ok
`

Bob-the-Kuhn, 2020-01-16:
The L6474 stepper drivers hadn't been selected in **Configuration.h** and the SPI chain hadn't been setup in **Configuration_adv.h**.

Please try these [modified files](https://www.dropbox.com/s/qik2ancbus71fja/Configuration%28modified%29.zip?dl=0).  

Do you have any steppers connected besides X, Y, Z and E0?  If yes I'll modify the config files accordingly.

Please post the results of the M122 and M906 commands.  That'll tell me if the SPI chain is working.

---

FYI - I've also enabled the X2 and Y2 stepper drivers because  having 6 steppers in the chain is the simplest hardware (don't need to install R16 & R17).

fradaxx, 2020-01-16:
It works!
At the moment I have only 2 steppers connected to the X channel; I don't plan to connect others, at least in the near future.
I am using the STEVAL-3DP001V1 to finely control the position of a single linear actuator.
Here is the log of startup and M122/M906 commands:

`
start
echo: External Reset
Marlin bugfix-2.0.x

echo: Last Updated: 2020-01-15 | Author: (none, default config)
echo:Compiled: Jan 16 2020
echo: Free Memory: 89191  PlannerBufferBytes: 1280
echo:SD card ok
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)

echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z4000.00 E500.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:  M301 P22.20 I1.08 D114.00

M122
AXIS: X     status: fe13   1111 1110 0001 0011   L6474
...OUTPUT: OFF   BUSY:  NO    DIR: FORWARD   Last Command: VALID    COMPLETED
...THERMAL: OK                OVERCURRENT: NO    STALL: NA    STEP-CLOCK MODE: NA   UNDER VOLTAGE LOCKOUT:  NO
AXIS: X2    status: fc03   1111 1100 0000 0011   L6474
...OUTPUT: OFF   BUSY:  NO    DIR: FORWARD   Last Command: VALID    COMPLETED
...THERMAL: OK                OVERCURRENT: NO    STALL: NA    STEP-CLOCK MODE: NA   UNDER VOLTAGE LOCKOUT:  YES
AXIS: Y     status: fe13   1111 1110 0001 0011   L6474
...OUTPUT: OFF   BUSY:  NO    DIR: REVERSE   Last Command: VALID    COMPLETED
...THERMAL: OK                OVERCURRENT: NO    STALL: NA    STEP-CLOCK MODE: NA   UNDER VOLTAGE LOCKOUT:  NO
AXIS: Y2    status: fc03   1111 1100 0000 0011   L6474
...OUTPUT: OFF   BUSY:  NO    DIR: REVERSE   Last Command: VALID    COMPLETED
...THERMAL: OK                OVERCURRENT: NO    STALL: NA    STEP-CLOCK MODE: NA   UNDER VOLTAGE LOCKOUT:  YES
AXIS: Z     status: fe13   1111 1110 0001 0011   L6474
...OUTPUT: OFF   BUSY:  NO    DIR: FORWARD   Last Command: VALID    COMPLETED
...THERMAL: OK                OVERCURRENT: NO    STALL: NA    STEP-CLOCK MODE: NA   UNDER VOLTAGE LOCKOUT:  NO
AXIS: E0    status: fe13   1111 1110 0001 0011   L6474
...OUTPUT: OFF   BUSY:  NO    DIR: FORWARD   Last Command: VALID    COMPLETED
...THERMAL: OK                OVERCURRENT: NO    STALL: NA    STEP-CLOCK MODE: NA   UNDER VOLTAGE LOCKOUT:  NO
ok

M906
M906
AXIS: X     status: ff93   1111 1111 1001 0011
...OverCurrent Threshold:  4 (1875.00 mA)   TVAL: 47 (1500.00 mA   Motor Status: NA)
...MicroSteps: 16   ADC_OUT: 21
   Vs_compensation: NA
...KVAL_HOLD: NA   KVAL_RUN : NA   KVAL_ACC: NA   KVAL_DEC: NA   V motor max =  NA
...SLEW RATE: 110V/uS

AXIS: X2    status: ff83   1111 1111 1000 0011
...OverCurrent Threshold:  8 (3375.00 mA)   TVAL: 41 (1312.50 mA   Motor Status: NA)
...MicroSteps: 128   ADC_OUT: 21
   Vs_compensation: NA
...KVAL_HOLD: NA   KVAL_RUN : NA   KVAL_ACC: NA   KVAL_DEC: NA   V motor max =  NA
...SLEW RATE: 110V/uS

AXIS: Y     status: ff93   1111 1111 1001 0011
...OverCurrent Threshold:  4 (1875.00 mA)   TVAL: 47 (1500.00 mA   Motor Status: NA)
...MicroSteps: 16   ADC_OUT: 21
   Vs_compensation: NA
...KVAL_HOLD: NA   KVAL_RUN : NA   KVAL_ACC: NA   KVAL_DEC: NA   V motor max =  NA
...SLEW RATE: 110V/uS

AXIS: Y2    status: ff83   1111 1111 1000 0011
...OverCurrent Threshold:  8 (3375.00 mA)   TVAL: 41 (1312.50 mA   Motor Status: NA)
...MicroSteps: 128   ADC_OUT: 21
   Vs_compensation: NA
...KVAL_HOLD: NA   KVAL_RUN : NA   KVAL_ACC: NA   KVAL_DEC: NA   V motor max =  NA
...SLEW RATE: 110V/uS

AXIS: Z     status: ff93   1111 1111 1001 0011
...OverCurrent Threshold:  4 (1875.00 mA)   TVAL: 47 (1500.00 mA   Motor Status: NA)
...MicroSteps: 16   ADC_OUT: 21
   Vs_compensation: NA
...KVAL_HOLD: NA   KVAL_RUN : NA   KVAL_ACC: NA   KVAL_DEC: NA   V motor max =  NA
...SLEW RATE: 110V/uS

AXIS: E0    status: ff93   1111 1111 1001 0011
...OverCurrent Threshold:  4 (1875.00 mA)   TVAL: 47 (1500.00 mA   Motor Status: NA)
...MicroSteps: 16   ADC_OUT: 21
   Vs_compensation: NA
...KVAL_HOLD: NA   KVAL_RUN : NA   KVAL_ACC: NA   KVAL_DEC: NA   V motor max =  NA
...SLEW RATE: 110V/uS

ok
`

Bob-the-Kuhn, 2020-01-16:
FANTASTIC!!!

Please close this if you think we're done.

fradaxx, 2020-01-16:
I tried to set the current limit with the M906 command but it appears to don't work 
If I give, as an example, the command

M906 X500 Y500 Z500

to limit the current for XYZ axes to 500 mA and then ask again their values with 

M906

I obtain the same output of my last post.

In any case I think that for clarity, should this problem be confirmed, it can be treated with another bug report; do you agree?

Bob-the-Kuhn, 2020-01-16:
Bug is confirmed.

It is treating the numbers as TVAL counts.  Current is (TVAL + 1) * 31.25ma

The M906 description is definitely wrong.  I'm not sure yet how I'm going to change it yet.  on a L6474 TVAL directly affect peak current.  All the others this sets KVAL_HOLD which affects the voltage PWM duty cycle which affects the current.

There's a couple of ways you can handle this:
- In Configuration_adv.h you can change the STALLCURRENT settings in the L64XX section.  You should probably also drop the OVERCURRENT settings while you are at it.
- Use the M906 command but manually calculate the TVAL count needed.  For 500mA that would be 15.  To set the X stepper to 500mA issue **M906 X15**. To set the X2 stepper to 500mA issue **M906 X15 I2**.

Bob-the-Kuhn, 2020-01-16:
Here is the correct code.  It replaces the following file:
Marlin\src\gcode\feature\L6470\M906.cpp

[M906.zip](https://github.com/MarlinFirmware/Marlin/files/4074192/M906.zip)

fradaxx, 2020-01-17:
Thank you very much for your assistance and willingness!
Regards.

fradaxx, 2020-01-17:
Ooooops, unfortunately when the modified M906.cpp is substituted to the original one, I get a compilation error:

`
Processing STM32F401VE_STEVAL (platform: ststm32; board: STEVAL_STM32F401VE; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/STEVAL_STM32F401VE.html
PLATFORM: ST STM32 6.0.0 > STM32F401VE (96k RAM. 512k Flash)
HARDWARE: STM32F401VET6 84MHz, 96KB RAM, 502.23KB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32 3.10700.191028 (1.7.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit MAX31865 library> 1.0.3
|   |-- <SPI> 1.0
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.8.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\RingBuffer.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\USBSerial.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\WSerial.cpp.o
C:\Users\Francesco\.platformio\packages\framework-arduinoststm32\cores\arduino\Tone.cpp:134:2: warning: #warning "TIMER_TONE or HAL_TIM_MODULE_ENABLED not defined" [-Wcpp]
 #warning "TIMER_TONE or HAL_TIM_MODULE_ENABLED not defined"
  ^~~~~~~
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\abi.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\board.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\main.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\new.cpp.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\pins_arduino.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_adc_ex.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_can.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cec.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_comp_ex.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cordic.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cortex.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_crc_ex.c.o
Compiling .pio\build\STM32F401VE_STEVAL\FrameworkArduino\stm32\HAL\stm32yyxx_hal_cryp.c.o
[... 420 lines cut ...]
mks_robin_mini                IGNORED
mks_robin_nano                IGNORED
mks_robin                     IGNORED
mks_robin_pro                 IGNORED
mks_robin_lite                IGNORED
mks_robin_lite3               IGNORED
jgaurora_a5s_a1               IGNORED
STM32F103CB_malyan            IGNORED
chitu_f103                    IGNORED
STM32F401VE_STEVAL            FAILED    00:01:18.171
FLYF407ZG                     IGNORED
FYSETC_S6                     IGNORED
STM32F407VE_black             IGNORED
BIGTREE_SKR_PRO               IGNORED
BIGTREE_BTT002                IGNORED
teensy31                      IGNORED
teensy35                      IGNORED
esp32                         IGNORED
linux_native                  IGNORED
SAMD51_grandcentral_m4        IGNORED
rumba32_f446ve                IGNORED
mks_rumba32                   IGNORED
include_tree                  IGNORED
========================================================================= 1 failed, 0 succeeded in 00:01:18.171 ========================================================================= 
The terminal process terminated with exit code: 1`

Bob-the-Kuhn, 2020-01-17:
That is not what I wanted to hear. ###  😞

Now to figure out how to reproduce the problem on my walking wounded board ...

fradaxx, 2020-01-17:
If you like I can test your mods on my board; the only problem can be with the time zone (I'm in Italy).
Of course I will also do some tries to solve the problem by myself.

Bob-the-Kuhn, 2020-01-17:
You definitely uncovered a strange one.  As long as **X_DUAL_STEPPER_DRIVERS** there will be a compile error.  Doesn't matter what stepper driver is selected.

Try replacing this file with the attached: \Marlin\src\core\drivers.h
https://www.dropbox.com/s/bo4b4zcdwq35ugo/drivers.zip?dl=0

fradaxx, 2020-01-17:
After the drivers.h substitution Marlin compiles without error ever re-enabling lines 668 and 669 in Marlin\Configuration.h.
M906 doesn't change behaviour: writing seems to work with motor changing behaviour in function of stall current programmed, but read reports always the same values.

Bob-the-Kuhn, 2020-01-17:
Please ZIP & attach your M122 and M906 results.

fradaxx, 2020-01-17:
Here is the log of a command session in text format where you can see the beaviour of M122/M906 commands when I change the value of stall current of X axis. As you can see the TVAL value reported by M906 command is always 0 (corresponding to a stall current of 31.25 mA) , even if I reprogram the axis current to 300 mA.
Nonetheless the behaviour of the motor depends on the value of current that I program with M906 command: if it is too low the motor doesn't spin at all (but emit some noise), if it is too high it tend to overheat rapidly (it is a small motor). For this reason I think that the TVAL register of L6474 is programmed correctly but is not read (or interpreted) correctly.

Please, feel free to ask me for other tries!

[log.txt](https://github.com/MarlinFirmware/Marlin/files/4078847/log.txt)

Bob-the-Kuhn, 2020-01-17:
Your SPI chain is not working.  The status print hex and binary values should look something like this:
```
N22 M122*18
AXIS: Y     status: fe12   1111 1110 0001 0010   L6474
...OUTPUT: ON    BUSY:  NO    DIR: REVERSE   Last Command: VALID    COMPLETED
...THERMAL: OK                OVERCURRENT: NO    STALL: NA    STEP-CLOCK MODE: NA   UNDER VOLTAGE LOCKOUT:  NO
AXIS: Y2    status: fe02   1111 1110 0000 0010   L6474
...OUTPUT: ON    BUSY:  NO    DIR: REVERSE   Last Command: VALID    COMPLETED
...THERMAL: OK                OVERCURRENT: NO    STALL: NA    STEP-CLOCK MODE: NA   UNDER VOLTAGE LOCKOUT:  NO
AXIS: E0    status: fe12   1111 1110 0001 0010   L6474
...OUTPUT: ON    BUSY:  NO    DIR: FORWARD   Last Command: VALID    COMPLETED
...THERMAL: OK                OVERCURRENT: NO    STALL: NA    STEP-CLOCK MODE: NA   UNDER VOLTAGE LOCKOUT:  NO

N23 M906*29
M906
AXIS: Y     status: ff92   1111 1111 1001 0010
...OverCurrent Threshold:  4 (1875.00 mA)   TVAL: 47 (1500.00 mA   Motor Status: NA)
...MicroSteps: 16   ADC_OUT: 21   Vs_compensation: NA
...KVAL_HOLD: NA   KVAL_RUN : NA   KVAL_ACC: NA   KVAL_DEC: NA   V motor max =  NA
...SLEW RATE: 110V/uS
AXIS: Y2    status: ff82   1111 1111 1000 0010
...OverCurrent Threshold:  8 (3375.00 mA)   TVAL: 41 (1312.50 mA   Motor Status: NA)
...MicroSteps: 128   ADC_OUT: 21   Vs_compensation: NA
...KVAL_HOLD: NA   KVAL_RUN : NA   KVAL_ACC: NA   KVAL_DEC: NA   V motor max =  NA
...SLEW RATE: 110V/uS
AXIS: E0    status: ff92   1111 1111 1001 0010
...OverCurrent Threshold:  4 (1875.00 mA)   TVAL: 47 (1500.00 mA   Motor Status: NA)
...MicroSteps: 16   ADC_OUT: 21   Vs_compensation: NA
...KVAL_HOLD: NA   KVAL_RUN : NA   KVAL_ACC: NA   KVAL_DEC: NA   V motor max =  NA
...SLEW RATE: 110V/uS
```

Assuming that you are using the last config files you posted, I don't understand the problem.  They look correct.  Please post your **pins_STEVAL_3DP001V1.h** file.  maybe the problem is there.

---

Hmmm  ....  looks like the M906 status prints are wrong.  They should be the same as the M122 results.  One more bug!

Bob-the-Kuhn, 2020-01-18:
Here's the corrected M906.

[M906.zip](https://github.com/MarlinFirmware/Marlin/files/4079775/M906.zip)

Bob-the-Kuhn, 2020-01-18:
Please delete the **.platformio\packages\framework-arduinoststm32** directory and the **\.pio** directory & see if the **SWITCH_ENABLED_DUAL_X_CARRIAGE** errors go away.

They disappeared for me when I deleted those two directories.

fradaxx, 2020-01-18:
Please note that I don't have anymore errors related to SWITCH_ENABLED_DUAL_X_CARRIAGE after the substitution of the drivers.h files you shared yesterday here https://www.dropbox.com/s/bo4b4zcdwq35ugo/drivers.zip?dl=0.

Probably the problem is related to the pins_STEVAL_3DP001V1.h file; if I use this version
[pins_STEVAL_3DP001V1.txt](https://github.com/MarlinFirmware/Marlin/files/4080149/pins_STEVAL_3DP001V1.txt)
with your last version of the M906.cpp the board is unstable and restarts periodically due to watchdog reset. Also the M906 command returns always 0 for all read registers.

If I use this version (included in the current bugfix-2.0.x)
[pins_STEVAL_3DP001V1_2.txt](https://github.com/MarlinFirmware/Marlin/files/4080152/pins_STEVAL_3DP001V1_2.txt)
the board is stable (doesn't reset by watchdog) but command M906 doesn't work in read mode (it returns always 0 for all registers).

boelle, 2020-01-23:
@fradaxx so this one is solved ?

fradaxx, 2020-01-23:
@boelle no, unfortunately the problem with M906 command persists: it seems that it works only in "write" mode but not in read mode. Stall currents appear to be correctly configured but not correctly read because the M906 command (whitout arguments) returns always all 0s.
However for my current needs this behaviour is acceptable; I will try to solve the reading problem in next days.
However I don't have any precise scheduling; so, if you want we can close this bug for the moment.

boelle, 2020-01-23:
no no, we now know it could be a week or 2 before you report back :-D

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle