MarlinFirmware/Marlin #16566: LCD Positioning error in Z menu, for menu_motion.cpp

andylh, 2020-01-14:
The position of the menu item for the Z menu when doing an 0.025mm correction is incorrect being too far to the left. The change lines up the menu item with those above.

Needs the following change in line 252 of menu_moition.cpp (Change the start position from 0 to 1) 

MENU_ITEM_ADDON_START(1);

boelle, 2020-01-14:
changed the title since you want something specific changed

borland1, 2020-01-16:
I don't see that offset on my 12864 graphic display.  But I see a problem that is inconsistent with other menus. Namely, the solid lines above and below the heading "Move Z".
Menu looks like this, but with solid lines above and below the 'Move Z' header, plus above and below menu selection..

  Move Z
Move Axis -^
Move 1mm ->
Move 0.1mm ->
Move 0.025mm ->

Big problem is when 'Move 1mm' is selected there are four solid lines on the screen, making it difficult to tell what menu item is currently selected.

Suggest removing the header "Move Z" and re-label each menu selection like this:

Move Axes -^
Move Z 1mm ->
Move Z 0.1mm ->
Move Z 0.025mm ->

That could apply to other Axis menus as well.

ellensp, 2020-02-02:
This does seem to be a bug. 

On a 20x4 char display:
The menu "Move Z|Move 0.025m"  is one character left of where it should be, interfering with the ">" cursor on that line.
If you apply the patch "MENU_ITEM_ADDON_START(1);"  then this does correct the display.

BUT

On a 12864 GLCD display: 
All the menus are left justified as there is no need for a ">" cursor, its uses boxes.
The menu "Move Z|Move 0.025m" is in the correct location on this display.
If you apply the patch "MENU_ITEM_ADDON_START(1);"  then this menu is one character to the right of where it should display.

What does this mean?

There does look to a bug, but its is not as simple as changing the start position.

ellensp, 2020-02-03:
After much digging in circles I have found the issue (but not solved it, yet) 
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/lcd/menu/menu_motion.cpp#L250
This routine adds a blank menu item, then comes back and updates the lcd line with the custom text using  lcd_put_u8str(tmp). The issue is that displaying the menu is LCD dependent. This is handled differently if your on a 20x4 or a 12864. But lcd_put_u8str is common across screens.

ellensp, 2020-02-03:
Looked at several ways to fix this, went for the simplest. 
Which is to set MENU_ITEM_ADDON_START(1); only on character displays.

boelle, 2020-02-03:
@andylh what board do you use?

ellensp, 2020-02-04:
This fix is now part of marlin bugfix 2.0.x.

closed 2020-02-04 by boelle