MarlinFirmware/Marlin #16567: M218 T1 X??? with Dual_X_Carriage

pbrgle1, 2020-01-14:
Hello,

I've been using dual x carriage for a while and it seems that M218 T1 for X doesn't work. 

It's working fine in the Y direction, but the X position doesn't change. If I want to calibrate the distance between the extruders, I have to go back to the firmware and change #define X2_MAX_POS and reflash, which is really annoying.

Any idea why it's not respecting the X offset?

I'm using the 2.0 Bugfix from a week ago.

boelle, 2020-01-14:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
[Tom's 3D Forums](https://discuss.toms3d.org/)
Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
[Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

InsanityAutomation, 2020-01-14:
@boelle if this is occurring its a bug as M218 should work. Ill update my trex and test it soon. Asked a couple questions to verify on discord, but this should be reopened.

boelle, 2020-01-23:
@pbrgle1 how often do you update? bugfix is updated almost daily

boelle, 2020-01-30:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

pbrgle1, 2020-02-02:
Not that often to be honest. I don't know how to update without having to reconfigure everything from scratch, so I only update when I have the time to do so (which hasn't been often lately).

boelle, 2020-02-02:
use a tool like winmerge to compare config files and then copy over your settings

that is how i do it now but yes before i found winmerge it was a pain to update

boelle, 2020-02-03:
@pbrgle1 when you have time can i ask what board you use?

pbrgle1, 2020-02-03:
I have a Rumba.

Primož Brglez

boelle, 2020-06-23:
@pbrgle1 

when you have time please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

closed 2020-07-29 by github-actions[bot]