MarlinFirmware/Marlin #16568: TMC2209 QUICK_HOME not working with sensor-less homing

tmartin007, 2020-01-14:
**Description** 

This is very similar to #10075, which had been closed almost a year ago.

This concerns Marlin-bugfix-2.0.x running on a SKR 1.1V Pro with 2209 drivers, controlling a CNC 6040 - not for milling, but for building a special pick and place robot.

Sensor-less homing is basically working, however:

**Steps to Reproduce**
1) QUICK_HOME is enabled in Configuration_adv.h (attached)
    However: 
**Actual behavior** 
the X- and Y-axis movements happen in sequence 

**Expected behavior** 
rather than simultaneously when issuing a G28XY

**Steps to Reproduce** 
2) X_BED_SIZE and Y_BED_SIZE (thus X_MAX_POS and Y_MAX_POS) are correctly defined.

**Expected behavior**
     E.g.: G0X1000 should either return an error or only move to X_MAX_POS, since "1000"  
    exceeds X_MAX_POS (same issue with the Y-axis/Z-axis), or, better:

    the stall-guard should kick in, in the same fashion as it is causing the homing to work correctly. 

**Actual behavior**
    Instead, all steps are executed regardless, the carriage neither stops  
    at e.g. X_MAX_POS, nor is X_STALL_SENSITIVITIES respected. The carriage bounces right into 
    the physical limit of the machine (and keeps driving the motor until marlin believes that the 
    destination position has been reached).

boelle, 2020-01-14:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

tmartin007, 2020-01-15:
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4062401/configuration.zip)

Thanks for your prompt comment!

Hopefully, the upload word this time ...

tmartin007, 2020-01-15:
ISSUE 1) has suddenly disappeared - I have not changed a thing, though ... - I'll report, should it re-appear. 

ISSUE 2) remains

tmartin007, 2020-01-17:
Hello,

ISSU 1) is back
but I found a reproducible pattern:
The issue seems to be related to the X_ and Y_STALL_SENSITIVITY.

With values up to 137, QUICK_HOME works as expected.
With values between 137 and 156, both axis move simultaneously for a while (the lower the value, the longer) until the y-axis suddenly stops, only to resume once the x-axis is done homing.
With values beyond 156, QUICK_HOME does not work, instead both axis move in sequence.

I would prefer a higher sensitivity, since it seems that lower sensitivities yield less accurate homing positions.

ISSUE 2) still remains

boelle, 2020-01-23:
how often do you update? bugfix is updated almost daily

tmartin007, 2020-01-25:
Thanks for that hint. - So I have installed yesterdays version of Marlin-bugfix-2.0.x.

Regarding the described issues the behaviour is the same.

There is a **3rd issue** (I already had discovered before updating yesterday):

Oftentimes (about 6 out of 10 times), it takes 2, (sometimes 3) consecutive attempt, before homing works correctly. - On the first and second attempt, Marlin seems to think, the X-, Y- or Z-axis are already homed and just performs the HOMING_BACKOFF_MM on that axis. 

Should I open a separate issue for this, since it does not depend on the QUICK_HOME option ?

boelle, 2020-01-25:
yep, that would be the best i think

boelle, 2020-01-28:
btw... even on prusa printers with sensorless homing there is a deviation of about 0.3mm so its not a feature i would recommend

use endstops, they have been proven to work for ages, sensorless is just a gimmick

closed 2020-02-29 by thinkyhead

tmartin007, 2020-03-11:
Sorry for asking: since this issue is closed, does this mean it is solved - as in "bug fixed" (and I have missed it) ?

thinkyhead, 2020-03-11:
It is presumed solved. Please confirm.

tmartin007, 2020-03-12:
I do confirm - this appears to be solved now. - Thanks!
However, #17128 occurred - and might be related ?