MarlinFirmware/Marlin #16569: Premature KILL on idle time (M85 need proper reset somewhere

AndKe, 2020-01-14:
### Description

If printer is preheating, left alone for a while, starting a job result in KILL due to inactivity

### Steps to Reproduce

1: Connect
2: M85 S30
3: M85 S30  ;  YES, repeat the command after one second or two.
4: observe KILL

Reconnect (step 1)
do M999 , or else it will KILL on step 2
now you can reproduce it over again. (step3)

So: 
If your jobs start with M85 S300  and end with M85 S0  , then any cancelled job that do never get to S0 , will fail instantly on new M85 Sxxx

**Expected behavior:** [What you expect to happen]
expected it not to fail like that, 

**Actual behavior:** [What actually happens]
```
2019-01-02 16:28:05,921 - Recv: ok
2019-01-02 16:28:05,924 - Send: N5 M85 S720*93
2019-01-02 16:28:05,938 - Recv: ok
2019-01-02 16:28:05,942 - Send: N6 G28*21
2019-01-02 16:28:05,968 - Recv: Error:KILL caused by too much inactive time - current command: M85 S720
2019-01-02 16:28:05,970 - Recv: Error:Printer halted. kill() called!
2019-01-02 16:28:05,973 - Changing monitoring state from "Printing" to "Error: KILL caused by too much inactive time - current command: M85 S720 - Printer halted. kill() called!"
2019-01-02 16:28:05,987 - Changing monitoring state from "Error: KILL caused by too much inactive time - current command: M85 S720 - Printer halted. kill() called!" to "Offline (Error: KILL caused by too much inactive time - current command: M85 S720 - Printer halted. kill() called!)"

```

#### Additional Information
 this is recent master 2.0-bugfix

ellensp, 2020-01-15:
the test is ELAPSED(ms, gcode.previous_move_ms + max_inactive_time)
M85 just resets resets max_inactive_time, but ms is now greater than previous_move_ms...
give this a try (I can't test with real hardware atm.) 

edit src/gcode/control/M85.cpp so it looks like the following.  

```
void GcodeSuite::M85() {

  if (parser.seen('S')) { 
    max_inactive_time = parser.value_millis_from_seconds();
    reset_stepper_timeout();
  }
} 
```

ellensp, 2020-01-17:
This fixed it. PR submitted.

AndKe, 2020-01-17:
Thank you. - you beat me to testing it..

closed 2020-01-21 by boelle