MarlinFirmware/Marlin #16570: [BUG] (BLtouch V3 scratching bed, doesn't seem to compensate with Z-axis)

sardeepskhera, 2020-01-15:
### Bug Description

So I got a genuine BLTouch V3 for my Ender 3 and after a variety of different issues I finally got it working as intended. However after setting the Z-offset and have it pass the "paper test", I would run the G29 for a print (and observe it set the nozzle at the correct height) only for it to scratch into bed on some of the corners.

- I'm running TH3D's firmware

- bilinear bed leveling enabled

- x and y probe offsets are correct

- uncommented 'restore_leveling_after_g28'

- enable a 5x5 grid, to no avail

(Some hardware stuff if it matters)

- Leveled it manually before running the ABL

- Used a bed visualizer plugin in Octoprint to get within 0.2 mm

- Leveled the x-gantry

- I do have Dupont extensions attached to the cable for the probe because it wouldn't reach my board (if that's important)

Here's the configs
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4062331/Configuration.zip)

No matter what, same result every time! I literally am losing my mind trying to figure out the cause. Everything else I've looked up makes it seems that the BLtouch should compensate for the variations of an uneven bed much worse than what I have currently. So what gives?

BeatSlayer, 2020-01-15:
Your using:
https://github.com/houseofbugs/TH3D-Unified-U1.R2
This is for Marlin firmware issues.

sardeepskhera, 2020-01-15:
> Your using:
> https://github.com/houseofbugs/TH3D-Unified-U1.R2
> This is for Marlin firmware issues.

I apologize. Thank you!

closed 2020-01-15 by sardeepskhera