MarlinFirmware/Marlin #16571: [BUG] SDHC card doesn't work on STM32 (BTT_SKR_MINI_E3_V1_2)

dp250f, 2020-01-15:
### Bug Description
SDHC cards don't initialize with STM32 (LPC1768 seems to work)

### My Configurations
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4063227/Configs.zip)

### Steps to Reproduce
Compile firmware with SDSUPPORT enabled for an STM32 board
Observe SDSC cards work but SDHC cards don't.

**Expected behavior:** [What you expect to happen]
Insert SDHC card (>8GB) and initialize it - it should work
Insert SDSC card (<=8GB) and initialize it - it should work

**Actual behavior:**
Insert SDHC card (>8GB) and attempt to initialize it - it doesn't work
Insert SDSC card (<=8GB) and initialize it - it works

#### Additional Information
I'm using BOARD_BTT_SKR_MINI_E3_V1_2 and the following SD cards to test:
128MB SDSC (works)
4GB Class 4 SDSC (works)
16GB Class 10 SDHC (doesn't work)

Lord-Quake, 2020-01-15:
I have the board in my Ender-3 and cannot replicate your issue.
I'm using SanDisk Ultra with 32GB. I've tried all possible variations. Initialization has worked every time.
I can't find anything relevant in your configs why it shouldn't work, at least with my little know-how on the subject.

Maybe changing to this in configuration_adv might help:
`#if ENABLED(SDSUPPORT)`
`  #define BLOCK_BUFFER_SIZE 32 // SD,LCD,Buttons take more memory, block buffer needs to be smaller`
`#else`
` #define BLOCK_BUFFER_SIZE 32 // maximize block buffer`
`#endif`

or

RudyBenner, 2020-01-16:
Try <4gig card formatted FAT16. Or larger card partitioned to <4gig

Lord-Quake, 2020-01-16:
I'm using FAT32 with mine.

thisiskeithb, 2020-01-16:
I just tried with PNY & SanDisk 16GB SDHC cards formatted as FAT32 on an SKR Mini E3 1.2 and they work fine, so it's probably not SDHC specific.

From what I've seen on Facebook, switching to a different card/manufacturer usually solves the problem.

boelle, 2020-01-16:
looks like this is more a hardware issue then?

dp250f, 2020-01-17:
Interesting. The only SDHC card I had tried is a 16GB Toshiba FlashAir W-04. It works fine in my SKR 1.3 and my older PrintrBoard RevD (using Marlin 1.1.9)
This card says it's a 16GB, SDHC, UHS-1, Speed Class 10, UHS Speed Class 3.

I found a couple other random SD cards which work:
PNY 4GB SDHC, Speed Class 4 (those are the only specs I can see on it)
Unbranded 4GB SDHC, Class 4 (those are the only specs I can see on it)

I also tried formatting the FlashAir card with a 4GB FAT32 partition and even a 1GB FAT partition, neither would initialize. Seems to only be a problem with FlashAir on STM32

thisiskeithb, 2020-01-17:
> Seems to only be a problem with FlashAir on STM32

If someone else using a FlashAir could runs some tests, that would be helpful. I don't own one, or I'd check.

dp250f, 2020-01-18:
I don't really know enough to say it's a hardware problem or not.
I can say this FlashAir card works with Marlin 2.0.1 on LPC1768 and Marlin 1.1.9 on AT90USB1286, but does not work with Marlin 2.0.1 on STM32F1.

Not sure if this is helpful, but it also reads fine in the BTT-TFT35-E3-V3.0 when it's in Host mode. That controller uses an STM32F207 VCT6. Given - it's not running Marlin, but shows that card can work with at least that variety of STM32.

boelle, 2020-01-21:
have you tried with bugfix 2.0.x ? its newer than 2.0.1

dp250f, 2020-01-21:
Sorry about that - it is bugfix 2.0.x, last one I tried was downloaded 1/19/2020
> have you tried with bugfix 2.0.x ? its newer than 2.0.1

boelle, 2020-01-30:
but a normal sd card works fine? i would then expect it to be a hardware issue of some sort

from what i know a flashair is just and sd card with a wifi module build in so that you can transfer files to it over wifi, but seen from the controller side its just an sd card

dp250f, 2020-01-30:
> 
> 
> but a normal sd card works fine? i would then expect it to be a hardware issue of some sort
> 
> from what i know a flashair is just and sd card with a wifi module build in so that you can transfer files to it over wifi, but seen from the controller side its just an sd card

I would agree, but that exact card works fine with other devices and with Marlin 2.0.x and 1.1.9 with different processors. It is clearly working normally. I would be happy to donate a FlashAir card to a Marlin developer to get it working with STM32F1 and Marlin 2.0.x if that would help.

boelle, 2020-01-30:
@thinkyhead @shitcreek @Vertabreak @InsanityAutomation are the only ones i know, are you US based?

dp250f, 2020-01-30:
> 
> 
> @thinkyhead @shitcreek @Vertabreak @InsanityAutomation are the only ones i know, are you US based?

Yes I am.

spattinson, 2020-02-03:
I have an SKR Pro, and 16GB flashair, it works fine on bugfix branch reading files and saving EEPROM. I was using a MultiSPI fork to enable access to both card readers, the inbuilt micro one on the skr pro and the external one on the lcd. When I used that my SD card reader on the LCD only worked with one out of 4 cards I have, the only one that did work was SDHC, not sure what the others were. I downloaded bugfix over the weekend it reads all cards.

Now my flashair has a different issue, I think it may have got its firmware updated at some point now STA mode does not work, only AP mode, including when its in my laptop or my skr pro, and in AP mode and that is slow, so I've given up using the wifi side of it.

While we talking about firmware and flashair, the firmware on the flashair times out and disables wifi if no one connects in a certain time. You can trigger the wifi to back  on by changing the attributes of a designated file on it. If there was a gcode command to do that it may make life easier. I believe prusa firmware can do this.

dp250f, 2020-02-04:
> I have an SKR Pro, and 16GB flashair, it works fine on bugfix branch reading files and saving EEPROM. I was using a MultiSPI fork to enable access to both card readers, the inbuilt micro one on the [...]

This indicates it does work with STM32F4 HAL. Seems there is something different when it comes to SD access between F1 and F4.

dp250f, 2020-03-08:
I compiled from bugfix 2.0.x today and this still isn't working. I can still send a developer a free FlashAir card. I'll figure out how to ship it internationally if necessary.

dp250f, 2020-04-08:
@victorpv @xC0000005 @thisiskeithb @tpruvot 
According to README.md in Marlin\src\HAL\STM32F1, you guys are the experts. I can donate any hardware required to test (FlashAir card, BTT_SKR_MINI_E3_V1_2, microSD to SD adapter)
It would be worth it to me. Thanks in advance.

thisiskeithb, 2020-04-08:
Odd. I shouldn’t be listed there.

xC0000005, 2020-04-09:
I have an SKR E3-DIP V1.0. Wonder if it would have the same problem. What environment are you using to build for that board? (I want to see which of the F1 hals it is using).

dp250f, 2020-04-09:
I'm using VSCode with PlatformIO. The last bugfix build I tried was from 3/31/2020.
I've tried both STM32F103RC_btt_512K and STM32F103RC_btt_512K_USB environments.

xC0000005, 2020-04-09:
Ok, that uses HAL_STM32F1, which is base on the original STM32Duino core. The F4/F7 (and 070) are based on ST’s arduino core. I would try reducing SPI speed (in configuration.h, SPI_SPEED) before we go mailing boards around.

dp250f, 2020-04-09:
Do you have a FlashAir card you can try with your SKR E3-DIP 1.0? I'm curious if you get the same results I'm getting.

dp250f, 2020-04-11:
Also, I tried it using SPI_HALF_SPEED, SPI_QUARTER_SPEED, and SPI_EIGHTH_SPEED.
Made no difference.

boelle, 2020-06-20:
@dp250f is the issue still there?

dp250f, 2020-06-21:
The last bugfix I have tried is from 5/29/2020, and the issue persists.

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

dp250f, 2020-06-28:
I compiled and tested bugfix 2.0.x which I downloaded today (commit 48151d1)
FlashAir card still cannot be initialized by Marlin.

FusionCHP, 2020-07-14:
dp250F, I have the same exact problem as you.  The wifi card is getting 3.3 volts, but I can't see it on my PC as a mapped drive.  I tried writing one byte to the card to initialize it like what is done on Prusa printers but it didn't work or I wasn't doing it correctly or writing the byte too late.

"Not sure if this is helpful, but it also reads fine in the BTT-TFT35-E3-V3.0 when it's in Host mode. That controller uses an STM32F207 VCT6. Given - it's not running Marlin, but shows that card can work with at least that variety of STM32."

Same for me.

FusionCHP, 2020-07-16:
> 
> 
> I compiled and tested bugfix 2.0.x which I downloaded today (commit [48151d1](https://github.com/MarlinFirmware/Marlin/commit/48151d177890d102c3e7adbc00ec5ee6f367a8fb))
> FlashAir card still cannot be initialized by Marlin.

Did you get your Toshiba WiFi SD to work?  I checked the sd card voltage and it is steady at 3.3 volts and doesn't dip.  My wifi on the SD stays on.  Some I'm guessing communication is trying to take place between the SKR and the SD.

xC0000005, 2020-07-17:
This link indicates there was a hack at some point t to support these. Someone should see if it is still there and missing in a HAL. 

https://reprap.org/forum/read.php?219,844807

AnHardt, 2020-07-17:
I copied  that patch over from PrusaFirmware. At that time my FlashAir worked on a MEGA + RAMPS + display. (can't find the card these days - so i can't test)
The additionally sent byte is still there. It now resides in 
```cpp
void Sd2Card::chipDeselect() {
  extDigitalWrite(chipSelectPin_, HIGH);
  spiSend(0xFF); // Ensure MISO goes high impedance
}
```
and seems to be called in many more situations.

FusionCHP, 2020-07-17:
The TFT35 V3.0 uses an STM32 processor and can read the Toshiba WiFi card fine.  I looked at the SD card code for the TFT35 firmware and its different from Marlin, and that’s where my skills pretty much end.

xC0000005, 2020-07-17:
Is it using SDIO instead of SPI?

xC0000005, 2020-07-17:
Ok, it’s not in SDIO mode, it’s SPI. They initialize in low speed, though. I’m not sure if you’ve tried reducing SPI speed or not (if you have, sorry).

FusionCHP, 2020-07-17:
I didn't, but dp250f did.  So, I didn't try.

FusionCHP, 2020-07-17:
xC0000005, Here is the TFT35 firmware (not sure if it helps).  I deeply appreciate you looking at this.

FusionCHP, 2020-07-17:
Sorry, the link didn't post:  https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

xC0000005, 2020-07-17:
Given how much their code is retrying, it might not hurt to enable SD_CHECK_AND_RETRY and see.

FusionCHP, 2020-07-18:
I did try this.

FusionCHP, 2020-07-21:
I tried reducing speed but no luck.

FusionCHP, 2020-08-09:
Tried latest version of Marlin and still have the issue.

dp250f, 2020-09-11:
Flashed Bugfix 2.0.x downloaded today - still cannot use FlashAir card with STM32F103RC.

chilman408, 2020-11-05:
Hello, wondering whey this issue was closed...  Is this fixed?  I'm having the same issue with Marlin 2.0.7.2 and wondering if there is a solution.

chilman408, 2020-11-05:
> ### Bug Description
> SDHC cards don't initialize with STM32 (my other board - LPC1768 - reads SDHC cards fine)
> 
> ### My Configurations
> [Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4063227/Configs.zip)
> 
> ### Steps to Reproduce
> Compile firmware with SDSUPPORT enabled for an STM32 board
> Observe SDSC cards work but SDHC cards don't.
> 
> **Expected behavior:** [What you expect to happen]
> Insert SDSC card (<=8GB) and initialize it - it should work
> Insert SDHC card (>8GB) and initialize it - it should work
> 
> **Actual behavior:**
> Insert SDSC card (<=8GB) and initialize it - it works
> Insert SDHC card (>8GB) and attempt to initialize it - it doesn't work
> 
> #### Additional Information
> I'm using BOARD_BTT_SKR_MINI_E3_V1_2 and the following SD cards to test:
> 128MB SDSC (works)
> 4GB Class 4 SDSC (works)
> 16GB Class 10 SDHC (doesn't work)

Hello, were you able to get this problem fixed?

sjasonsmith, 2020-11-05:
It looks like this was incorrectly closed. The purpose of the "stale bot" is to identify and close issues for which nobody is still interesting in pursuing a fix. A simple reply should prevent the issue from closing. In this case some manual intervention caused it to close anyway.

I'll go ahead and open this back up, but don't want to set any unrealistic expectations. This is a niche product, and I don't when (or if) somebody will spend the time to actually resolve it.

dp250f, 2020-11-05:
Never got it to work properly. My current workaround is to print from my BTT TFT35 in host mode instead of Marlin mode.

chilman408, 2020-11-05:
> Never got it to work properly. My current workaround is to print from my BTT TFT35 in host mode instead of Marlin mode.

I see....that is not a good workaround as that is going through a serial connection to the printer....and I don't have a TFT35...lol

closed 2020-12-13 by github-actions[bot]