MarlinFirmware/Marlin #16577: [FR] Z Safe Homing for IDEX (with Z offset)

pbrgle1, 2020-01-15:
I have an IDEX printer with nozzles at different heights. My Z endstop position is at the higher nozzle, which means there is a potential for the lower nozzle to crash into the bed. 

In my case, the X home position of T0 is -84 and for T1 it's 284. My bed is 200mm wide which means that a safe X position for T0 would be between -84 and 0 and for T1 it's 200-284.

It would be great, if it were possible to specify safe XY points for each tool with Z_SAFE_HOMING.

Roxy-3D, 2020-01-19:
I'm not understanding your post (or problem).    Are you confusing the Home position with the Park position?    Typically, the Home position is on the bed, and usually in the center of the bed.

Can you edit your post to be a little more clear?

pbrgle1, 2020-01-20:
Basically, I want to prevent any of the two X axis from being within the bed region X0 - X200 while homing Z.

Safe Z currently makes sure that the probe is not outside the bed area. 
I would need the reverse - to make sure the extruder is not over the bed.

This is my situation:
[image]

And from that you can imagine something like this can happen if the lower nozzle is above the bed while homing Z.

[image]